openhab/org.openhab.binding.zigbee #577: Workaround causes serial port discovery issues

wborn, 2020-05-13:
The binding currently [uses serialPortManager.getIdentifiers()](https://github.com/openhab/org.openhab.binding.zigbee/blob/c4c40739bf9cebc251c316bd1ab37f5f408bd904/org.openhab.binding.zigbee/src/main/java/org/openhab/binding/zigbee/handler/ZigBeeSerialPort.java#L139-L145) to get the `SerialPortIdentifier`.

This workaround was added in https://github.com/openhab/org.openhab.binding.zigbee/pull/335 by @triller-telekom.

The workaround causes serial port detection issues because the `getIdentifiers()` method will only return discovered ports.

As a result:
* Undiscovered (non-standard) RXTX ports will not work this way (e.g ports defined in udev rules), the [transport adds undiscovered ports](https://github.com/openhab/openhab-core/blob/178ffdf9c275fd086f76f02d6be64ca29edbdc35/bundles/org.openhab.core.io.transport.serial.rxtx/src/main/java/org/openhab/core/io/transport/serial/internal/SerialPortUtil.java#L51-L63) to `gnu.io.rxtx.SerialPorts` so they can be used without users having to configure this
* RFC2217 ports cannot be used because there is [no discovery logic](https://github.com/openhab/openhab-core/blob/178ffdf9c275fd086f76f02d6be64ca29edbdc35/bundles/org.openhab.core.io.transport.serial.rxtx.rfc2217/src/main/java/org/openhab/core/io/transport/serial/rxtx/rfc2217/internal/RFC2217PortProvider.java#L50-L54) for these ports.

We've upgraded nrjavaserial in OH3 so we should retest if this is still an issue.
We should also test if it is fixed with nrjavaserial 5.1.0+ because it contains [fixes for hardware that gets removed](https://github.com/NeuronRobotics/nrjavaserial/releases/tag/5.1.0).

If this is no longer an issue we should remove the workaround by instead detecting that a controller is disconnected by just opening the port and handling any resulting exceptions.

If it is still an issue, we should create an issue for it in the nrjavaserial [issue tracker](https://github.com/NeuronRobotics/nrjavaserial/issues) with a reproduction scenario. That way we can fix the root cause and don't need to add such workarounds (causing discovery issues) to other add-ons.

Related to:
* https://github.com/openhab/openhab-addons/pull/7609#discussion_r423830007
* https://github.com/openhab/openhab-core/pull/1471
* https://github.com/openhab/openhab-core/issues/1462

wborn, 2020-05-22:
The crash/exit can also be reproduced with other USB devices and will probably be resolved when https://github.com/NeuronRobotics/nrjavaserial/issues/180 is fixed.

wborn, 2020-05-27:
The workaround can be removed in OH3 because nrjavaserial 5.2.1 no longer exits the JVM when in certain scenarios when ports are closed/reopened.

cdjackson, 2020-09-26:
> nrjavaserial 5.2.1 no longer exits the JVM 

Unfortunately that's not true on a Mac at least.

moodyblue, 2021-02-05:
It would be good if this can be fixed. Sonoff ZBBridge can be used with with zigbee binding, but only with socat because RFC2217 does not work.

wborn, 2021-05-07:
> Unfortunately that's not true on a Mac at least.

Hopefully your Mac issue will be fixed by https://github.com/NeuronRobotics/nrjavaserial/pull/211

dschall, 2021-07-08:
I've created a PR to make Zigbee work with RFC2217:
https://github.com/openhab/org.openhab.binding.zigbee/pull/661