openhab/org.openhab.binding.zigbee #578: Update gnu.io dependency or use serial transport for OH3 compatibility

wborn, 2020-05-14:
Looks like the OH3 distro build fails because the OH3 zigbee binding still depends on gnu.io.

See: https://ci.openhab.org/job/openHAB3-Distribution/1850/console

OH3 uses nrjavaserial 5.0.2 with https://github.com/openhab/openhab-core/pull/1464 which has the same API.

Now I did create https://github.com/openhab/org.openhab.binding.zigbee/pull/513 some time ago to get rid of this dependency here, but it is not merged on the master branch. So I suggest the @openhab/zigbee-maintainers to apply it to the master branch or update the gnu.io import packages similar to https://github.com/openhab/openhab-addons/pull/7589.

cdjackson, 2020-05-14:
I don’t think the master branch has been maintained at all, so it is not just the serial classes that will fail - but all the ESH classes have not been migrated to OH either. At some stage I will probably look to migrate these over - once OH3 is a bit more “in use” but for now my main maintenance needs to remain on OH2 as I have users that I need to support.

wborn, 2020-05-14:
I don't use this add-on so it's not important to me that it is completely up to date as long as the distro build succeeds. If it is difficult to fix the gnu.io issue for OH3 with the zigbee binding we can also remove it from the OH3 distro or add an older nrjavaserial bundle. Another way could be to do just a one time sync of the 2.5.x branch and then sync it again when needed.

cdjackson, 2020-05-14:
I guess the bistro will not build for OH3 until the 2.5 branch is refactored to change the namespace which I’ve not done yet as I need to support 2.5 still. I know we discussed this a while back, but I will refactor for OH3 once OH3 is a bit closer - in the meantime I’d suggest to simply remove ZigBee from the OH3 distro until this time to avoid having to rework everything.

wborn, 2020-05-14:
The namespace change was already done in https://github.com/openhab/org.openhab.binding.zigbee/pull/538. :wink:

cdjackson, 2020-05-14:
Oh :( That will presumably make refactoring from 2.5 branch now more difficult if there are modifications going on in both branches.

I would still suggest to remove ZigBee from the build rather than try and fix things in both branches. This issue is already resolved in the current branch, so once this is refactored for OH3 it will not be an issue.

kaikreuzer, 2020-05-15:
How shall we best exclude it @wborn? Changing the build plan is probably not possible, so we would rather have to remove it from the distro feature on master, wdyt?

wborn, 2020-05-15:
Yes that would probably be the quickest fix. The script we use for porting the other add-ons can probably also be slightly adapted to port the 2.5.x branch here.

kaikreuzer, 2020-05-15:
> Yes that would probably be the quickest fix.

Like https://github.com/openhab/openhab-distro/pull/1111?

wborn, 2020-05-15:
That will do! Let's close this issue.

closed 2020-05-15 by wborn