openhab/org.openhab.binding.zigbee #583: Only perform discovery on existing devices

cdjackson, 2020-06-06:
Currently the binding performs discovery on all devices - even those that are already `Things`. This is because the discovery service has no way to know if a device is already discovered and added as a `Thing`.

Since devices may (re)join the network at any time we either need to add devices when discovery is started, or add support for the OH background discovery concept.

We also need to handle one other case which is if Things are deleted, but not removed from the network. In this case, if we don't add all known things during discovery, we will loose devices completely.

falmanna, 2020-08-20:
Does it make sense to check on the existing things using `thingRegistery` when discovering existing devices after starting a scan?
```
if (node.getNetworkAddress() == 0 || thingRegistry.stream().anyMatch(thing -> thing.getUID().toString()
      .toLowerCase().trim().contains(node.getIeeeAddress().toString().toLowerCase().trim()))) {
      continue;
}
```
I've made this change and it seems to be working, but I am not sure if I missed something as I have little knowledge about the internals of the ZigBee binding.

cdjackson, 2020-08-20:
It's not allowed for the discovery service to interface with the thingRegistry. This is what we used to have in the past, but it was removed some time ago.

cdjackson, 2020-08-20:
Sorry - to be clearer - what I meant was this was originally part of ESH - I think it was called the "extended discovery service" and this feature was removed from the core a while back.

PR #600 (2020-08-27): Only perform discovery on existing devices
Don't perform discovery on discovered things
Closes #583
closed 2020-08-27 by cdjackson