openhab/org.openhab.binding.zigbee #585: Support for different IKEA Symfonisk E1744 button presses

nohum, 2020-06-15:
I have paired an [IKEA Symfonisk](https://www.ikea.com/at/de/p/symfonisk-fernbedienung-fuer-soundsystem-weiss-60370480/) with OpenHAB. Basically it just a dimmer if you rotate it in either direction (the binding detects a level control channel which is a dimmer type).

Additionally, it supports a single, double or triple press on its button. Within the Zigbee binding, a single press is detected as a toggle of the level control channel, but double and triple presses are not detected. I propose the feature of including a way to detect and provide a possibility to react on the different kind of presses. I have attached the log output of all presses so that it could be supported. 

Note: If somebody could point me in the right direction on how and where to implement support for this within the binding, I could also implement this myself and submit an PR.

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used | CC2531 |
| openHAB version  | 2.5.4 |
| Hardware         | Raspberry Pi 3 |
| Memory           | 1024 MB |
| Java version     | 1.8.0_191 |
| Devices          | IKEA Symfonisk E1744 |

### Logs

<details>
<summary>Output of a single press (working correctly)</summary>

```

18:43:12.590 [DEBUG] [531.network.impl.CommandInterfaceImpl] - <-- AF_INCOMING_MSG (FE 1B 44 81 00 00 01 00 CB BE 01 01 00 1F 00 33 95 67 00 00 07 18 31 0A 21 00 20 3C CB BE 1D 05)
18:43:12.618 [DEBUG] [531.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=27, apiId=44 81, data=FE 1B 44 81 00 00 01 00 CB BE 01 01 00 1F 00 33 95 67 00 00 07 18 31 0A 21 00 20 3C CB BE 1D 05, checksum=05, error=false
18:43:12.648 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=BECB/1, destinationAddress=0000/1, profile=0104, cluster=0001, addressMode=null, radius=0, apsSecurity=false, ackRequest=false, apsCounter=--, rssi=--, lqi=--, payload=18 31 0A 21 00 20 3C]
18:43:12.669 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=SERVER_TO_CLIENT, disableDefaultResponse=true, manufacturerCode=0, sequenceNumber=49, commandId=10]
18:43:12.693 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReportAttributesCommand [Power Configuration: BECB/1 -> 0000/1, cluster=0001, TID=31, reports=[AttributeReport [attributeDataType=UNSIGNED_8_BIT_INTEGER, attributeIdentifier=33, attributeValue=60]]]
18:43:12.715 [DEBUG] [.transaction.ZigBeeTransactionManager] - notifyTransactionCommand: ReportAttributesCommand [Power Configuration: BECB/1 -> 0000/1, cluster=0001, TID=31, reports=[AttributeReport [attributeDataType=UNSIGNED_8_BIT_INTEGER, attributeIdentifier=33, attributeValue=60]]] 
18:43:12.883 [DEBUG] [ee.dongle.cc2531.ZigBeeDongleTiCc2531] - Unhandled ZToolPacket type 0x4481
18:43:13.164 [DEBUG] [531.network.impl.CommandInterfaceImpl] - <-- AF_INCOMING_MSG (FE 17 44 81 00 00 06 00 CB BE 01 01 00 22 00 35 9C 67 00 00 03 11 32 02 CB BE 1D 07)
18:43:13.192 [DEBUG] [531.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=23, apiId=44 81, data=FE 17 44 81 00 00 06 00 CB BE 01 01 00 22 00 35 9C 67 00 00 03 11 32 02 CB BE 1D 07, checksum=07, error=false
18:43:13.217 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=BECB/1, destinationAddress=0000/1, profile=0104, cluster=0006, addressMode=null, radius=0, apsSecurity=false, ackRequest=false, apsCounter=--, rssi=--, lqi=--, payload=11 32 02]
18:43:13.241 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=true, manufacturerCode=0, sequenceNumber=50, commandId=2]
18:43:13.261 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ToggleCommand [On/Off: BECB/1 -> 0000/1, cluster=0006, TID=32]
18:43:13.281 [DEBUG] [.transaction.ZigBeeTransactionManager] - notifyTransactionCommand: ToggleCommand [On/Off: BECB/1 -> 0000/1, cluster=0006, TID=32] 
18:43:13.304 [DEBUG] [ee.dongle.cc2531.ZigBeeDongleTiCc2531] - Unhandled ZToolPacket type 0x4481
``` 

</details>

<details>
<summary>Output for double presses</summary>

```
18:44:05.503 [DEBUG] [531.network.impl.CommandInterfaceImpl] - <-- AF_INCOMING_MSG (FE 1B 44 81 00 00 01 00 CB BE 01 01 00 2C 00 20 1B 6A 00 00 07 18 33 0A 21 00 20 05 CB BE 1D 9D)
18:44:05.526 [DEBUG] [531.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=27, apiId=44 81, data=FE 1B 44 81 00 00 01 00 CB BE 01 01 00 2C 00 20 1B 6A 00 00 07 18 33 0A 21 00 20 05 CB BE 1D 9D, checksum=9D, error=false
18:44:05.550 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=BECB/1, destinationAddress=0000/1, profile=0104, cluster=0001, addressMode=null, radius=0, apsSecurity=false, ackRequest=false, apsCounter=--, rssi=--, lqi=--, payload=18 33 0A 21 00 20 05]
18:44:05.581 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=SERVER_TO_CLIENT, disableDefaultResponse=true, manufacturerCode=0, sequenceNumber=51, commandId=10]
18:44:05.612 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReportAttributesCommand [Power Configuration: BECB/1 -> 0000/1, cluster=0001, TID=33, reports=[AttributeReport [attributeDataType=UNSIGNED_8_BIT_INTEGER, attributeIdentifier=33, attributeValue=5]]]
18:44:05.642 [DEBUG] [.transaction.ZigBeeTransactionManager] - notifyTransactionCommand: ReportAttributesCommand [Power Configuration: BECB/1 -> 0000/1, cluster=0001, TID=33, reports=[AttributeReport [attributeDataType=UNSIGNED_8_BIT_INTEGER, attributeIdentifier=33, attributeValue=5]]] 
18:44:05.666 [DEBUG] [ee.dongle.cc2531.ZigBeeDongleTiCc2531] - Unhandled ZToolPacket type 0x4481
18:44:06.306 [DEBUG] [531.network.impl.CommandInterfaceImpl] - <-- AF_INCOMING_MSG (FE 1D 44 81 00 00 08 00 CB BE 01 01 00 3F 00 EC 24 6A 00 00 09 11 34 02 00 01 00 00 00 00 CB BE 1D 7F)
18:44:06.326 [DEBUG] [531.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=29, apiId=44 81, data=FE 1D 44 81 00 00 08 00 CB BE 01 01 00 3F 00 EC 24 6A 00 00 09 11 34 02 00 01 00 00 00 00 CB BE 1D 7F, checksum=7F, error=false
18:44:06.351 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=BECB/1, destinationAddress=0000/1, profile=0104, cluster=0008, addressMode=null, radius=0, apsSecurity=false, ackRequest=false, apsCounter=--, rssi=--, lqi=--, payload=11 34 02 00 01 00 00 00 00]
18:44:06.375 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=true, manufacturerCode=0, sequenceNumber=52, commandId=2]
18:44:06.403 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: StepCommand [Level Control: BECB/1 -> 0000/1, cluster=0008, TID=34, stepMode=0, stepSize=1, transitionTime=0]
18:44:06.428 [DEBUG] [.transaction.ZigBeeTransactionManager] - notifyTransactionCommand: StepCommand [Level Control: BECB/1 -> 0000/1, cluster=0008, TID=34, stepMode=0, stepSize=1, transitionTime=0] 
18:44:06.455 [DEBUG] [ee.dongle.cc2531.ZigBeeDongleTiCc2531] - Unhandled ZToolPacket type 0x4481
18:44:06.474 [DEBUG] [531.network.impl.CommandInterfaceImpl] - <-- AF_INCOMING_MSG (FE 1D 44 81 00 00 08 00 CB BE 01 01 00 41 00 FD 24 6A 00 00 09 11 34 02 00 01 00 00 00 00 CB BE 1D 10)
18:44:06.490 [DEBUG] [531.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=29, apiId=44 81, data=FE 1D 44 81 00 00 08 00 CB BE 01 01 00 41 00 FD 24 6A 00 00 09 11 34 02 00 01 00 00 00 00 CB BE 1D 10, checksum=10, error=false
18:44:06.512 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=BECB/1, destinationAddress=0000/1, profile=0104, cluster=0008, addressMode=null, radius=0, apsSecurity=false, ackRequest=false, apsCounter=--, rssi=--, lqi=--, payload=11 34 02 00 01 00 00 00 00]
[... cut ...] dId=2]
18:44:24.525 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX CMD: StepCommand [Level Control: BECB/1 -> 0000/1, cluster=0008, TID=3B, stepMode=1, stepSize=1, transitionTime=0]
18:44:24.542 [DEBUG] [.transaction.ZigBeeTransactionManager] - notifyTransactionCommand: StepCommand [Level Control: BECB/1 -> 0000/1, cluster=0008, TID=3B, stepMode=1, stepSize=1, transitionTime=0] 
18:44:24.559 [DEBUG] [ee.dongle.cc2531.ZigBeeDongleTiCc2531] - Unhandled ZToolPacket type 0x4481
```

</details>

TheNON75, 2020-07-17:
I would be already glad if the toggle could be used somehow as it seems like appears in debug only, isn't it?
Furthermore the dimmer should provide increment and decrement values. Currently reaches 0 or 100% within 30-40 degrees while it sends hex values and physically has no end points to any directions (you can rotate as many times as you want).
With the single key press I see no issue (unless the fact I cannot use it), as if you set up a rule eg within 1 second 2 presses or 2 seconds 3 presses, you can simulate the expected behavior, but first the button press should be somehow captured.

I am looking forward to see this in work.

nohum, 2020-07-29:
> I would be already glad if the toggle could be used somehow as it seems like appears in debug only, isn't it?
> Furthermore the dimmer should provide increment and decrement values. Currently reaches [...]

Yes, this is something I am also experiencing, but I guess that this is an OpenHAB limitation (I think this could be circumvented if the binding would send out dimmer channel events) .

cdjackson, 2020-07-29:
I don't have this device. If this sufficient interest I'll purchase one for integration and testing.

TheNON75, 2020-08-15:
What is considered as sufficient interest?
Shall we make a voting in the community or would donation help?

nohum, 2020-08-17:
I would support the development with a donation too.

cdjackson, 2020-08-17:
If you guys wanted to donate £11 each, or you can get a few others to chip in to the same amount, then I'll get one and take a look. That will cover the cost of the device and shipping (£19 total), and paypal fees...

I think it's this device - right? -:
https://www.ikea.com/gb/en/p/symfonisk-sound-remote-white-60370480/

nohum, 2020-08-18:
I would be in, where should we donate to? Regarding the link to the IKEA website: yes, this is the device.

cdjackson, 2020-08-18:
I think this [paypal link](https://www.paypal.com/donate/?token=Yu6VPCQ88pbTwbLhVgDCKxg8EPkXlfy-kXf1HbJKecbSl74jcvFqzYuwRLsCawa-xaMHFm&country.x=GB&locale.x=) should work.

nohum, 2020-08-20:
I tried the hyperlink but sadly it did not work for me (it tells me that the session is expired).

cdjackson, 2020-08-20:
Sorry - I thought I'd used this successfully in the past.

Try the following page - at the bottom is a donate button - this should work.

Thanks.

nohum, 2020-08-25:
Sadly there was no page linked in your last response, did you mean the [openHAB donate page](https://www.openhab.org/about/donate.html)?

cdjackson, 2020-08-25:
I'm not sure why that didn't come through. No - if you donate to OH, then this doesn't really help me (and this personally cost me around 2000USD in the past with ZWave :( ). 

Please use the following https://www.paypal.com/paypalme/jacksoncd

I've already ordered this a few days back - I think it's due early Sept.

nohum, 2020-08-25:
Thank you, I just sent you a donation.

TheNON75, 2020-08-25:
Donation sent too, thanks in advance!

cdjackson, 2020-08-25:
Thanks. I'll let you know when it arrives and I take a look...

cdjackson, 2020-09-10:
This arrived the a day or two ago...

This device is using the dimmer (level-control) step-up and step-down commands for the double and triple click. The binding maps the step commands into the dimmer channel and I'm not sure that I can change this without breaking other devices. Maybe we can add configuration to stop this, but that's a larger chunk of work than I'd like to do right now.

The step value is very small - 1% or less (it's 1 step - the binding might map this as percent or ~1/4% - I can't remember off the top of my head). In any case, unless you're clicking a lot, probably you won't notice the change in brightness/volume (and I guess not since the reported issue is that it does nothing - not that it changes the level).

So, what I can probably do is to create a separate channel that provides this information - it will still change the level, but you probably won't notice, but you'll also get this trigger channel.

@ nohum and @TheNON75 - Please let me know if that sounds ok and I'll take a look at this in the coming days.

---

For my reference...

* To wake this damn thing up, hold the button down for a few seconds!!!

```
Application Version   33
Date Code             20190710
Generic Device Class  0
Generic Device Type   0
HW Version            1
Manufacturer Name     IKEA of Sweden
Model Identifier      SYMFONISK Sound Controller
Power Source          3
Product Code          ByteArray [value=45 31 37 34 34]
Product URL           
SW Build ID           2.1.024
Stack Version         98
ZCL Version           3
```

nohum, 2020-09-10:
Do you mean that you would another channel that just sends trigger events? If so, that would be nice (e.g. a channel trigger event for each click type).

Would it also be possible to have channel trigger events for rotating/dimming the device? 
I guess this would be useful for the usecase of @TheNON75 (and another one of mine actually) where you could then receive events even if you rotate the device out of the 0-100% bounds.

cdjackson, 2020-09-10:
> Do you mean that you would another channel that just sends trigger events? If so, that would be nice (e.g. a channel trigger event for each click type).

Yes, that's what I'm thinking about.

> Would it also be possible to have channel trigger events for rotating/dimming the device?

I think this might also be possible.