openhab/org.openhab.binding.zigbee #601: Xiaomi door window sensor not working - on off channel consolidation issue

mattibal, 2020-08-24:
Hi all,

I have a Xiaomi door window sensor MCCGQ01LM (the rounded non-Aqara ones) that's not working with the Zigbee binding (with both the stable 2.5.7 and snapshot version).
The thing is correctly recognized and a "Level Control" channel shows up, but when you link an item to that channel, it doesn't get any event and doesn't change status when you open or close the door.

I looked at what's happening with a debugger and I think there is an issue in the Zigbee binding code.
The sensor seems to support On Off and Level Control as client clusters, but only On Off as server cluster. The Zigbee binding creates anyway just a Level Control channel (and not also an On Off channel) due to "consolidation", without taking into account that the level control cluster is supported by the sensor only as client and not as server.
So a ```ZigBeeConverterSwitchLevel``` is getting created, and it does initializes the device as "client" instead of as "server" because it gets confused by the fact that the level control server cluster is missing, so it doesn't register an AttributeListener for the On Off cluster and doesn't get On Off notifications.

As a workaround, I tried to disable the On Off - Level Control channels "consolidation" by commenting this line: https://github.com/openhab/org.openhab.binding.zigbee/blob/db37991a066c428835b6851c9a1da348bad41d24/org.openhab.binding.zigbee/src/main/java/org/openhab/binding/zigbee/internal/converter/ZigBeeChannelConverterFactoryImpl.java#L65-L66
and it seems to work fine. 
A "Switch" channel gets created alongside to the "Level Control" one, and when linking an item to the Switch channel, it does receive events and changes status normally when the door is open or closed.

I know that's not a proper fix for the issue, but might it be a good enough temporary solution? 
I've tested it with my IKEA Tradfri lights and an Innr SP120 plug, and it doesn't seems to break anything. 

BTW thank you for your work on the binding!

cdjackson, 2020-08-24:
> The sensor seems to support On Off and Level Control as client clusters, but only On Off as server cluster. 

That's incorrect. Both client and server are supported - they have been for a long time as we support wall switches that send OnOff (ie client) and lights that receive OnOff (ie server).

I don't know why a window sensor is a level control device anyway - it should be an IAS device - level doesn't really make sense. However, that's an asside.

I won't disable the channel consolidation if that's what you are suggesting - this goes against the OH design philosophy which is why it was added in the first place.  You are better off considering a static device definition.

mattibal, 2020-08-24:
> That's incorrect. Both client and server are supported - they have been for a long time as we support wall switches that send OnOff (ie client) and lights that receive OnOff (ie server).

Sorry maybe I haven't explained it well. 
I was saying that the Xiaomi sensor I've on hands does support these clusters:
- client clusters: OnOff, Level control
- server clusters: OnOff, but **not** Level control

This is its XML file in ```userdata/zigbee``` directory:

<details>
 <summary>Click to expand</summary>

```xml
<ZigBeeNode>
  <ieeeAddress>00158D000460E3C9</ieeeAddress>
  <networkAddress>42338</networkAddress>
  <nodeDescriptor>
    <apsFlags>0</apsFlags>
    <bufferSize>127</bufferSize>
    <complexDescriptorAvailable>false</complexDescriptorAvailable>
    <manufacturerCode>4151</manufacturerCode>
    <logicalType>END_DEVICE</logicalType>
    <serverCapabilities class="sorted-set"/>
    <incomingTransferSize>100</incomingTransferSize>
    <outgoingTransferSize>100</outgoingTransferSize>
    <userDescriptorAvailable>false</userDescriptorAvailable>
    <frequencyBands>
      <FrequencyBandType>FREQ_2400_MHZ</FrequencyBandType>
    </frequencyBands>
    <macCapabilities class="sorted-set">
      <MacCapabilitiesType>REDUCED_FUNCTION_DEVICE</MacCapabilitiesType>
    </macCapabilities>
    <extendedEndpointListAvailable>false</extendedEndpointListAvailable>
    <extendedSimpleDescriptorListAvailable>false</extendedSimpleDescriptorListAvailable>
    <stackCompliance>0</stackCompliance>
  </nodeDescriptor>
  <powerDescriptor>
    <currentPowerMode>RECEIVER_ON_IDLE</currentPowerMode>
    <availablePowerSources>
      <PowerSourceType>DISPOSABLE_BATTERY</PowerSourceType>
    </availablePowerSources>
    <currentPowerSource>DISPOSABLE_BATTERY</currentPowerSource>
    <powerLevel>FULL</powerLevel>
  </powerDescriptor>
  <endpoints>
    <ZigBeeEndpoint>
      <endpointId>1</endpointId>
      <profileId>260</profileId>
      <deviceId>260</deviceId>
      <deviceVersion>1</deviceVersion>
      <inputClusterIds/>
      <outputClusterIds/>
      <inputClusters>
        <ZclCluster>
          <clusterId>0</clusterId>
          <isClient>false</isClient>
          <attributes>
            <entry>
              <int>0</int>
              <ZclAttribute>
                <id>0</id>
                <name>ZCL Version</name>
                <dataType>UNSIGNED_8_BIT_INTEGER</dataType>
                <mandatory>true</mandatory>
                <implemented>false</implemented>
                <readable>true</readable>
                <writable>false</writable>
                <reportable>false</reportable>
                <minimumReportingPeriod>0</minimumReportingPeriod>
                <maximumReportingPeriod>0</maximumReportingPeriod>
[... 696 lines cut ...]
          </supportedCommandsReceived>
          <supportedCommandsGenerated class="java.util.Collections$UnmodifiableSet">
            <c class="sorted-set"/>
          </supportedCommandsGenerated>
        </ZclCluster>
      </outputClusters>
    </ZigBeeEndpoint>
  </endpoints>
  <bindingTable/>
</ZigBeeNode>
```
</details>

The sensor that I have seems to be available from italian resellers since 2019, so it could be different from the one sold in China since several years ago, I don't know.

> I don't know why a window sensor is a level control device anyway - it should be an IAS device - level doesn't really make sense. However, that's an asside.

I think the clusters configuration of that Xiaomi device (even if it's not the smartest) is allowed by Zigbee specs, so the Openhab Zigbee binding should correctly get attribute update reports from the OnOff server cluster of the device.
Unfortunately that's not working for the reason that I've explained in my previous message. 
I think that's a bug of the Zigbee binding.

> I won't disable the channel consolidation if that's what you are suggesting - this goes against the OH design philosophy which is why it was added in the first place. You are better off considering a static device definition.

Thank you for the suggestion, I didn't know the static device definition feature existed, I'll have a look at it.
Could it be used to "disable" the LevelControl client cluster of the device, so that the OnOff server cluster is picked by ```ZigBeeConverterSwitchOnoff``` ? This would be a better workaround for the issue.