openhab/org.openhab.binding.zigbee #603: Tradfri bulb not going OFFLINE

weakfl, 2020-08-29:
As discussed [here](https://community.openhab.org/t/zigbee-devices-going-offline/101249/42).

@cdjackson I've set the polling period to 60 seconds to hopefully speed things up a bit. 15 minutes after the `LastChance Timeout` has expired the device is still online and being polled:

```
2020-08-29 13:30:10.770 [DEBUG] [.zigbee.handler.ZigBeeIsAliveTracker] - IsAlive Tracker LastChance Timeout has been reached for thingUID=zigbee:device:ba8eb59b:000b57fffeb1bab5
[...]
2020-08-29 13:45:44.976 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 000B57FFFEB1BAB5: Polling [zigbee:device:ba8eb59b:000b57fffeb1bab5:000B57FFFEB1BAB5_1_dimmer, zigbee:device:ba8eb59b:000b57fffeb1bab5:000B57FFFEB1BAB5_1_colortemperature] channels...
2020-08-29 13:45:44.978 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 000B57FFFEB1BAB5: Polling zigbee:device:ba8eb59b:000b57fffeb1bab5:000B57FFFEB1BAB5_1_dimmer
2020-08-29 13:46:09.018 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 000B57FFFEB1BAB5: Polling zigbee:device:ba8eb59b:000b57fffeb1bab5:000B57FFFEB1BAB5_1_colortemperature
```

And the debug log: 
[openhab.log.zip](https://github.com/openhab/org.openhab.binding.zigbee/files/5145176/openhab.log.zip)

cdjackson, 2020-08-29:
Thanks - I'll take a look.

Note that polling period will not impact this since the devices will be using reporting...

weakfl, 2020-08-29:
> Note that polling period will not impact this since the devices will be using reporting...

Noted :)
And thanks for looking into this.

cdjackson, 2020-08-29:
I have hopefully found the issue. The last chance timer is meant to be set to 30 seconds - the timer was being initialised in seconds, but I'm used to defining things in milliseconds, so it was 30000 seconds. If you can wait 8 hours, it should work (hopefully!) :) 

I'll create a fix for this later today.

weakfl, 2020-08-29:
> If you can wait 8 hours, it should work (hopefully!) :)

😂 

Perfect, thanks 👍

PR #604 (2020-08-29): Update last chance timer period
Closes #603
closed 2020-08-29 by cdjackson