openhab/org.openhab.binding.zigbee #605: Support for CC2562 coordinator

filewalker1, 2020-09-05:
Since you support the older TI CC2531 with Zigbee Firmware i tried to use the newer one but without success. Log says: "CC2531 seems to be in bootloader mode". Is it possible to support this newer Stick in the near future?

cdjackson, 2020-09-06:
I'm not personally familiar with this device, so probably will not add support for it myself. If someone wants to do this, then can add it to the underlying ZigBee libraries and I'd suggest to start an issue in that repository.

https://github.com/zsmartsystems/com.zsmartsystems.zigbee

filewalker1, 2020-09-06:
Thank you ver your reply! I am nut quite sure why this is not wokring, since the firmware is the same for both... I am quite new to openHAB, so i need to learn how everything fits together. If i get deeper and get to know the structure for this addon/binding i can possibly add support and create a pull request.

Hedda, 2021-05-12:
@filewalker1 Please see https://github.com/openhab/org.openhab.binding.zigbee/issues/649 and more importantly please see discussion in ->  https://github.com/zsmartsystems/com.zsmartsystems.zigbee/issues/1226

Hedda, 2021-12-01:
Those interested in Z-Stack 3 support to use CC2652 and CC1352 with openHAB's Zigbee Binding should check out:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/1280

and

https://community.openhab.org/t/discussion-about-ti-z-stack-3-x-and-cc2652-and-cc1352-zigbee-coordinator-adapters-with-openhab-zigbee-binding/129385

That branch is currently only for developers however it could also soon become ready for testing by advanced users.

Hedda, 2022-09-28:
FYI; @leonschenk posted now in OpenHAB community forum here https://community.openhab.org/t/discussion-about-texas-instruments-z-stack-3-0-and-cc2652-and-cc1352-zigbee-coordinator-adapters-with-openhab-zigbee-binding/129385/4 that he now picked up the development where @t-8ch left it, forking his branch of the zstack driver with Z-Stack 3 support to https://github.com/leonschenk/com.zsmartsystems.zigbee/tree/zstack/initial in order to continue the work of adding initial support for CC2652/CC1352 based adapters with Z-Stack 3.x.0 Zigbee Coordinator firmware https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin so hopefully once he feels it works good enough for review he will create a new pull request to the com.zsmartsystems.zigbee ZigBee Cluster Library Java framework that OpenHAB's ZigBee Binding and other projects depend on https://github.com/zsmartsystems/com.zsmartsystems.zigbee