openhab/org.openhab.binding.zigbee #608: Power Meter channels don't show up

mjhas, 2020-09-16:
### Outline

I recently bought a [Lupusec Smart Plug](https://www.lupus-electronics.de/shop/Alarm-Smarthome/Smarthome-Hausautomation/LUPUSEC-Funksteckdose-mit-Stromz%E4hler-und-ZigBee-Repeater-p.html), which is rebranded version of 
[Climax Device](https://www.climax.com.tw/new/psmd29zb.php) (see also [Reference in Zigbee2Mqtt](https://www.zigbee2mqtt.io/devices/PSM-29ZBSR.html)) while discovery works and also the on/off channel is found, the power meter readings are not available. Are smart meters generally not yet supported, as indicated by discussions around other products? 

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used |             Ember EM35x Coordinator                                                                   |
| openHAB version  | 2.5.8     |
| Hardware         |                      Raspberry Pi 4                                                          |
| Memory           |                                           4GB                                     |
| Java version     |                                                11                                |
| Devices          |  zigbee:coordinator_ember:0118E7EC (Type=Bridge, Status=ONLINE, Label=BV 2010/10 (ZigBee USB dongle), Bridge=null) zigbee:device:0118E7EC:00124b001c0f3318 (Type=Thing, Status=ONLINE, Label=powermeter2, Bridge=zigbee:coordinator_ember:0118E7EC)     |

### Logs

```
openhab>smarthome:zigbee node 00124B001C0F3318
IEEE Address     : 00124B001C0F3318
Network Address  : 48161
Node Descriptor  : NodeDescriptor [apsFlags=0, bufferSize=80, complexDescriptorAvailable=false, manufacturerCode=0000, logicalType=ROUTER, serverCapabilities=[], incomingTransferSize=160, outgoingTransferSize=160, userDescriptorAvailable=false, frequencyBands=[FREQ_2400_MHZ], macCapabilities=[FULL_FUNCTION_DEVICE, MAINS_POWER, RECEIVER_ON_WHEN_IDLE], extendedEndpointListAvailable=false, extendedSimpleDescriptorListAvailable=false, stackCompliance=0]
Power Descriptor : PowerDescriptor [currentPowerMode=RECEIVER_ON_IDLE, availablePowerSources=[MAINS], currentPowerSource=MAINS, powerLevel=FULL]
Associations     : []
Endpoints        : 
            1    : Profile     0104 ZIGBEE_HOME_AUTOMATION
                 : Device Type 0051 SMART_PLUG
                   -> BASIC
                   -> IDENTIFY
                   -> GROUPS
                   -> SCENES
                   -> ON_OFF
                   -> METERING
                   <- BASIC
            4    : Profile     0104 ZIGBEE_HOME_AUTOMATION
                 : Device Type 0051 SMART_PLUG
                   <- OTA_UPGRADE
Neighbors:
Routes:
openhab> smarthome:things show zigbee:device:0118E7EC:00124b001c0f3318
UID: zigbee:device:0118E7EC:00124b001c0f3318
Type: zigbee:device
Label: powermeter2
Status: ONLINE
Bridge: zigbee:coordinator_ember:0118E7EC

Properties:
	zigbee_logicaltype : ROUTER
	zigbee_powerlevel : FULL
	zigbee_manufacturercode : 0x0000
	modelId : PSMP5_00.00.05.10TC
	zigbee_networkaddress : 48161
	zigbee_powersource : MAINS
	zigbee_datecode : 20180801
	zigbee_zclversion : 1
	vendor : ClimaxTechnology
	zigbee_powermode : RECEIVER_ON_IDLE
	zigbee_powersources : [MAINS]
	hardwareVersion : 0
	firmwareVersion : 0x000001FE
	zigbee_applicationVersion : 3
	zigbee_device_initialised : true

Configuration parameters:
	zigbee_macaddress : 00124B001C0F3318

Channels:
	ID: 00124B001C0F3318_1_switch
	Label: Switch
	Type: zigbee:switch_onoff

```

```
2020-09-16 21:32:29.766 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 00124B001C0F3318: Starting ZigBee device discovery
2020-09-16 21:32:29.773 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - 00124B001C0F3318: Creating ZigBee device zigbee:device with bridge zigbee:coordinator_ember:0118E7EC
2020-09-16 21:32:29.765 [DEBUG] [iscovery.ZigBeeNodeServiceDiscoverer] - 00124B001C0F3318: Node SVC Discovery: start discovery
2020-09-16 21:32:29.782 [INFO ] [g.discovery.internal.PersistentInbox] - Added new thing 'zigbee:device:0118E7EC:00124b001c0f3318' to inbox.
2020-09-16 21:32:29.783 [DEBUG] [iscovery.ZigBeeNodeServiceDiscoverer] - 00124B001C0F3318: Node SVC Discovery: starting new tasks [NWK_ADDRESS, POWER_DESCRIPTOR, ACTIVE_ENDPOINTS, NODE_DESCRIPTOR]
2020-09-16 21:32:29.790 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - 00124B001C0F3318: Node discovery not complete
2020-09-16 21:32:29.794 [DEBUG] [iscovery.ZigBeeNodeServiceDiscoverer] - 00124B001C0F3318: Node SVC Discovery: scheduled [NWK_ADDRESS, NODE_DESCRIPTOR, ACTIVE_ENDPOINTS, POWER_DESCRIPTOR]
2020-09-16 21:32:29.823 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=6, ackNum=6, reTx=false, data=05 90 59 AB 9F 6A 36 0A 0A AA 3E B0 7C FF B1 00]
2020-09-16 21:32:29.831 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingRouteRecordHandler [networkId=0, source=9FAB, sourceEui=7CB03EAA0A0A366A, lastHopLqi=255, lastHopRssi=-79, relayList=]
2020-09-16 21:32:29.837 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=7, notRdy=false]
2020-09-16 21:32:29.878 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=7, ackNum=6, reTx=false, data=05 90 24 21 BC 18 33 0F 1C 00 4B 12 00 01 00 AB 9F]
2020-09-16 21:32:29.882 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspTrustCenterJoinHandler [networkId=0, newNodeId=BC21, newNodeEui64=00124B001C0F3318, status=EMBER_STANDARD_SECURITY_UNSECURED_JOIN, policyDecision=EMBER_USE_PRECONFIGURED_KEY, parentOfNewNodeId=9FAB]
2020-09-16 21:32:29.890 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - 00124B001C0F3318: nodeStatusUpdate - node status is UNSECURED_JOIN, network address is BC21.
2020-09-16 21:32:29.896 [DEBUG] [pp.discovery.ZigBeeNetworkDiscoverer] - 00124B001C0F3318: Device status updated. NWK=BC21
2020-09-16 21:32:29.898 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=0, notRdy=false]
2020-09-16 21:32:29.901 [DEBUG] [pp.discovery.ZigBeeNetworkDiscoverer] - 00124B001C0F3318: NWK Discovery add node BC21
2020-09-16 21:32:29.904 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeNode ] - 00124B001C0F3318: Node state updated from UNKNOWN to ONLINE
2020-09-16 21:32:29.907 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - 00124B001C0F3318: Updating node NWK=BC21
2020-09-16 21:32:29.910 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - 00124B001C0F3318: Node update. NWK Address=BC21
2020-09-16 21:32:29.913 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - 00124B001C0F3318: Node BC21 is not updated
2020-09-16 21:32:30.007 [DEBUG] [atabase.ZigBeeNetworkDatabaseManager] - 00124B001C0F3318: Data store: Writing node.
2020-09-16 21:32:30.022 [DEBUG] [ding.zigbee.internal.ZigBeeDataStore] - 00124B001C0F3318: ZigBee saving network state complete.
2020-09-16 21:32:30.092 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue size: 1
2020-09-16 21:32:30.094 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=6, ackNum=0, reTx=false, data=06 00 18]
2020-09-16 21:32:30.099 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=0, ackNum=6, reTx=false, data=05 90 45 04 00 00 13 00 00 00 00 00 00 00 00 FF C8 21 BC FF FF 0C 00 21 BC 18 33 0F 1C 00 4B 12 00 8E 04]
2020-09-16 21:32:30.103 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingMessageHandler [networkId=0, type=EMBER_INCOMING_BROADCAST, apsFrame=EmberApsFrame [profileId=0000, clusterId=0013, sourceEndpoint=0, destinationEndpoint=0, options=[], groupId=0, sequence=00], lastHopLqi=255, lastHopRssi=-56, sender=BC21, bindingIndex=255, addressIndex=255, messageContents=00 21 BC 18 33 0F 1C 00 4B 12 00 8E]
2020-09-16 21:32:30.106 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=BC21/0, destinationAddress=0000/0, profile=0000, cluster=0013, addressMode=DEVICE, radius=0, apsSecurity=false, ackRequest=false, apsCounter=00, rssi=-56, lqi=FF, payload=00 21 BC 18 33 0F 1C 00 4B 12 00 8E]

cdjackson, 2020-10-03:
I would need to see logs during a complete reinitialisation of the device to see why it's not showing so that I can determine what features this device supports.

cdjackson, 2021-05-12:
Closing this as the logs aren't available. If you can provide the logs please reopen and I will take a look.

closed 2021-05-12 by cdjackson