openhab/org.openhab.binding.zigbee #609: rfc2217 support (cc2531, others ?)

gerwsx, 2020-10-01:
The serial port stays connected using rfc2217 after suppressing “Unsupported comm operation” exception, but initialization of cc2531 is failing.

https://community.openhab.org/t/solved-zwave-zigbee-rfc2217-remote-serial-port-howto/81633/89

cdjackson, 2020-10-17:
In general RFC2217 support should be handled outside the binding - that was the reason to add the ESH serial abstraction layer. If you have a PR that can solve any niggles in the binding with the way the serial abstraction is interfaced, then I'm happy to look at that, but at the moment this issue isn't really something that I think is a binding issue.

gerwsx, 2020-10-23:
It would be an improvement, I think, to include the small patch I posted in the thread on the openhab forum (catch the UnsupportedCommOperationException). Interesting to see if other zigbee sticks also have initialization problems.

cdjackson, 2020-10-23:
If you want to propose an PR, then please do so.

gerwsx, 2020-10-23:
Thanks for the quick reply. I don't have the code anymore since I switched to Conbee2 (deConz). It's a really small change, feel free to include with another change.

closed 2021-11-10 by cdjackson