openhab/org.openhab.binding.zigbee #614: NLG-remote channels for COLOR_SCENE_CONTROLLER missing after discovery

mnasec, 2020-10-15:
OpenHab Version: 2.5.9
Bug?: Adding thing "Neuhaus Lighting Group Q Remote Controller" discovers only one channel (LEVEL_CONTROL)
Other channels/clusters are missing, so color oder scenes are not usable

Controller:

> BV 2010/10 (ZigBee USB dongle)
> Ember EM35x Coordinator
> Ember EM35x NCP
> 

Output Karaf:
```
openhab> zigbee nodes
Total known nodes in network: 3
Network  Addr  IEEE Address      Logical Type  State      EP   Profile                    Device Type                Manufacturer     Model          
      0  0000  000D6F000DD029EB  COORDINATOR   UNKNOWN  
  37639  9307  7CB03EAA0A090992  ROUTER        ONLINE      3  ZIGBEE_LIGHT_LINK          DATA_COLLECTION_UNIT       OSRAM            Plug 01        
  62606  F48E  00158D00044D7BEF  END_DEVICE    ONLINE      1  ZIGBEE_HOME_AUTOMATION     COLOR_SCENE_CONTROLLER     Neuhaus Lighting Group  NLG-remote 
```

```
openhab> zigbee node 62606
IEEE Address     : 00158D00044D7BEF
Network Address  : 62606
Node Descriptor  : NodeDescriptor [apsFlags=0, bufferSize=127, complexDescriptorAvailable=false, manufacturerCode=1037, logicalType=END_DEVICE, serverCapabilities=[], incomingTransferSize=100, outgoingTransferSize=100, userDescriptorAvailable=false, frequencyBands=[FREQ_2400_MHZ], macCapabilities=[REDUCED_FUNCTION_DEVICE], extendedEndpointListAvailable=false, extendedSimpleDescriptorListAvailable=false, stackCompliance=22]
Power Descriptor : PowerDescriptor [currentPowerMode=RECEIVER_ON_IDLE, availablePowerSources=[DISPOSABLE_BATTERY], currentPowerSource=DISPOSABLE_BATTERY, powerLevel=FULL]
Associations     : []
Endpoints        : 
            1    : Profile     0104 ZIGBEE_HOME_AUTOMATION
                 : Device Type 0810 COLOR_SCENE_CONTROLLER
                   -> BASIC
                   -> IDENTIFY
                   -> 0x1000
                   <- IDENTIFY
                   <- GROUPS
                   <- SCENES
                   <- ON_OFF
                   <- LEVEL_CONTROL
                   <- COLOR_CONTROL
                   <- 0x1000
Neighbors:
Routes:

```
```
openhab> zigbee bindtable 62606
Binding table for node 62606 [00158D00044D7BEF]
Src Address          | Dest Address         | Group | Mode    | Cluster
00158D00044D7BEF/1   | 000D6F000DD029EB/1   |       | Address | 0008:LEVEL_CONTROL
00158D00044D7BEF/1   | 000D6F000DD029EB/1   |       | Address | 0006:ON_OFF

```

In Paper UI only the channel LEVEL_CONTROL is available.

Switch ON works:
2020-10-16 01:27:20.287 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingMessageHandler [networkId=0, type=EMBER_INCOMING_BROADCAST, apsFrame=EmberApsFrame [profileId=0104, clusterId=0006, sourceEndpoint=1, destinationEndpoint=255, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY], groupId=0, sequence=68], lastHopLqi=255, lastHopRssi=-57, sender=F48E, bindingIndex=255, addressIndex=255, messageContents=11 32 01]
2020-10-16 01:27:20.288 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=F48E/1, destinationAddress=0000/255, profile=0104, cluster=0006, addressMode=DEVICE, radius=0, apsSecurity=false, ackRequest=false, apsCounter=68, rssi=-57, lqi=FF, payload=11 32 01]
2020-10-16 01:27:20.289 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=CLUSTER_SPECIFIC_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=true, manufacturerCode=0, sequenceNumber=50, commandId=1]
2020-10-16 01:27:20.291 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: OnCommand [On/Off: F48E/1 -> 0000/255, cluster=0006, TID=32]
2020-10-16 01:27:20.292 [DEBUG] [transaction.ZigBeeTransactionManager] - notifyTransactionCommand: OnCommand [On/Off: F48E/1 -> 0000/255, cluster=0006, TID=32] 
2020-10-16 01:27:20.295 [DEBUG] [converter.ZigBeeConverterSwitchLevel] - 00158D00044D7BEF: ZigBee command received OnCommand [On/Off: F48E/1 -> 0000/255, cluster=0006, TID=32]
2020-10-16 01:27:20.296 [DEBUG] [converter.ZigBeeBaseChannelConverter] - 00158D00044D7BEF: Channel zigbee:device:01381159:00158d00044d7bef:00158D00044D7BEF_1_dimmer updated to 100
2020-10-16 01:27:20.297 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00158D00044D7BEF: Updating ZigBee channel state zigbee:device:01381159:00158d00044d7bef:00158D00044D7BEF_1_dimmer to 100
2020-10-16 01:27:20.298 [DEBUG] [.zigbee.handler.ZigBeeIsAliveTracker] - IsAlive Tracker reset for handler with thingUID=zigbee:device:01381159:00158d00044d7bef
2020-10-16 01:27:20.299 [DEBUG] [.zigbee.handler.ZigBeeIsAliveTracker] - IsAlive Tracker cancelled task for thingUID=zigbee:device:01381159:00158d00044d7bef
2020-10-16 01:27:20.300 [DEBUG] [.zigbee.handler.ZigBeeIsAliveTracker] - IsAlive Tracker scheduled task for thingUID=zigbee:device:01381159:00158d00044d7bef in 28 seconds
``
also for Switch OFF, but Color Wheel does not work:
[... 21 lines cut ...]
2020-10-16 00:08:08.455 [DEBUG] [p.discovery.ZigBeeDiscoveryExtension] - 00158D00044D7BEF: DISCOVERY Extension: Adding discoverer for node
2020-10-16 00:08:08.455 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00158D00044D7BEF: Node updated - ZigBeeNode [state=ONLINE, IEEE=00158D00044D7BEF, NWK=F48E, endpoints=[]]
2020-10-16 00:08:08.456 [DEBUG] [iscovery.ZigBeeNodeServiceDiscoverer] - 00158D00044D7BEF: Node SVC Discovery: created discoverer
2020-10-16 00:08:08.456 [WARN ] [zsmartsystems.zigbee.ZigBeeExecutors] - Uncaught exception in thread NotificationService-thread-503
java.lang.NullPointerException: null
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.nodeUpdated(ZigBeeThingHandler.java:872) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.nodeAdded(ZigBeeThingHandler.java:815) ~[?:?]
        at com.zsmartsystems.zigbee.ZigBeeNetworkManager$12.run(ZigBeeNetworkManager.java:1652) ~[?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149) ~[?:1.8.0_265]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624) ~[?:1.8.0_265]
        at java.lang.Thread.run(Thread.java:748) [?:1.8.0_265]
2020-10-16 00:08:08.456 [DEBUG] [iscovery.ZigBeeNodeServiceDiscoverer] - 00158D00044D7BEF: Node SVC Discovery: start discovery
2020-10-16 00:08:08.459 [DEBUG] [iscovery.ZigBeeNodeServiceDiscoverer] - 00158D00044D7BEF: Node SVC Discovery: starting new tasks [NODE_DESCRIPTOR, NWK_ADDRESS, POWER_DESCRIPTOR, ACTIVE_ENDPOINTS]
2020-10-16 00:08:08.460 [DEBUG] [iscovery.ZigBeeNodeServiceDiscoverer] - 00158D00044D7BEF: Node SVC Discovery: scheduled [NWK_ADDRESS, NODE_DESCRIPTOR, POWER_DESCRIPTOR, ACTIVE_ENDPOINTS]
2020-10-16 00:08:08.465 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 00158D00044D7BEF: Starting ZigBee device discovery
2020-10-16 00:08:08.466 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - 00158D00044D7BEF: Creating ZigBee device zigbee:device with bridge zigbee:coordinator_ember:01381159
2020-10-16 00:08:08.467 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - 00158D00044D7BEF: Node discovery not complete
2020-10-16 00:08:08.706 [DEBUG] [atabase.ZigBeeNetworkDatabaseManager] - 00158D00044D7BEF: Data store: Writing node.
2020-10-16 00:08:08.717 [DEBUG] [ding.zigbee.internal.ZigBeeDataStore] - 00158D00044D7BEF: ZigBee saving network state complete.
2020-10-16 00:08:09.181 [DEBUG] [iscovery.ZigBeeNodeServiceDiscoverer] - 00158D00044D7BEF: Node SVC Discovery: running NWK_ADDRESS
2020-10-16 00:08:09.183 [DEBUG] [transaction.ZigBeeTransactionManager] - 00158D00044D7BEF: Creating new Transaction Queue
2020-10-16 00:08:09.185 [DEBUG] [e.transaction.ZigBeeTransactionQueue] - 00158D00044D7BEF: Set profile to ZigBeeTransactionProfile [maxOutstandingTransactions=1, interTransactionDelay=50, maxRetries=2]
2020-10-16 00:08:09.187 [DEBUG] [e.transaction.ZigBeeTransactionQueue] - 00158D00044D7BEF: Added transaction to queue, len=1, transaction=ZigBeeTransaction [ieeeAddress=00158D00044D7BEF queueTime=0, state=WAITING, sendCnt=0, command=NetworkAddressRequest [0000/0 -> F48E/0, cluster=0000, TID=--, ieeeAddr=00158D00044D7BEF, requestType=0, startIndex=0]]

```

cdjackson, 2020-10-16:
Currently these features are not supported in the binding. I think only on/off and level control clients are supported - not color control.

mnasec, 2020-10-16:
Hi Chris,
from the documentation about the binding, at least Cluster 0x300 (Color Control) should work, for the other clusters you are right, e.g. Scene is not mentioned.

cdjackson, 2020-10-16:
Yes, the binding supports CONTROLLING colour devices - this is not what you are doing. You are using a controller as I understand it?

mnasec, 2020-10-16:
Hi Chris,
in my scenario i have a openhab instance with a usb zigbee controller and i am able to control these devices by paper ui or openhab app. Additional i have a Zigbee Remote control (https://zigbeealliance.org/de/zigbee_products/Funkfernbedienung-rgbcct-e0040044/), but after discovering this EndDevice in paper ui, i only see one channel (LEVEL CONTROL) for cluster 0x8. I got not a channel for cluster 0x6 (ON_OFF) and not a channel for 0x300 (COLOR CONTROL). I this this is a bug in discovering this special device.

Manual modifying the bindtable of this endpoint does not help.

So now, pressing Level Control at the remote device works for cluster 0x8, but the color wheel for cluster 0x300 not, because the channel is missing in openhab.

Hope the situation is a little bit undestandable.

Best regards
Marc

cdjackson, 2020-10-16:
Hi Marc,
It's not a bug - as per my initial response - it's unsupported (if I understand this device correctly).

This is a remote control - right? In ZigBee, clusters can be servers (eg lights), or they can be clients (eg light switches, remote controls). I believe that your device is the latter - a remote control, supporting the client side.  If so, this is not supported at this time. As per my original answer - only the on/off and level control clusters are currently supported for the client.

If this is a colour light, then you can control it with the binding. Since it's a controller, the binding doesn't accept commands from the device.

I hope that makes sense. Maybe I still don't understand the device?

mnasec, 2020-10-17:
Hi Chris,
thank you for your response.
I think 'remote control' is correct and Zigbee client device is correct also, normally the lamps are linked directly with the remote control without OH. In case that cluster 0x300 is not supported by the binding, should i have not at least two channels in OH, one for LEVEL_CONTROL and one for ON_OFF?? I just got only one channel for LEVEL_CONTROL, so i think i am missing one channel from type SWITCH for cluster 0x6. Possible the discovery phase stops due to the java thread dump.

May it help to manual modify the bindtable for the endpoint by 'zigbee bind' to link cluster 0x300 with the light devices directly, till OH supports this?
May be it is a good feature request to extend the binding to support cluster 0x300 for remote controls to operate the lights remote?

Best regards 
Marc

cdjackson, 2020-10-17:
> should i have not at least two channels in OH, one for LEVEL_CONTROL and one for ON_OFF??

No - you will just have one channel - this is normal for OH since there should be no need for both channels.

> May it help to manual modify the bindtable for the endpoint by 'zigbee bind' to link cluster 0x300 with the light devices directly, till OH supports this?

Sure - you can do this if you want to control lights directly.

> May be it is a good feature request to extend the binding to support cluster 0x300 for remote controls to operate the lights remote?

That's fine - I don't disagree that it should be added, but unless someone has the time, it won't happen :). At some stage in future, maybe I'll look at it, but it won't be any time soon as I've too much else on at he moment and it's clearly not a common issue.

cdjackson, 2021-11-12:
Closed by #696

closed 2021-11-12 by cdjackson