openhab/org.openhab.binding.zigbee #615: Issue adding Sunricher CCT Lighting

0xE1, 2020-11-15:
### Outline

Issue adding Sunricher CCT Lighting (2 channel LED, WW and CW), there an exception in logs related to it:
com.zsmartsystems.zigbee.dongle.cc2531.network.packet.ZToolParseException: Packet checksum failed

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used | TI CC2531EMK |
| openHAB version  | 2.5.0 (installed via packet manager) |
| Hardware         | Raspberry Pi 3 B+ |
| Memory           | 1GB |
| Java version     | 1.8.0_152-b76 |
| Devices          | 2x OSRAM Plug 01 (7CB03EAA00B0C79E, 7CB03EAA0A0A4F6B), 1x Sunricher CCT Lighting (000D6F00137473A9) |

### Logs
events.log
```2020-11-15 21:04:59.550 [home.event.InboxAddedEvent] - Discovery Result with UID 'zigbee:device:b72bea66:000d6f00137473a9' has been added.
2020-11-15 21:06:31.132 [hingStatusInfoChangedEvent] - 'zigbee:device:b72bea66:000d6f00137473a9' changed from UNINITIALIZED to INITIALIZING
2020-11-15 21:06:31.158 [hingStatusInfoChangedEvent] - 'zigbee:device:b72bea66:000d6f00137473a9' changed from INITIALIZING to UNKNOWN
2020-11-15 21:06:31.278 [me.event.ThingUpdatedEvent] - Thing 'zigbee:device:b72bea66:000d6f00137473a9' has been updated.
2020-11-15 21:06:31.294 [nt.FirmwareStatusInfoEvent] - Firmware status of thing zigbee:device:b72bea66:000d6f00137473a9 changed to UNKNOWN.
2020-11-15 21:06:31.313 [me.event.ThingUpdatedEvent] - Thing 'zigbee:device:b72bea66:000d6f00137473a9' has been updated.
2020-11-15 21:08:19.380 [hingStatusInfoChangedEvent] - 'zigbee:device:b72bea66:000d6f00137473a9' changed from UNKNOWN to ONLINE
2020-11-15 21:08:34.544 [.ItemChannelLinkAddedEvent] - Link 'SunricherCCTLighting_ColorTemperature-zigbee:device:b72bea66:000d6f00137473a9:000D6F00137473A9_1_colortemperature' has been added.
2020-11-15 21:08:43.609 [.ItemChannelLinkAddedEvent] - Link 'SunricherCCTLighting_LevelControl-zigbee:device:b72bea66:000d6f00137473a9:000D6F00137473A9_1_dimmer' has been added.
```

openhab.log
```2020-11-15 21:06:01.803 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - 7CB03EAA00B0C79E: Discovery: Starting discovery for existing device
2020-11-15 21:06:01.807 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - 000D6F00137473A9: Discovery: Starting discovery for existing device
2020-11-15 21:06:01.810 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - 7CB03EAA0A0A4F6B: Discovery: Starting discovery for existing device
2020-11-15 21:06:01.817 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 7CB03EAA00B0C79E: Starting ZigBee device discovery
2020-11-15 21:06:01.821 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - 7CB03EAA00B0C79E: Creating ZigBee device zigbee:device with bridge zigbee:coordinator_cc2531:b72bea66
2020-11-15 21:06:01.821 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 000D6F00137473A9: Starting ZigBee device discovery
2020-11-15 21:06:01.816 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - Starting ZigBee scan for zigbee:coordinator_cc2531:b72bea66
2020-11-15 21:06:01.825 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - 000D6F00137473A9: Creating ZigBee device zigbee:device with bridge zigbee:coordinator_cc2531:b72bea66
2020-11-15 21:06:01.826 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 7CB03EAA0A0A4F6B: Starting ZigBee device discovery
2020-11-15 21:06:01.828 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - Permit join to 65532/0 for 60 seconds.
2020-11-15 21:06:01.831 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - 7CB03EAA0A0A4F6B: Creating ZigBee device zigbee:device with bridge zigbee:coordinator_cc2531:b72bea66
2020-11-15 21:06:01.831 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 7CB03EAA00B0C79E: ZigBee node property discovery start
2020-11-15 21:06:01.835 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ManagementPermitJoiningRequest [0/0 -> 65532/0, cluster=0036, TID=F4, permitDuration=60, tcSignificance=true]
2020-11-15 21:06:01.837 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 7CB03EAA00B0C79E: ZigBee node property discovery using basic cluster on endpoint 32474/3
2020-11-15 21:06:01.840 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 7CB03EAA0A0A4F6B: ZigBee node property discovery start
2020-11-15 21:06:01.840 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=65532/0, profile=0000, cluster=0036, addressMode=DEVICE, radius=31, apsSecurity=false, apsCounter=7D, payload=00 3C 01]
2020-11-15 21:06:01.847 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 000D6F00137473A9: ZigBee node property discovery start
2020-11-15 21:06:01.852 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 000D6F00137473A9: ZigBee node property discovery using basic cluster on endpoint 29117/1
2020-11-15 21:06:01.852 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 7CB03EAA0A0A4F6B: ZigBee node property discovery using basic cluster on endpoint 5131/3
2020-11-15 21:06:01.869 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ManagementPermitJoiningRequest [0/0 -> 0/0, cluster=0036, TID=F3, permitDuration=60, tcSignificance=true]
2020-11-15 21:06:01.876 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=0/0, profile=0000, cluster=0036, addressMode=DEVICE, radius=31, apsSecurity=false, apsCounter=7E, payload=00 3C 01]
2020-11-15 21:06:01.902 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ReadAttributesCommand [Basic: 0/0 -> 5131/3, cluster=0000, TID=F5, identifiers=[5, 4, 3, 2, 6, 0, 1]]
2020-11-15 21:06:01.905 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=245, commandId=0]
2020-11-15 21:06:01.910 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/1, destinationAddress=5131/3, profile=0104, cluster=0000, addressMode=DEVICE, radius=31, apsSecurity=false, apsCounter=7F, payload=00 F5 00 05 00 04 00 03 00 02 00 06 00 00 00 01 00]
2020-11-15 21:06:01.925 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=0/0, profile=0000, cluster=8036, addressMode=null, radius=0, apsSecurity=false, apsCounter=--, payload=00 00]
2020-11-15 21:06:01.927 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ManagementPermitJoiningResponse [0/0 -> 0/0, cluster=8036, TID=--, status=SUCCESS]
2020-11-15 21:06:01.951 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ManagementPermitJoiningRequest [0/0 -> 0/0, cluster=0036, TID=F6, permitDuration=60, tcSignificance=true]
2020-11-15 21:06:01.953 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=0/0, profile=0000, cluster=0036, addressMode=DEVICE, radius=31, apsSecurity=false, apsCounter=80, payload=00 3C 01]
2020-11-15 21:06:01.983 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ReadAttributesCommand [Basic: 0/0 -> 32474/3, cluster=0000, TID=F0, identifiers=[5, 4, 3, 2, 6, 0, 1]]
2020-11-15 21:06:01.984 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=240, commandId=0]
2020-11-15 21:06:01.986 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0/1, destinationAddress=32474/3, profile=0104, cluster=0000, addressMode=DEVICE, radius=31, apsSecurity=false, apsCounter=81, payload=00 F0 00 05 00 04 00 03 00 02 00 06 00 00 00 01 00]
2020-11-15 21:06:02.000 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=0/0, destinationAddress=0/0, profile=0000, cluster=8036, addressMode=null, radius=0, apsSecurity=false, apsCounter=--, payload=00 00]
2020-11-15 21:06:02.002 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ManagementPermitJoiningResponse [0/0 -> 0/0, cluster=8036, TID=--, status=SUCCESS]
2020-11-15 21:06:02.040 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ReadAttributesCommand [Basic: 0/0 -> 32474/3, cluster=0000, TID=F7, identifiers=[5, 4, 3, 2, 6, 0, 1]]
[... cut ...]
2020-11-15 21:08:19.742 [DEBUG] [ding.zigbee.internal.ZigBeeDataStore] - 000D6F00137473A9: ZigBee saving network state complete.
2020-11-15 21:08:34.542 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 000D6F00137473A9: Channel zigbee:device:b72bea66:000d6f00137473a9:000D6F00137473A9_1_colortemperature linked - polling started.
2020-11-15 21:08:43.607 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 000D6F00137473A9: Channel zigbee:device:b72bea66:000d6f00137473a9:000D6F00137473A9_1_dimmer linked - polling started.
```

cdjackson, 2020-11-15:
> Issue adding Sunricher CCT Lighting (2 channel LED, WW and CW)

What is the problem though?

The error you mention is not an issue - it's just a checksum failure. Please describe the problem you are having with the device -or is it really just the checksum error you're reporting?

0xE1, 2020-11-15:
Controller is not initialized, zigbee_device_initialised is always false despite it also showing "Online".
After the controller is added, there's no further communications in either direction from the looks of it,
openhab is unaware of the state of the controller (on/off/dimming, color) and any changes to color or dimming have no affect on controller.

0xE1, 2020-11-16:
Found the [original documentation](https://www.sunricher.com/media/resources/manual/SR-ZG9101FA-CCT%20instruction%20V1.pdf) from Manufacturer instead of a resellers cut version, it had different workflow than describe by reseller - and it actually works now.

Model name SR-ZG9101FA-CCT, Operation 3 is for pairing with ZigBee controller like TI CC2531EMK

closed 2020-11-16 by 0xE1