openhab/org.openhab.binding.zigbee #625: Missing channels for Bitron Home 902010/25

adahmen, 2021-01-01:
### Outline

I was able to add a smartplug from Bitron (Bitron Home 902010/25) to OpenHAB, but I only see the Switch-Channel and not to ActivePower, RmsCurrent or RmsVoltage channels.
I now that the device is able to deliver these values, since I used the devices with another smarthome plattform (from german Telekom - Qivicon) and these values were available.

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used |  zigbee:coordinator_telegesis:0500201F (Qivicon-USB-Stick)                                                                         |
| openHAB version  | 2.5.6     |
| Hardware         |  Raspberry Pi 4                                                                              |
| Memory           |   2 GB                                                                             |
| Java version     |   1.8.0_252                                                                             |
| Devices          |                                   |

### Logs

I'm not sure if this part of the logfile is correct or enough ... I have a lot more available.

`
2021-01-01 15:57:10.924 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:SEQ:38
2021-01-01 15:57:10.925 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:OK
2021-01-01 15:57:10.926 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisSendUnicastCommand [address=39399, sourceEp=0, destEp=0, profileId=0, clusterId=49, messageData=08 03, messageId=56, status=SUCCESS]
2021-01-01 15:57:10.927 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX STA: msgTag=08 state=TX_ACK
2021-01-01 15:57:10.928 [DEBUG] [transaction.ZigBeeTransactionManager] - notifyTransactionProgress: TID=08, state=TX_ACK, outstanding=5
2021-01-01 15:57:10.929 [DEBUG] [zigbee.transaction.ZigBeeTransaction] - Transaction state updated: TID 08 -> TX_ACK == TRANSMITTED
2021-01-01 15:57:11.034 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:SR:00,00124B0007502BF8,99E7
2021-01-01 15:57:11.036 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisRouteRecordMessageEvent [hops=0, remoteAddress=00124B0007502BF8, networkRoute=[39399]]
2021-01-01 15:57:11.037 [DEBUG] [ngle.telegesis.ZigBeeDongleTelegesis] - Unhandled Telegesis Frame: TelegesisRouteRecordMessageEvent [hops=0, remoteAddress=00124B0007502BF8, networkRoute=[39399]]
2021-01-01 15:57:11.037 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:ACK:38
2021-01-01 15:57:11.038 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisAckMessageEvent [messageId=56]
2021-01-01 15:57:11.039 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX STA: msgTag=08 state=RX_ACK
2021-01-01 15:57:11.040 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:SR:00,00124B0007502BF8,99E7
2021-01-01 15:57:11.040 [DEBUG] [transaction.ZigBeeTransactionManager] - notifyTransactionProgress: TID=08, state=RX_ACK, outstanding=5
2021-01-01 15:57:11.041 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisRouteRecordMessageEvent [hops=0, remoteAddress=00124B0007502BF8, networkRoute=[39399]]
2021-01-01 15:57:11.045 [DEBUG] [zigbee.transaction.ZigBeeTransaction] - Transaction state updated: TID 08 -> RX_ACK == ACKED
2021-01-01 15:57:11.046 [DEBUG] [ngle.telegesis.ZigBeeDongleTelegesis] - Unhandled Telegesis Frame: TelegesisRouteRecordMessageEvent [hops=0, remoteAddress=00124B0007502BF8, networkRoute=[39399]]
2021-01-01 15:57:11.050 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:RX:99E7,0000,00,00,8031,47: ÉÜ»9¢!GUÿÿÿÿÿÿÿÿ3$9ÿ¬ÉÜ»9¢!GUÿÿÿÿÿÿÿÿ  8ÿÉÜ»9¢!GUÿÿÿÿÿÿÿÿð
9ÿ:,-64,FF
2021-01-01 15:57:11.052 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisReceiveMessageEvent [ieeeAddress=null, networkAddress=39399, profileId=0, destinationEp=0, sourceEp=0, clusterId=32817, messageData=08 00 0B 03 03 C9 DC BB 39 A2 21 47 55 FF FF FF FF FF FF FF FF 33 24 39 02 FF AC C9 DC BB 39 A2 21 47 55 FF FF FF FF FF FF FF FF 00 00 38 02 FF 17 C9 DC BB 39 A2 21 47 55 FF FF FF FF FF FF FF FF F0 0D 39 02 FF 3A, rssi=-100, lqi=255]
2021-01-01 15:57:11.055 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=99E7/0, destinationAddress=0000/0, profile=0000, cluster=8031, addressMode=null, radius=0, apsSecurity=false, ackRequest=false, apsCounter=--, rssi=--, lqi=--, payload=08 00 0B 03 03 C9 DC BB 39 A2 21 47 55 FF FF FF FF FF FF FF FF 33 24 39 02 FF AC C9 DC BB 39 A2 21 47 55 FF FF FF FF FF FF FF FF 00 00 38 02 FF 17 C9 DC BB 39 A2 21 47 55 FF FF FF FF FF FF FF FF F0 0D 39 02 FF 3A]
2021-01-01 15:57:11.058 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ManagementLqiResponse [99E7/0 -> 0000/0, cluster=8031, TID=08, status=SUCCESS, neighborTableEntries=11, startIndex=3, neighborTableList=[NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=FFFFFFFFFFFFFFFF, networkAddress=9267, deviceType=ROUTER, rxOnWhenIdle=UNKNOWN, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=255, lqi=172], NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=FFFFFFFFFFFFFFFF, networkAddress=0, deviceType=COORDINATOR, rxOnWhenIdle=UNKNOWN, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=255, lqi=23], NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=FFFFFFFFFFFFFFFF, networkAddress=3568, deviceType=ROUTER, rxOnWhenIdle=UNKNOWN, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=255, lqi=58]]]
2021-01-01 15:57:11.060 [DEBUG] [transaction.ZigBeeTransactionManager] - notifyTransactionCommand: ManagementLqiResponse [99E7/0 -> 0000/0, cluster=8031, TID=08, status=SUCCESS, neighborTableEntries=11, startIndex=3, neighborTableList=[NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=FFFFFFFFFFFFFFFF, networkAddress=9267, deviceType=ROUTER, rxOnWhenIdle=UNKNOWN, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=255, lqi=172], NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=FFFFFFFFFFFFFFFF, networkAddress=0, deviceType=COORDINATOR, rxOnWhenIdle=UNKNOWN, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=255, lqi=23], NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=FFFFFFFFFFFFFFFF, networkAddress=3568, deviceType=ROUTER, rxOnWhenIdle=UNKNOWN, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=255, lqi=58]]] 
2021-01-01 15:57:11.062 [DEBUG] [transaction.ZigBeeTransactionManager] - notifyTransactionCommand: ManagementLqiResponse [99E7/0 -> 0000/0, cluster=8031, TID=08, status=SUCCESS, neighborTableEntries=11, startIndex=3, neighborTableList=[NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=FFFFFFFFFFFFFFFF, networkAddress=9267, deviceType=ROUTER, rxOnWhenIdle=UNKNOWN, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=255, lqi=172], NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=FFFFFFFFFFFFFFFF, networkAddress=0, deviceType=COORDINATOR, rxOnWhenIdle=UNKNOWN, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=255, lqi=23], NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=FFFFFFFFFFFFFFFF, networkAddress=3568, deviceType=ROUTER, rxOnWhenIdle=UNKNOWN, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=255, lqi=58]]] ZigBeeTransaction [ieeeAddress=000D6F000AF8E2B3 queueTime=7685, state=RESPONDED, sendCnt=1, command=ManagementRoutingRequest [0000/0 -> 0000/0, cluster=0032, TID=06, startIndex=0]]
2021-01-01 15:57:11.064 [DEBUG] [transaction.ZigBeeTransactionManager] - notifyTransactionCommand: ManagementLqiResponse [99E7/0 -> 0000/0, cluster=8031, TID=08, status=SUCCESS, neighborTableEntries=11, startIndex=3, neighborTableList=[NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=FFFFFFFFFFFFFFFF, networkAddress=9267, deviceType=ROUTER, rxOnWhenIdle=UNKNOWN, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=255, lqi=172], NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=FFFFFFFFFFFFFFFF, networkAddress=0, deviceType=COORDINATOR, rxOnWhenIdle=UNKNOWN, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=255, lqi=23], NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=FFFFFFFFFFFFFFFF, networkAddress=3568, deviceType=ROUTER, rxOnWhenIdle=UNKNOWN, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=255, lqi=58]]] ZigBeeTransaction [ieeeAddress=null queueTime=7952, state=RESPONDED, sendCnt [... cut ...]
2021-01-01 15:57:11.799 [DEBUG] [transaction.ZigBeeTransactionManager] - Transaction complete: ZigBeeTransaction [ieeeAddress=000D6F000AF8E2B3 queueTime=150, state=COMPLETE, sendCnt=1, command=ManagementRoutingRequest [0000/0 -> 0000/0, cluster=0032, TID=0D, startIndex=15]]
2021-01-01 15:57:11.800 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisSendUnicastCommand [address=52910, sourceEp=1, destEp=1, profileId=260, clusterId=0, messageData=00 0E 00 02 00, messageId=null]
`

cdjackson, 2021-01-01:
We need to work out the attributes that the device requires. I can guess what they might be, but I noted that somewhere in the forum post there's mention about "non standard".

In the `userdata/zigbee/network` folder will be an XML for this device - can you post that here please.

adahmen, 2021-01-01:
I found a file zigbee-network--zigbee_coordinator_telegesis_0500201F.xml located in /var/lib/openhab2/zigbee.
Is this the correct file?

If yes, the file has about 21.000 lines.
Wich part do you need ... I think I should not post 21.000 lines ;-)

cdjackson, 2021-01-01:
Hmmm - I'm surprised it's that long. Is that a single "network.xml" file, or is it named based on the single device? The older bindings used a single file for all devices, but this changed quite a while ago so that files were shorter and writing was quicker on slow computers.

I need all the information for the device in question...

adahmen, 2021-01-01:
I can not find the recently added device (zigbee:device:0500201F:00124b00073009aa) in this file :-(
Whatever the reason is.

But I found I similar device with another id (zigbee:device:0500201F:00124b00072a6c3e) - maybe this helps.

`
      <ZigBeeEndpoint>
        <profileId>260</profileId>
        <endpointId>1</endpointId>
        <inputClusterIds/>
        <outputClusterIds/>
        <inputClusters>
          <ZclCluster>
            <clusterId>0</clusterId>
            <isClient>false</isClient>
            <attributes>
              <entry>
                <int>0</int>
                <ZclAttribute>
                  <cluster>BASIC</cluster>
                  <id>0</id>
                  <name>ZCLVersion</name>
                  <dataType>UNSIGNED_8_BIT_INTEGER</dataType>
                  <mandatory>true</mandatory>
                  <implemented>false</implemented>
                  <readable>true</readable>
                  <writeable>false</writeable>
                  <reportable>false</reportable>
                  <minimumReportingPeriod>0</minimumReportingPeriod>
                  <maximumReportingPeriod>0</maximumReportingPeriod>
                  <reportingTimeout>0</reportingTimeout>
                  <lastReportTime>
                    <time>1586068366579</time>
                    <timezone>Europe/Berlin</timezone>
                  </lastReportTime>
                  <lastValue class="int">1</lastValue>
                </ZclAttribute>
              </entry>
              <entry>
                <int>16384</int>
                <ZclAttribute>
                  <cluster>BASIC</cluster>
                  <id>16384</id>
                  <name>SWBuildID</name>
                  <dataType>CHARACTER_STRING</dataType>
                  <mandatory>false</mandatory>
                  <implemented>false</implemented>
                  <readable>true</readable>
                  <writeable>false</writeable>
                  <reportable>false</reportable>
                  <minimumReportingPeriod>0</minimumReportingPeriod>
                  <maximumReportingPeriod>0</maximumReportingPeriod>
                  <reportingTimeout>0</reportingTimeout>
                </ZclAttribute>
              </entry>
              <entry>
                <int>1</int>
                <ZclAttribute>
                  <cluster>BASIC</cluster>
                  <id>1</id>
                  <name>ApplicationVersion</name>
                  <dataType>UNSIGNED_8_BIT_INTEGER</dataType>
                  <mandatory>true</mandatory>
                  <implemented>false</implemented>
                  <readable>true</readable>
                  <writeable>false</writeable>
                  <reportable>false</reportable>
                  <minimumReportingPeriod>0</minimumReportingPeriod>
                  <maximumReportingPeriod>0</maximumReportingPeriod>
                  <reportingTimeout>0</reportingTimeout>
[... 676 lines cut ...]
              <entry>
                <int>20</int>
                <ZclAttribute>
                  <cluster>BASIC</cluster>
                  <id>20</id>
                  <name>DisableLocalConfig</name>
                  <dataType>BITMAP_8_BIT</dataType>
                  <mandatory>true</mandatory>
                  <implemented>false</implemented>
                  <readable>true</readable>
                  <writeable>true</writeable>
                  <reportable>false</reportable>
                  <minimumReportingPeriod>0</minimumReportingPeriod>
                  <maximumReportingPeriod>0</maximumReportingPeriod>
                  <reportingTimeout>0</reportingTimeout>
                </ZclAttribute>
              </entry>
            </attributes>
            <supportedCommandsReceived/>
            <supportedCommandsGenerated/>
            <supportedAttributes class="sorted-set"/>
          </ZclCluster>
        </outputClusters>
      </ZigBeeEndpoint>
`

adahmen, 2021-01-01:
Hhhmmm, maybe this an oder file from a backup that is not any longer used.

I found a directory /var/lib/openhab2/zigbee/zigbee_coordinator_telegesis_0500201F

Inside this directory I found an XML file named 00124B00073009AA.xml (that is the id of the recently added device).

This file has 14.500 lines - so I'm not able to post the content ... too many chars :-(
Any idea how to give you access?

adahmen, 2021-01-01:
Here the first around 1000 lines:

`<ZigBeeNode>
  <ieeeAddress>00124B00073009AA</ieeeAddress>
  <networkAddress>52910</networkAddress>
  <nodeDescriptor>
    <apsFlags>0</apsFlags>
    <bufferSize>80</bufferSize>
    <complexDescriptorAvailable>false</complexDescriptorAvailable>
    <manufacturerCode>0</manufacturerCode>
    <logicalType>ROUTER</logicalType>
    <serverCapabilities class="sorted-set"/>
    <incomingTransferSize>160</incomingTransferSize>
    <outgoingTransferSize>160</outgoingTransferSize>
    <userDescriptorAvailable>false</userDescriptorAvailable>
    <frequencyBands>
      <FrequencyBandType>FREQ_2400_MHZ</FrequencyBandType>
    </frequencyBands>
    <macCapabilities class="sorted-set">
      <MacCapabilitiesType>FULL_FUNCTION_DEVICE</MacCapabilitiesType>
      <MacCapabilitiesType>MAINS_POWER</MacCapabilitiesType>
      <MacCapabilitiesType>RECEIVER_ON_WHEN_IDLE</MacCapabilitiesType>
    </macCapabilities>
    <extendedEndpointListAvailable>false</extendedEndpointListAvailable>
    <extendedSimpleDescriptorListAvailable>false</extendedSimpleDescriptorListAvailable>
    <stackCompliance>0</stackCompliance>
  </nodeDescriptor>
  <powerDescriptor>
    <currentPowerMode>RECEIVER_ON_IDLE</currentPowerMode>
    <availablePowerSources>
      <PowerSourceType>MAINS</PowerSourceType>
    </availablePowerSources>
    <currentPowerSource>MAINS</currentPowerSource>
    <powerLevel>FULL</powerLevel>
  </powerDescriptor>
  <endpoints>
    <ZigBeeEndpoint>
      <endpointId>1</endpointId>
      <profileId>260</profileId>
      <deviceId>81</deviceId>
      <deviceVersion>0</deviceVersion>
      <inputClusterIds/>
      <outputClusterIds/>
      <inputClusters>
        <ZclCluster>
          <clusterId>0</clusterId>
          <isClient>false</isClient>
          <attributes>
            <entry>
              <int>0</int>
              <ZclAttribute>
                <id>0</id>
                <name>ZCL Version</name>
                <dataType>UNSIGNED_8_BIT_INTEGER</dataType>
                <mandatory>true</mandatory>
                <implemented>false</implemented>
                <readable>true</readable>
                <writable>false</writable>
                <reportable>false</reportable>
                <minimumReportingPeriod>0</minimumReportingPeriod>
                <maximumReportingPeriod>0</maximumReportingPeriod>
                <reportingTimeout>0</reportingTimeout>
                <lastReportTime>
                  <time>1609513031526</time>
                  <timezone>Europe/Berlin</timezone>
                </lastReportTime>
                <lastValue class="int">1</lastValue>
              </ZclAttribute>
            </entry>
            <entry>
[... 1564 lines cut ...]
                <name>Image Stamp</name>
                <dataType>UNSIGNED_32_BIT_INTEGER</dataType>
                <mandatory>true</mandatory>
                <implemented>false</implemented>
                <readable>true</readable>
                <writable>false</writable>
                <reportable>false</reportable>
                <minimumReportingPeriod>0</minimumReportingPeriod>
                <maximumReportingPeriod>0</maximumReportingPeriod>
                <reportingTimeout>0</reportingTimeout>
              </ZclAttribute>
            </entry>
          </attributes>
          <supportedCommandsReceived class="java.util.Collections$UnmodifiableSet">
            <c class="sorted-set"/>
          </supportedCommandsReceived>
          <supportedCommandsGenerated class="java.util.Collections$UnmodifiableSet">
            <c class="sorted-set"/>
          </supportedCommandsGenerated>
        </ZclCluster>
      </outputClusters>
    </ZigBeeEndpoint>
  </endpoints>
  <bindingTable/>
</ZigBeeNode>`

cdjackson, 2021-01-01:
Can you put one of these files somewhere like dropbox and then paste the link here?

I think that at least one of these is using the SmartEnergy clusters which I wasn't really expecting. It's not a problem, but just shows that we need to properly understand what to implement.

adahmen, 2021-01-01:
Here is the link: https://www.dropbox.com/s/ckctw2k2mgzn170/00124B00073009AA.xml?dl=0

cdjackson, 2021-01-02:
So this confirms that it is using the SEP clusters, which is fine, but these clusters have a huge number of attributes and most devices only support a small number...

Are you able to upgrade to the latest 2.5 binding?  If so, we can try a new console command I added yesterday. This was created recently in the ZSS library for pretty much this exact reason - to find out EXACTLY what a device supports...

adahmen, 2021-01-02:
Do you have a download link for the latest version?
What is the correct directory for the binding? Is it /usr/share/openhab2/addons?

cdjackson, 2021-01-02:
Are you using Linux? If so, the best thing is to use the script from @5iver on the forum as it will sort out the dependencies.

https://community.openhab.org/t/zigbee-and-zwave-manual-install-script/50270

If you don’t use a ’nix device, then let me know and I’ll try and provide links - it’s not just a single JAR though so it’s not quite as simple as some bindings as we need to manage the dependencies.

adahmen, 2021-01-02:
I‘m using a Raspberry Pi 4 ... so I will try the script ...

adahmen, 2021-01-03:
Before I upgrade : is the latest version of the binding compatible with OpenHAB 2.5.6 ?

cdjackson, 2021-01-03:
Hmmm - that is a good question. I have a funny feeling that there were some breaking changes introduced in the core at some point that then needed changes in bindings, so the answer might be no. I’m just not sure unfortunately.

Let’s do something different…

Please try the following command

zigbee attsupported ENDPOINT 0x702 

Where endpoint is the node/endpoint format - so 1234/1 or ox1234/1. You should be able to use the zigbee endpoints command to list the endpoints.

That should dump a list of attributes supported by that cluster and hopefully will tell us what is needed.

adahmen, 2021-01-03:
That is the output:
openhab> smarthome:zigbee attsupported 52910/1 0x702                                                                                                                                     
Failed to retrieve supported attributes

Maybe this helps:
`openhab> smarthome:zigbee endpoint 52910/1
IEEE Address     : 00124B00073009AA
Network Address  : 52910
Endpoint         : 1
Device Profile   : 0x0104, ZIGBEE_HOME_AUTOMATION
Device Type      : 0x0051, SMART_PLUG
Device Version   : 0
Input Clusters   : (Server)
   0000 Basic
     - APS Security disabled
        S       0 r-- UNSIGNED_8_BIT_INTEGER    ZCL Version                              Fri Jan 01 15:57:11 CET 2021 1
        S       1 r-- UNSIGNED_8_BIT_INTEGER    Application Version                      Fri Jan 01 15:57:11 CET 2021 15
        S       2 r-- UNSIGNED_8_BIT_INTEGER    Stack Version                             
        S       3 r-- UNSIGNED_8_BIT_INTEGER    HW Version                               Fri Jan 01 15:57:11 CET 2021 0
        S       4 r-- CHARACTER_STRING          Manufacturer Name                        Fri Jan 01 15:57:11 CET 2021 Bitron Home
        S       5 r-- CHARACTER_STRING          Model Identifier                         Fri Jan 01 15:57:11 CET 2021 902010/25
        S       6 r-- CHARACTER_STRING          Date Code                                Fri Jan 01 15:57:11 CET 2021 20150629
        S       7 r-- ENUMERATION_8_BIT         Power Source                              
        S       8 r-- ENUMERATION_8_BIT         Generic Device Class                      
        S       9 r-- ENUMERATION_8_BIT         Generic Device Type                       
        S      10 r-- CHARACTER_STRING          Product Code                              
        S      11 r-- CHARACTER_STRING          Product URL                               
        S      16 rw- CHARACTER_STRING          Location Description                      
        S      17 rw- ENUMERATION_8_BIT         Physical Environment                      
        S      18 rw- BOOLEAN                   Device Enabled                            
        S      19 rw- BITMAP_8_BIT              Alarm Mask                                
        S      20 rw- BITMAP_8_BIT              Disable Local Config                      
        S   16384 r-- CHARACTER_STRING          SW Build ID                               
   0003 Identify
     - APS Security disabled
        S       0 rw- UNSIGNED_16_BIT_INTEGER   Identify Time                             
   0004 Groups
     - APS Security disabled
        S       0 r-- BITMAP_8_BIT              Name Support                              
   0005 Scenes
     - APS Security disabled
        S       0 r-- UNSIGNED_8_BIT_INTEGER    Scene Count                               
        S       1 r-- UNSIGNED_8_BIT_INTEGER    Current Scene                             
        S       2 r-- UNSIGNED_16_BIT_INTEGER   Current Group                             
        S       3 r-- BOOLEAN                   Scene Valid                               
        S       4 r-- BITMAP_8_BIT              Name Support                              
        S       5 r-- IEEE_ADDRESS              Last Configured By                        
   0006 On/Off
     - APS Security disabled
        S       0 r-s BOOLEAN                   On Off                                   Sun Jan 03 12:01:35 CET 2021 true
        U   16384 r-- BOOLEAN                   Global Scene Control                      
        U   16385 rw- UNSIGNED_16_BIT_INTEGER   On Time                                   
        U   16386 rw- UNSIGNED_16_BIT_INTEGER   Off Wait Time                             
        U   16387 rw- ENUMERATION_8_BIT         Start Up On Off                           
   0702 Metering
     - APS Security disabled
        S       0 r-- UNSIGNED_48_BIT_INTEGER   Current Summation Delivered               
        S       1 r-- UNSIGNED_48_BIT_INTEGER   Current Summation Received                
        S       2 r-- UNSIGNED_48_BIT_INTEGER   Current Max Demand Delivered              
        S       3 r-- UNSIGNED_48_BIT_INTEGER   Current Max Demand Received               
        S       4 r-- UNSIGNED_48_BIT_INTEGER   Dft Summation                             
        S       5 r-- UNSIGNED_16_BIT_INTEGER   Daily Freeze Time                         
        S       6 r-- SIGNED_8_BIT_INTEGER      Power Factor                              
        S       7 r-- UTCTIME                   Reading Snapshot Time                     
        S       8 r-- UTCTIME                   Current Max Demand Delivered Time         
        S       9 r-- UTCTIME                   Current Max Demand Received Time          
        S      10 r-- UNSIGNED_8_BIT_INTEGER    Default Update Period                     
        S      11 r-- UNSIGNED_8_BIT_INTEGER    Fast Poll Update Period                   
        S      12 r-- UNSIGNED_48_BIT_INTEGER   Current Block Period Consumption Delivered  
        S      13 r-- UNSIGNED_24_BIT_INTEGER   Daily Consumption Target                  
[... 821 lines cut ...]
        S    3145 r-- UNSIGNED_32_BIT_INTEGER   Previous Month 7 Alternative Consumption Delivered  
        S    3146 r-- UNSIGNED_32_BIT_INTEGER   Previous Month 7 Alternative Consumption Received  
        S    3147 r-- UNSIGNED_32_BIT_INTEGER   Previous Month 9 Alternative Consumption Delivered  
[... cut ...]
        S    3160 r-- UNSIGNED_32_BIT_INTEGER   Previous Month 21 Alternative Consumption Received  
        S    3161 r-- UNSIGNED_32_BIT_INTEGER   Previous Month 23 Alternative Consumption Delivered  
        S    3162 r-- UNSIGNED_32_BIT_INTEGER   Previous Month 23 Alternative Consumption Received  
        S    3163 r-- UNSIGNED_32_BIT_INTEGER   Previous Month 25 Alternative Consumption Delivered  
        S    3164 r-- UNSIGNED_32_BIT_INTEGER   Previous Month 25 Alternative Consumption Received  
Output Clusters  : (Client)
   0000 Basic
     - APS Security disabled

`

cdjackson, 2021-01-03:
Unfortunately the list above is just the standard list of all the attributes, so this I know. I was hoping that the device would report what attributes is actually supports - it won't be all of them, and is probably only a very small number. I guess it doesn't support this then.

Ok, let's try something else.....

`smarthome:zigbee read 52910/1 0x702 ATTR`

Change ATTR to be one of the attributes you want to test from the list above. Eg -:

```
        S       1 r-- UNSIGNED_48_BIT_INTEGER   Current Summation Received                
```

would be `smarthome:zigbee read 52910/1 0x702 1`

Please provide the attribute numbers that return something, and also what they return - probably you only need to test a couple to get what you want.

adahmen, 2021-01-03:
HHmmmm, I tried a lot ... but without success:

`openhab> smarthome:zigbee read 52910/1 0x702 6
Reading endpoint CEAE/1, cluster server cluster Metering (0702), attributes Power Factor
Error executing command: CommandResult [TIMEOUT]

openhab> smarthome:zigbee read 52910/1 0x702 23
Reading endpoint CEAE/1, cluster server cluster Metering (0702), attributes Inlet Temperature
Error executing command: CommandResult [TIMEOUT]

openhab> smarthome:zigbee read 52910/1 0x702 257
Reading endpoint CEAE/1, cluster server cluster Metering (0702), attributes Current Tier 1 Summation Delivered
Error executing command: CommandResult [TIMEOUT]

openhab> smarthome:zigbee read 52910/1 0x702 518
Reading endpoint CEAE/1, cluster server cluster Metering (0702), attributes Current Meter ID
Error executing command: CommandResult [TIMEOUT]

openhab> smarthome:zigbee read 52910/1 0x702 776
Reading endpoint CEAE/1, cluster server cluster Metering (0702), attributes Meter Serial Number
Error executing command: CommandResult [TIMEOUT]

openhab> smarthome:zigbee read 52910/1 0x702 774
Reading endpoint CEAE/1, cluster server cluster Metering (0702), attributes Metering Device Type
Error executing command: CommandResult [TIMEOUT]

openhab> smarthome:zigbee read 52910/1 0x702 1025
Reading endpoint CEAE/1, cluster server cluster Metering (0702), attributes Current Day Consumption Delivered
Error executing command: CommandResult [TIMEOUT]

openhab> smarthome:zigbee read 54768/1 0x702 1025                               
Reading endpoint D5F0/1, cluster server cluster Metering (0702), attributes Current Day Consumption Delivered
Error executing command: CommandResult [TIMEOUT]

openhab> smarthome:zigbee read 52910/1 0x702 1058
Reading endpoint CEAE/1, cluster server cluster Metering (0702), attributes Previous Day 2 Consumption Delivered
Error executing command: CommandResult [TIMEOUT]

openhab> smarthome:zigbee read 52910/1 0x702 1059
Reading endpoint CEAE/1, cluster server cluster Metering (0702), attributes Previous Day 2 Consumption Received
Error executing command: CommandResult [TIMEOUT]

openhab> smarthome:zigbee read 52910/1 0x702 508
Reading endpoint CEAE/1, cluster server cluster Metering (0702), attributes CPP 1 Summation Delivered
Error executing command: CommandResult [TIMEOUT]`

adahmen, 2021-01-03:
I found this pdf : http://media.bitronvideo.eu/bitronhome/Bitron_Home_ZigBee_Clusters_and_Attributes_-_en_0814.pdf

Page 5 refers to the smartmeter.
Maybe this is the needed information?

cdjackson, 2021-01-03:
Try 0 and 1024 - it seems these are the only ones available.

adahmen, 2021-01-03:
:-(

openhab> smarthome:zigbee read 52910/1 0x0702 0
Reading endpoint CEAE/1, cluster server cluster Metering (0702), attributes Current Summation Delivered
Error executing command: CommandResult [TIMEOUT]

openhab> smarthome:zigbee read 52910/1 0x0702 1024
Reading endpoint CEAE/1, cluster server cluster Metering (0702), attributes Instantaneous Demand
Error executing command: CommandResult [TIMEOUT]

cdjackson, 2021-01-03:
It's strange that these are returning a timeout? Are you able to communicate at all with the device?

Eg. use the on/off commands (on 52910/1 and off 52910/1).

Another thing would be to get the debug log - I don't think it will help too much, but...

Also, please use three ticks on a separate line above and below when posting logs etc - not one. I've edited most of your earlier posts, but you can see in the last post above what is wrong. It just makes it easier to read :)

adahmen, 2021-01-03:
Yes, communication seems to work:

```
openhab> smarthome:zigbee read 52910/1 0x0006 0                                                                                                                                          
Reading endpoint CEAE/1, cluster server cluster On/Off (0006), attributes On Off
Response for cluster 0x0006
Attribute 0, type BOOLEAN, value: true

What do you mean with "use the on/off commands"? 

Switching between on and off using PaperUI works perfect. No issues.

adahmen, 2021-01-03:
Maybe this helps:

```openhab> smarthome:zigbee node 52910 
IEEE Address     : 00124B00073009AA
Network Address  : 52910
Node Descriptor  : NodeDescriptor [apsFlags=0, bufferSize=80, complexDescriptorAvailable=false, manufacturerCode=0000, logicalType=ROUTER, serverCapabilities=[], incomingTransferSize=160, outgoingTransferSize=160, userDescriptorAvailable=false, frequencyBands=[FREQ_2400_MHZ], macCapabilities=[FULL_FUNCTION_DEVICE, MAINS_POWER, RECEIVER_ON_WHEN_IDLE], extendedEndpointListAvailable=false, extendedSimpleDescriptorListAvailable=false, stackCompliance=0]
Power Descriptor : PowerDescriptor [currentPowerMode=RECEIVER_ON_IDLE, availablePowerSources=[MAINS], currentPowerSource=MAINS, powerLevel=FULL]
Associations     : []
Endpoints        : 
            1    : Profile     0104 ZIGBEE_HOME_AUTOMATION
                 : Device Type 0051 SMART_PLUG
                   -> BASIC
                   -> IDENTIFY
                   -> GROUPS
                   -> SCENES
                   -> ON_OFF
                   -> METERING
                   <- BASIC
            4    : Profile     0104 ZIGBEE_HOME_AUTOMATION
                 : Device Type 0009 MAINS_POWER_OUTLET
                   <- OTA_UPGRADE
Neighbors:
NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=FFFFFFFFFFFFFFFF, networkAddress=13800, deviceType=ROUTER, rxOnWhenIdle=UNKNOWN, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=255, lqi=54]
NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=00124B00074DC09E, networkAddress=3469, deviceType=END_DEVICE, rxOnWhenIdle=RX_OFF, relationship=CHILD, permitJoining=UNKNOWN, depth=4, lqi=125]
NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=FFFFFFFFFFFFFFFF, networkAddress=9267, deviceType=ROUTER, rxOnWhenIdle=UNKNOWN, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=255, lqi=99]
NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=FFFFFFFFFFFFFFFF, networkAddress=19851, deviceType=ROUTER, rxOnWhenIdle=UNKNOWN, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=255, lqi=66]
NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=FFFFFFFFFFFFFFFF, networkAddress=54768, deviceType=ROUTER, rxOnWhenIdle=UNKNOWN, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=255, lqi=58]
NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=FFFFFFFFFFFFFFFF, networkAddress=39399, deviceType=ROUTER, rxOnWhenIdle=UNKNOWN, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=255, lqi=94]
NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=FFFFFFFFFFFFFFFF, networkAddress=3568, deviceType=ROUTER, rxOnWhenIdle=UNKNOWN, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=255, lqi=33]
NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=FFFFFFFFFFFFFFFF, networkAddress=0, deviceType=COORDINATOR, rxOnWhenIdle=UNKNOWN, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=255, lqi=14]
NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=00124B00072A7797, networkAddress=25623, deviceType=ROUTER, rxOnWhenIdle=UNKNOWN, relationship=PARENT, permitJoining=UNKNOWN, depth=2, lqi=121]
NeighborTable [extendedPanId=554721A239BBDCC9, extendedAddress=FFFFFFFFFFFFFFFF, networkAddress=2889, deviceType=ROUTER, rxOnWhenIdle=UNKNOWN, relationship=UNKNOWN, permitJoining=UNKNOWN, depth=255, lqi=44]
Routes:
RoutingTable [destinationAddress=0, status=ACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=39399]

```

or this

```
openhab> smarthome:zigbee nodes
Network  Addr  IEEE Address      Logical Type  State      EP   Profile                    Device Type                Manufacturer     Model          
      0  0000  000D6F000AF8E2B3  COORDINATOR   UNKNOWN  
   2889  0B49  00124B00072A6C2E  ROUTER        UNKNOWN     1  ZIGBEE_HOME_AUTOMATION     SMART_PLUG                 Bitron Home      902010/25      
                                                           4  ZIGBEE_HOME_AUTOMATION     MAINS_POWER_OUTLET                                         
   3469  0D8D  00124B00074DC09E  END_DEVICE    ONLINE      1  ZIGBEE_HOME_AUTOMATION     IAS_ZONE                   Bitron Home      902010/24      
   3568  0DF0  00124B00072A6C3E  ROUTER        UNKNOWN     1  ZIGBEE_HOME_AUTOMATION     SMART_PLUG                 Bitron Home      902010/25      
                                                           4  ZIGBEE_HOME_AUTOMATION     MAINS_POWER_OUTLET                                         
   7823  1E8F  00158D00041CD620  END_DEVICE    UNKNOWN     1  ZIGBEE_HOME_AUTOMATION     5F01                       LUMI             lumi.sensor_magnet.aq2
   8842  228A  00158D00042192E1  END_DEVICE    UNKNOWN     1  ZIGBEE_HOME_AUTOMATION     5F01                       LUMI             lumi.weather   
   9012  2334  00158D00020182B2  END_DEVICE    UNKNOWN     1  ZIGBEE_HOME_AUTOMATION     5F01                       LUMI             lumi.weather   
   9267  2433  00124B00072ADA35  ROUTER        UNKNOWN     1  ZIGBEE_HOME_AUTOMATION     SMART_PLUG                 Bitron Home      902010/25      
                                                           4  ZIGBEE_HOME_AUTOMATION     MAINS_POWER_OUTLET                                         
  11726  2DCE  00158D0000723CBC  ROUTER        UNKNOWN     1  ZIGBEE_LIGHT_LINK          1000                                                       
[... cut ...]
  54768  D5F0  00124B00077E344D  ROUTER        ONLINE      1  ZIGBEE_HOME_AUTOMATION     SMART_PLUG                 Bitron Home      902010/25      
                                                           4  ZIGBEE_HOME_AUTOMATION     MAINS_POWER_OUTLET                                         
  56013  DACD  00158D0002C9F3C5  END_DEVICE    ONLINE      1  ZIGBEE_HOME_AUTOMATION     5F01                       LUMI             lumi.weather   
  58799  E5AF  00158D0002C8CBFB  END_DEVICE    UNKNOWN     1  ZIGBEE_HOME_AUTOMATION     5F01                       LUMI             lumi.weather   
```

cdjackson, 2021-01-03:
These are really just different views of the same information I already have. I don't really know why the device isn't responding to this...

Hmmm - I think I know what might be up. Please get a debug log - I think there will be an error about unsupported local cluster...

> What do you mean with "use the on/off commands"?

Exactly as I wrote - there are simple console commands that send on and off commands. If it works in P-UI, then fine.

I think the issue will be what I've written above. If you can confirm that, I will look to implement these two channels...

At some point though, we will come back to your earlier question -:

> Before I upgrade : is the latest version of the binding compatible with OpenHAB 2.5.6 ?

Clearly you will need to upgrade to get this functionality...

adahmen, 2021-01-03:
What Debug-Log do you need / mean?

Just set the log level of the binding with this command: 
```
log:set DEBUG com.zsmartsystems.zigbee
```

And afterwards run this command : 
```
openhab> smarthome:zigbee read 52910/1 0x0702 1024   
```
?

adahmen, 2021-01-03:
Here is the output:

```
2021-01-03 16:12:56.328 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1

2021-01-03 16:12:56.329 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisDisplayNetworkInformationCommand [device=null, channel=null, power=null, panId=null, epanId=null]

2021-01-03 16:12:56.331 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis Data:AT+N?

2021-01-03 16:12:56.361 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:+N=COO,24,08,0B7C,554721A239BBDCC9

2021-01-03 16:12:56.364 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:OK

2021-01-03 16:12:56.365 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisDisplayNetworkInformationCommand [device=COO, channel=24, power=8, panId=2940, epanId=554721A239BBDCC9, status=SUCCESS]

2021-01-03 16:12:57.332 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1

2021-01-03 16:12:57.334 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisDisplayNetworkInformationCommand [device=null, channel=null, power=null, panId=null, epanId=null]

2021-01-03 16:12:57.335 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis Data:AT+N?

2021-01-03 16:12:57.373 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:+N=COO,24,08,0B7C,554721A239BBDCC9

2021-01-03 16:12:57.376 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:OK

2021-01-03 16:12:57.378 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisDisplayNetworkInformationCommand [device=COO, channel=24, power=8, panId=2940, epanId=554721A239BBDCC9, status=SUCCESS]

2021-01-03 16:12:58.033 [DEBUG] [e.transaction.ZigBeeTransactionQueue] - 00124B00073009AA: Added transaction to queue, len=2, transaction=ZigBeeTransaction [ieeeAddress=00124B00073009AA queueTime=0, state=WAITING, sendCnt=0, command=ReadAttributesCommand [Metering: 0000/0 -> CEAE/1, cluster=0702, TID=--, identifiers=[1024]]]

2021-01-03 16:12:58.036 [DEBUG] [transaction.ZigBeeTransactionManager] - CEAE/1: Sending ZigBeeTransaction [ieeeAddress=00124B00073009AA queueTime=95301, state=WAITING, sendCnt=1, command=ReadAttributesCommand [Metering: 0000/0 -> CEAE/1, cluster=0702, TID=E4, identifiers=[1024]]]

2021-01-03 16:12:58.038 [DEBUG] [transaction.ZigBeeTransactionManager] - transactionListenerAdded: 1 outstanding

2021-01-03 16:12:58.040 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ReadAttributesCommand [Metering: 0000/0 -> CEAE/1, cluster=0702, TID=E4, identifiers=[1024]]

2021-01-03 16:12:58.042 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=228, commandId=0]

2021-01-03 16:12:58.044 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0000/1, destinationAddress=CEAE/1, profile=0104, cluster=0702, addressMode=DEVICE, radius=31, apsSecurity=false, ackRequest=true, apsCounter=CA, rssi=--, lqi=--, payload=00 E4 00 00 04]

2021-01-03 16:12:58.046 [DEBUG] [ngle.telegesis.ZigBeeDongleTelegesis] - Telegesis send: TelegesisSendUnicastCommand [address=52910, sourceEp=1, destEp=1, profileId=260, clusterId=1794, messageData=00 E4 00 00 04, messageId=null]

2021-01-03 16:12:58.048 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis queue: 1

2021-01-03 16:12:58.050 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisSendUnicastCommand [address=52910, sourceEp=1, destEp=1, profileId=260, clusterId=1794, messageData=00 E4 00 00 04, messageId=null]

2021-01-03 16:12:58.054 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis Data:AT+SENDUCASTB:05,CEAE,01,01,0104,0702

ä

2021-01-03 16:12:58.086 [DEBUG] [transaction.ZigBeeTransactionManager] - CEAE/1: Sending ZigBeeTransaction [ieeeAddress=00124B00073009AA queueTime=53, state=WAITING, sendCnt=0, command=ReadAttributesCommand [Metering: 0000/0 -> CEAE/1, cluster=0702, TID=EB, identifiers=[1024]]]

2021-01-03 16:12:58.087 [DEBUG] [transaction.ZigBeeTransactionManager] - transactionListenerAdded: 2 outstanding

2021-01-03 16:12:58.089 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:SEQ:53

2021-01-03 16:12:58.091 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:OK

2021-01-03 16:12:58.091 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ReadAttributesCommand [Metering: 0000/0 -> CEAE/1, cluster=0702, TID=EB, identifiers=[1024]]

2021-01-03 16:12:58.092 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisSendUnicastCommand [address=52910, sourceEp=1, destEp=1, profileId=260, clusterId=1794, messageData=00 E4 00 00 04, messageId=83, status=SUCCESS]

2021-01-03 16:12:58.094 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX STA: msgTag=E4 state=TX_ACK

2021-01-03 16:12:58.095 [DEBUG] [transaction.ZigBeeTransactionManager] - notifyTransactionProgress: TID=E4, state=TX_ACK, outstanding=2

2021-01-03 16:12:58.095 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=235, commandId=0]

2021-01-03 16:12:58.101 [DEBUG] [zigbee.transaction.ZigBeeTransaction] - Transaction state updated: TID E4 -> TX_ACK == TRANSMITTED

[... cut ...]
2021-01-03 16:13:16.437 [DEBUG] [gesis.internal.TelegesisFrameHandler] - TX Telegesis Data:AT+N?

2021-01-03 16:13:16.469 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:+N=COO,24,08,0B7C,554721A239BBDCC9

2021-01-03 16:13:16.471 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis Data:OK

2021-01-03 16:13:16.473 [DEBUG] [gesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisDisplayNetworkInformationCommand [device=COO, channel=24, power=8, panId=2940, epanId=554721A239BBDCC9, status=SUCCESS]

```

adahmen, 2021-01-03:
This is what I performed:

````
openhab> log:set DEBUG com.zsmartsystems.zigbee
openhab> smarthome:zigbee read 52910/1 0x0702 1024
Reading endpoint CEAE/1, cluster server cluster Metering (0702), attributes Instantaneous Demand
Error executing command: CommandResult [TIMEOUT]

openhab> log:set ERROR com.zsmartsystems.zigbee
````

cdjackson, 2021-01-03:
That's perfect - thanks. That confirmed what I thought was the problem...

I will take a look at adding these two channels.

adahmen, 2021-01-03:
Good to read that you found the problem.
Looking forward for a new version of the binding :-)
Thanks a lot !!!

PR #646 (2021-04-25): Add metering channels
Adds metering instantaneous demand, sum delivered and sum received channels.

Closes #625 
Closes #164
closed 2021-04-25 by cdjackson