openhab/org.openhab.binding.zigbee #632: Aqara button no longer working

bartkummel, 2021-01-06:
### Guidelines

### Outline

I have an Xiaomi Aqara wallswitch. Recognized as  `LUMI lumi.remote.b286acn01`. It was working before I upgraded from openHAB 2.5M1 to 3.0. After the upgrade, the rules that used to be triggered by the channels of the button are no longer triggered.

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used | Ember EM35x                                                                               |
| openHAB version  | old version: 2.5M1, new version: 3.0     |
| Hardware         | Raspberry Pi 2 Model B Rev 1.1                                                                               |
| Memory           |  1 GB                                                                              |
| Java version     |  `openjdk 11.0.9 2020-10-20 LTS`<br/>`OpenJDK Runtime Environment Zulu11.43+88-CA (build 11.0.9+11-LTS)`<br/>`OpenJDK Client VM Zulu11.43+88-CA (build 11.0.9+11-LTS, mixed mode)`                                                                           |
| Devices          | see below |
```
openhab> zigbee nodes
Total known nodes in network: 8
Network  Addr  IEEE Address      Logical Type  State      EP   Profile                    Device Type                Manufacturer     Model
      0  0000  000D6F000C8691BA  COORDINATOR   UNKNOWN
  17792  4580  00158D0003F4382F  END_DEVICE    ONLINE      1  ZIGBEE_HOME_AUTOMATION     OCCUPANCY_SENSOR           LUMI             lumi.sensor_motion.aq2
  20762  511A  D0CF5EFFFE260C16  ROUTER        ONLINE      1  ZIGBEE_LIGHT_LINK          POINT_OF_SALE              IKEA of Sweden   TRADFRI bulb E27 WS opal 980lm
  21650  5492  000D6FFFFE03C639  ROUTER        ONLINE      1  ZIGBEE_HOME_AUTOMATION     COLOR_TEMPERATURE_LIGHT    IKEA of Sweden   TRADFRI bulb E27 WS opal 1000lm
                                                         242  A1E0                       ZGP_PROXY_BASIC
  42350  A56E  000D6FFFFEE843C4  ROUTER        ONLINE      1  ZIGBEE_LIGHT_LINK          ON_OFF_LIGHT               IKEA of Sweden   TRADFRI bulb E27 W opal 1000lm
  46444  B56C  00158D00031387A0  END_DEVICE    ONLINE      1  ZIGBEE_HOME_AUTOMATION     5F01                       LUMI             lumi.remote.b286acn01
                                                           2  ZIGBEE_HOME_AUTOMATION     5F02
                                                           3  ZIGBEE_HOME_AUTOMATION     5F03
  53960  D2C8  000D6FFFFEE80288  ROUTER        ONLINE      1  ZIGBEE_LIGHT_LINK          ON_OFF_LIGHT               IKEA of Sweden   TRADFRI bulb E27 W opal 1000lm
  56292  DBE4  90FD9FFFFED1AD53  ROUTER        ONLINE      1  ZIGBEE_LIGHT_LINK          POINT_OF_SALE              IKEA of Sweden   TRADFRI bulb E27 WS opal 980lm
```

### Logs
```
2021-01-04 20:44:28.100 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingRouteRecordHandler [networkId=0, source=B56C, sourceEui=00158D00031387A0, lastHopLqi=255, lastHopRssi=-72, relayList=511A]
2021-01-04 20:44:28.150 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingMessageHandler [networkId=0, type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=0104, clusterId=0012, sourceEndpoint=2, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY], groupId=0, sequence=4A], lastHopLqi=255, lastHopRssi=-72, sender=B56C, bindingIndex=255, addressIndex=255, messageContents=18 12 0A 55 00 21 00 00]
2021-01-04 20:44:28.153 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=B56C/2, destinationAddress=0000/1, profile=0104, cluster=0012, addressMode=DEVICE, radius=0, apsSecurity=false, ackRequest=false, apsCounter=4A, rssi=-72, lqi=FF, payload=18 12 0A 55 00 21 00 00]
2021-01-04 20:44:28.155 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - Incoming message from unknown node B56C: Notifying announce listeners
2021-01-04 20:44:28.162 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - Unknown remote node B56C
2021-01-04 20:44:28.164 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - Incoming message from node B56C did not translate to command
2021-01-04 20:44:30.100 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingRouteRecordHandler [networkId=0, source=B56C, sourceEui=00158D00031387A0, lastHopLqi=255, lastHopRssi=-73, relayList=511A]
2021-01-04 20:44:30.152 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingMessageHandler [networkId=0, type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=0104, clusterId=0000, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY], groupId=0, sequence=4B], lastHopLqi=255, lastHopRssi=-73, sender=B56C, bindingIndex=255, addressIndex=255, messageContents=1C 5F 11 13 0A F0 FF 41 09 AA 10 05 41 87 13 01 10 01]
2021-01-04 20:44:30.156 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=B56C/1, destinationAddress=0000/1, profile=0104, cluster=0000, addressMode=DEVICE, radius=0, apsSecurity=false, ackRequest=false, apsCounter=4B, rssi=-73, lqi=FF, payload=1C 5F 11 13 0A F0 FF 41 09 AA 10 05 41 87 13 01 10 01]
2021-01-04 20:44:30.158 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - Incoming message from unknown node B56C: Notifying announce listeners
2021-01-04 20:44:30.165 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - Unknown remote node B56C
2021-01-04 20:44:30.167 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - Incoming message from node B56C did not translate to command
2021-01-04 20:44:38.879 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspTrustCenterJoinHandler [networkId=0, newNodeId=B56C, newNodeEui64=00158D00031387A0, status=EMBER_DEVICE_LEFT, policyDecision=EMBER_NO_ACTION, parentOfNewNodeId=511A]
2021-01-04 20:44:38.881 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - 00158D00031387A0: nodeStatusUpdate - node status is DEVICE_LEFT, network address is B56C.
2021-01-04 20:44:38.900 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspTrustCenterJoinHandler [networkId=0, newNodeId=B56C, newNodeEui64=00158D00031387A0, status=EMBER_DEVICE_LEFT, policyDecision=EMBER_NO_ACTION, parentOfNewNodeId=511A]
2021-01-04 20:44:38.903 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - 00158D00031387A0: nodeStatusUpdate - node status is DEVICE_LEFT, network address is B56C.
2021-01-04 20:44:39.196 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspChildJoinHandler [networkId=0, index=0, joining=true, childId=B56C, childEui64=00158D00031387A0, childType=EMBER_SLEEPY_END_DEVICE]
2021-01-04 20:44:39.199 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - 00158D00031387A0: nodeStatusUpdate - node status is UNSECURED_JOIN, network address is B56C.
2021-01-04 20:44:39.203 [DEBUG] [pp.discovery.ZigBeeNetworkDiscoverer] - 00158D00031387A0: Device status updated. NWK=B56C
2021-01-04 20:44:39.206 [DEBUG] [pp.discovery.ZigBeeNetworkDiscoverer] - 00158D00031387A0: NWK Discovery add node B56C
2021-01-04 20:44:39.210 [DEBUG] [com.zsmartsystems.zigbee.ZigBeeNode ] - 00158D00031387A0: Node state updated from UNKNOWN to ONLINE
2021-01-04 20:44:39.212 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - 00158D00031387A0: Updating node NWK=B56C
2021-01-04 20:44:39.218 [DEBUG] [atabase.ZigBeeNetworkDatabaseManager] - 00158D00031387A0: Data store: Deferring write for 250ms.
2021-01-04 20:44:39.218 [DEBUG] [p.discovery.ZigBeeDiscoveryExtension] - 00158D00031387A0: DISCOVERY Extension: Adding discoverer for node
2021-01-04 20:44:39.223 [DEBUG] [iscovery.ZigBeeNodeServiceDiscoverer] - 00158D00031387A0: Node SVC Discovery: created discoverer
2021-01-04 20:44:39.227 [DEBUG] [iscovery.ZigBeeNodeServiceDiscoverer] - 00158D00031387A0: Node SVC Discovery: start discovery
2021-01-04 20:44:39.230 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 00158D00031387A0: Starting ZigBee device discovery
2021-01-04 20:44:39.230 [DEBUG] [iscovery.ZigBeeNodeServiceDiscoverer] - 00158D00031387A0: Node SVC Discovery: starting new tasks [NWK_ADDRESS, POWER_DESCRIPTOR, ACTIVE_ENDPOINTS, NODE_DESCRIPTOR]
2021-01-04 20:44:39.233 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - 00158D00031387A0: Creating ZigBee device zigbee:device with bridge zigbee:coordinator_ember:stick
[... cut ...] essIndex=255, messageContents=18 0A 0A 55 00 21 01 00]
2021-01-04 20:52:49.155 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=B56C/1, destinationAddress=0000/1, profile=0104, cluster=0012, addressMode=DEVICE, radius=0, apsSecurity=false, ackRequest=false, apsCounter=CF, rssi=-56, lqi=FF, payload=18 0A 0A 55 00 21 01 00]
2021-01-04 20:52:49.162 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - Incoming message from node B56C did not translate to command

```

bartkummel, 2021-01-06:
See also [this thread in teh openHAB community](https://community.openhab.org/t/aqara-button-not-working-anymore-after-oh3-upgrade) where is issue was discussed before I opened this ticket.

stage-rl, 2021-01-21:
I have similar problem, can't pair Aqara Temperature & Humidity sensor. The other 6 already paired work well.

cdjackson, 2021-04-24:
> It was working before I upgraded from openHAB 2.5M1 to 3.0. 

My guess is that there is an initialisation issue and probably reinitialising the device should solve it. There haven't been changes from 2.5 to 3 in the binding.

> I have similar problem, can't pair Aqara Temperature & Humidity sensor. The other 6 already paired work well.

@stage-rl how is that in any way related to this issue? If you have a problem with one sensor, then maybe it's a problem with the device - if you have 6 others that work fine, it's unlikely to be an issue with the binding. I'd suggest to discuss somewhere else unless it's somehow related to this switch.

bartkummel, 2021-04-25:
Thanks for your reply. I'm sorry to say, but in the meantime I switched to HomeAssistant for my home automation. It is more stable and far more easy to use in general. It runs faster on the same hardware: there is noticeable less lag between pushing a button and the reaction. So I won't be able to do any more tests. @cdjackson Thanks for your support in the past. Goog luck with further developments.

closed 2021-04-25 by cdjackson

stage-rl, 2021-04-27:
Well, just want to add, that I also chose the easy way - to migrate to something else. In my case, I stayed with OH3, but moved to Zigbee2mqtt. As a OH3 fan, I'm sad to see ppl solving it like this and seems to be a popular choice among users. Maybe it would make sense to have a compliant Zigbee addon version and a compatible version in OH3. If the reason for these "malfunctions" is incompatibility of specific devices (Xiaomi or others).

bartkummel, 2021-04-28:
I was also a fan once. I think there are problems with OH that go beyond the Zigbee binding. 

- The focus is too much on technology instead of on the end user. This discussion about compliancy vs. compatibility is a good example of that. One other example is how you update OH. You have to do this on the OS level. A user-friendly way is to do this via the web interface. Even if you aren't scared for SSH, it is a hassle to update. 
- The community is unfriendly. A lot of people in the community react like: "You are asking stupid questions, go away!" instead of realizing that people asking questions is a sign that the information available is not good enough. 
- Big bang releases. This is a bad idea and that is known in the software industry for very long now. Software should be release fast and often.

cdjackson, 2021-04-28:
Really there is limited point in discussing this sort of thing here. I'd suggest to discuss this on the forum - I agree with your points in general, but I have very limited influence on openHAB unfortunately.