openhab/org.openhab.binding.zigbee #636: Add support for SAGE Doorbell Sensor 206612

allanstevens, 2021-02-04:
Would it be possible to add support for the SAGE Doorbell Sensor 206612?

It pairs fine in OH3 as an Echostar Bell.  The  channels batterylevel, level control, battery alarm and battery voltage are added.  There is no action channel as shown over at mqtt (https://www.zigbee2mqtt.io/devices/SAGE206612.html)  It does not report anything back and the battery shows as 0% 0v.  Ringing the doorbell also returns nothing.

I hope someone would be able to get this added, if you need anymore information from me then please get in touch.

Many thanks for looking at this,

cdjackson, 2021-02-05:
Please provide logs, or some information that shows what this device supports. The XML file that OH produces would possibly also be of use. Alternatively if you are able to add the support yourself this would be appreciated as it may be difficult to do this without a device,

allanstevens, 2021-02-05:
Many thanks for getting back to me.  I'm very new to all this, I have certainly learned a bit so far!

When I ring the bell I get this in the logs once debug is switched on.  

```
2021-02-05 12:35:07.396 [DEBUG] [converter.ZigBeeConverterSwitchLevel] - 0008890001D9F36D: ZigBee command received OnCommand [On/Off: E828/18 -> 0000/1, cluster=0006, TID=00]

2021-02-05 12:35:07.397 [DEBUG] [converter.ZigBeeBaseChannelConverter] - 0008890001D9F36D: Channel zigbee:device:stick1:0008890001d9f36d:0008890001D9F36D_18_dimmer updated to 100

2021-02-05 12:35:07.398 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 0008890001D9F36D: Updating ZigBee channel state zigbee:device:stick1:0008890001d9f36d:0008890001D9F36D_18_dimmer to 100
```

I also list the XML below.  Would you be able to offer some advice on how to add support myself ?

Thanks again!

0008890001D9F36D.XML
```
<ZigBeeNode>
  <ieeeAddress>0008890001D9F36D</ieeeAddress>
  <networkAddress>59432</networkAddress>
  <nodeDescriptor>
    <apsFlags>0</apsFlags>
    <bufferSize>89</bufferSize>
    <complexDescriptorAvailable>false</complexDescriptorAvailable>
    <manufacturerCode>4116</manufacturerCode>
    <logicalType>END_DEVICE</logicalType>
    <serverCapabilities class="sorted-set"/>
    <incomingTransferSize>61</incomingTransferSize>
    <outgoingTransferSize>61</outgoingTransferSize>
    <userDescriptorAvailable>true</userDescriptorAvailable>
    <frequencyBands>
      <FrequencyBandType>FREQ_2400_MHZ</FrequencyBandType>
    </frequencyBands>
    <macCapabilities class="sorted-set">
      <MacCapabilitiesType>REDUCED_FUNCTION_DEVICE</MacCapabilitiesType>
    </macCapabilities>
    <extendedEndpointListAvailable>false</extendedEndpointListAvailable>
    <extendedSimpleDescriptorListAvailable>false</extendedSimpleDescriptorListAvailable>
    <stackCompliance>0</stackCompliance>
  </nodeDescriptor>
  <powerDescriptor>
    <currentPowerMode>RECEIVER_ON_PERIODICALLY</currentPowerMode>
    <availablePowerSources>
      <PowerSourceType>MAINS</PowerSourceType>
      <PowerSourceType>RECHARGABLE_BATTERY</PowerSourceType>
      <PowerSourceType>DISPOSABLE_BATTERY</PowerSourceType>
    </availablePowerSources>
    <currentPowerSource>DISPOSABLE_BATTERY</currentPowerSource>
    <powerLevel>FULL</powerLevel>
  </powerDescriptor>
  <endpoints>
    <ZigBeeEndpoint>
      <endpointId>18</endpointId>
      <profileId>260</profileId>
      <deviceId>260</deviceId>
      <deviceVersion>0</deviceVersion>
      <inputClusterIds/>
      <outputClusterIds/>
      <inputClusters>
        <ZclCluster>
          <clusterId>0</clusterId>
          <isClient>false</isClient>
          <attributes>
            <entry>
              <int>0</int>
              <ZclAttribute>
                <id>0</id>
                <name>ZCL Version</name>
                <dataType>UNSIGNED_8_BIT_INTEGER</dataType>
[... 996 lines cut ...]
                <implemented>false</implemented>
                <readable>true</readable>
                <writable>false</writable>
                <reportable>false</reportable>
                <minimumReportingPeriod>0</minimumReportingPeriod>
                <maximumReportingPeriod>0</maximumReportingPeriod>
                <reportingTimeout>0</reportingTimeout>
              </ZclAttribute>
            </entry>
          </attributes>
          <supportedCommandsReceived class="java.util.Collections$UnmodifiableSet">
            <c class="sorted-set"/>
          </supportedCommandsReceived>
          <supportedCommandsGenerated class="java.util.Collections$UnmodifiableSet">
            <c class="sorted-set"/>
          </supportedCommandsGenerated>
        </ZclCluster>
      </outputClusters>
    </ZigBeeEndpoint>
  </endpoints>
  <bindingTable/>
</ZigBeeNode>
```

LOG

allanstevens, 2021-02-05:
On further investigation, it's working 🤦‍♂️

When doorbell is rung it sets the dimmer control channel to 100.  But it was already at 100 when first setup so I had not realised it changed.  I have now set item to reset state 0 after 5 seconds as shown below.

zigbee.items
`Dimmer  DoorbellControl  "Doorbell Control"   {channel="zigbee:device:stick1:0008890001d9f36d:0008890001D9F36D_18_dimmer", expire="5s,state=0"}`

zigbee.rules
```
rule Doorbell
when 
	Item DoorbellControl changed to 100
then
	//Ding dong
end 
```

The other channels are a bit odd, as the battery is set to 0% and the voltage is 0v,  but its working great.  Thanks for your help.

closed 2021-02-05 by cdjackson