openhab/org.openhab.binding.zigbee #637: Improve compatibility of Ikea Shortcut Button

shorty707, 2021-02-13:
as per 
https://community.openhab.org/t/zigbee-ikea-shortcut-button-discovered-as-dimmer/116839

cdjackson, 2021-02-13:
Please make the issue stand alone so we don't have to hunt through the forum as well as it's unclear what you're asking for. It seems that this device provides a dimmer, and you want to use a rule, which is not really part of the binding. You should be clear about what the bug is that you're reporting - providing information on any device etc.

shorty707, 2021-02-18:
will create some logs as soon as possible to better describe what is discovered wrongly for this device.
however the rule thing mentioned in the thread is just my workaround.

the issue is that device is dicovered with 1 channel as dimmer and every click results in an ON or 100 for that channel.
As shared in the links from the thread in other forums/systems they state that the device actually has 3 different message types / event types (sorry if this is the wrong term) to differntiate the button clicks between "single click" "hold "release".

However like said in the OH Forum thread currently I can not further test the device as it drains the cell battery within very little time of max some hours since it seems to stay awake constantly. There was posting in the thread from somebody who thinks that the device is waiting for a response from the controller to a firmware upgrade... if this is true it might be a possibility that the zigbee binding needs to send something to the device to actually go sleep again.... but that would be another story and another discussion.

cdjackson, 2021-02-18:
> There was posting in the thread from somebody who thinks that the device is waiting for a response from the controller to a firmware upgrade.

Seems very unlikely! The device is responsible for requesting a firmware update, and if it does, the binding will respond that there is no firmware available.

> if this is true it might be a possibility that the zigbee binding needs to send something to the device to actually go sleep again

This is not the way that ZigBee works.