openhab/org.openhab.binding.zigbee #638: Add support for SONOFF SNZB-02ZigBee Temperature & Humidity Sensor

walermanril, 2021-02-23:
as per https://community.openhab.org/t/zigbee-sonoff-snzb-02-discovery-cannot-be-completed-error-node-has-not-completed-discovery/116661
there needs to be some changes to make that switches work with the zigbee binding

cdjackson, 2021-04-24:
> there needs to be some changes to make that switches work with the zigbee binding

Please clearly define what these changes are, or preferably provide a PR with the changes made. Since you have the device (I assume?) you are in the best position to do this.

pmknowles, 2021-04-28:
This is the openhab.log entry when I scan for
1) A SNZB-02 T & H Sensor
2) A SNZB-01 Wireless Switch
2021-04-28 17:24:08.800 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 00124B00226924FC: Starting ZigBee device discovery
2021-04-28 17:24:08.804 [INFO ] [g.discovery.internal.PersistentInbox] - Added new thing 'zigbee:device:ZigBeeUSB:00124b00226924fc' to inbox.
2021-04-28 17:24:24.385 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 00124B00226587F1: Starting ZigBee device discovery
2021-04-28 17:24:24.395 [INFO ] [g.discovery.internal.PersistentInbox] - Added new thing 'zigbee:device:ZigBeeUSB:00124b00226587f1' to inbox.
In events.log
2021-04-28 17:24:08.805 [INFO ] [openhab.event.InboxAddedEvent       ] - Discovery Result with UID 'zigbee:device:ZigBeeUSB:00124b00226924fc' has been added.
2021-04-28 17:24:24.396 [INFO ] [openhab.event.InboxAddedEvent       ] - Discovery Result with UID 'zigbee:device:ZigBeeUSB:00124b00226587f1' has been added.
2021-04-28 17:24:58.930 [INFO ] [openhab.event.InboxRemovedEvent     ] - Discovery Result with UID 'zigbee:device:ZigBeeUSB:00124b00226587f1' has been removed.
2021-04-28 17:24:58.933 [INFO ] [openhab.event.InboxRemovedEvent     ] - Discovery Result with UID 'zigbee:device:ZigBeeUSB:00124b00226924fc' has been removed.
2021-04-28 17:24:58.940 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'zigbee:device:ZigBeeUSB:00124b00226587f1' changed from UNINITIALIZED to INITIALIZING
2021-04-28 17:24:58.944 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'zigbee:device:ZigBeeUSB:00124b00226924fc' changed from UNINITIALIZED to INITIALIZING
2021-04-28 17:24:58.952 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'zigbee:device:ZigBeeUSB:00124b00226587f1' changed from INITIALIZING to UNKNOWN
2021-04-28 17:24:58.955 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'zigbee:device:ZigBeeUSB:00124b00226924fc' changed from INITIALIZING to UNKNOWN
2021-04-28 17:24:58.967 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'zigbee:device:ZigBeeUSB:00124b00226587f1' changed from UNKNOWN to OFFLINE: Node has not completed discovery
2021-04-28 17:24:58.969 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'zigbee:device:ZigBeeUSB:00124b00226924fc' changed from UNKNOWN to OFFLINE: Node has not completed discovery
2021-04-28 17:24:59.384 [INFO ] [penhab.event.FirmwareStatusInfoEvent] - Firmware status of thing zigbee:device:ZigBeeUSB:00124b00226587f1 changed to UNKNOWN.
2021-04-28 17:24:59.428 [INFO ] [penhab.event.FirmwareStatusInfoEvent] - Firmware status of thing zigbee:device:ZigBeeUSB:00124b00226924fc changed to UNKNOWN.

In the status screen for the Thing it says OFFLINE and 'Node has not completed discovery'

There are no other log entries. Without some form of error message explaining why it hasn't completed discovery I wouldn't know where to look.

cdjackson, 2021-04-28:
It's pretty hard to comment on this log as there's so little information. Please take a look at the binding docs to see how to provide debug logging.

However I suspect one of the following is likely your issue -:

* This device is a new device, requiring the new join functionality and your coordinator doesn't support this.
* This device goes to sleep too quickly, and doesn't allow the binding to complete the discovery phase after the device has joined in order to work out what the device is, and what it supports.

pmknowles, 2021-04-28:
Here is the debug log for adding the 2 devices
2021-04-28 22:16:12.914 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - Starting ZigBee scan for zigbee:coordinator_cc2531:ZigBeeUSB
2021-04-28 22:16:12.916 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - Permit join to FFFC/0 for 60 seconds.
2021-04-28 22:16:12.917 [DEBUG] [e.transaction.ZigBeeTransactionQueue] - Broadcast: Added transaction to queue, len=1, transaction=ZigBeeTransaction [ieeeAddress=null queueTime=0, state=WAITING, sendCnt=0, command=ManagementPermitJoiningRequest [0000/0 -> FFFC/0, cluster=0036, TID=--, permitDuration=60, tcSignificance=true]]
2021-04-28 22:16:12.918 [DEBUG] [transaction.ZigBeeTransactionManager] - Transaction Manager: Send Next transaction. outstandingTransactions=0, outstandingQueues=2, sleepy=0/3
2021-04-28 22:16:12.919 [DEBUG] [transaction.ZigBeeTransactionManager] - FFFC/0: Sending ZigBeeTransaction [ieeeAddress=null queueTime=2, state=WAITING, sendCnt=0, command=ManagementPermitJoiningRequest [0000/0 -> FFFC/0, cluster=0036, TID=30, permitDuration=60, tcSignificance=true]]
2021-04-28 22:16:12.920 [DEBUG] [transaction.ZigBeeTransactionManager] - transactionListenerAdded: 1 outstanding
2021-04-28 22:16:12.921 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ManagementPermitJoiningRequest [0000/0 -> FFFC/0, cluster=0036, TID=30, permitDuration=60, tcSignificance=true]
2021-04-28 22:16:12.922 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0000/0, destinationAddress=FFFC/0, profile=0000, cluster=0036, addressMode=DEVICE, radius=8, apsSecurity=false, ackRequest=true, apsCounter=E6, rssi=--, lqi=--, payload=30 3C 01]
2021-04-28 22:16:12.924 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=13, apiId=24 01, data=FE 0D 24 01 FC FF 00 00 36 00 E6 30 08 03 30 3C 01 CD, checksum=CD, error=false)
2021-04-28 22:16:12.924 [DEBUG] [e.transaction.ZigBeeTransactionQueue] - 00124B001CD406D4: Added transaction to queue, len=1, transaction=ZigBeeTransaction [ieeeAddress=00124B001CD406D4 queueTime=0, state=WAITING, sendCnt=0, command=ManagementPermitJoiningRequest [0000/0 -> 0000/0, cluster=0036, TID=--, permitDuration=60, tcSignificance=true]]
2021-04-28 22:16:12.925 [DEBUG] [transaction.ZigBeeTransactionManager] - Transaction Manager: Send Next transaction. outstandingTransactions=1, outstandingQueues=2, sleepy=0/3
2021-04-28 22:16:12.926 [DEBUG] [transaction.ZigBeeTransactionManager] - 0000/0: Sending ZigBeeTransaction [ieeeAddress=00124B001CD406D4 queueTime=2, state=WAITING, sendCnt=0, command=ManagementPermitJoiningRequest [0000/0 -> 0000/0, cluster=0036, TID=31, permitDuration=60, tcSignificance=true]]
2021-04-28 22:16:12.927 [DEBUG] [transaction.ZigBeeTransactionManager] - transactionListenerAdded: 2 outstanding
2021-04-28 22:16:12.928 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ManagementPermitJoiningRequest [0000/0 -> 0000/0, cluster=0036, TID=31, permitDuration=60, tcSignificance=true]
2021-04-28 22:16:12.929 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0000/0, destinationAddress=0000/0, profile=0000, cluster=0036, addressMode=DEVICE, radius=8, apsSecurity=false, ackRequest=true, apsCounter=E7, rssi=--, lqi=--, payload=31 3C 01]
2021-04-28 22:16:12.938 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  AF_DATA_SRSP (FE 01 64 01 00 64)
2021-04-28 22:16:12.939 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX STA: msgTag=30 state=TX_ACK
2021-04-28 22:16:12.940 [DEBUG] [transaction.ZigBeeTransactionManager] - notifyTransactionProgress: TID=30, state=TX_ACK, outstanding=2
2021-04-28 22:16:12.941 [DEBUG] [31.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=13, apiId=24 01, data=FE 0D 24 01 00 00 00 00 36 00 E7 30 08 03 31 3C 01 CE, checksum=CE, error=false)
2021-04-28 22:16:12.941 [DEBUG] [transaction.ZigBeeTransactionManager] - Transaction complete: ZigBeeTransaction [ieeeAddress=null queueTime=23, state=COMPLETE, sendCnt=1, command=ManagementPermitJoiningRequest [0000/0 -> FFFC/0, cluster=0036, TID=30, permitDuration=60, tcSignificance=true]]
2021-04-28 22:16:12.942 [DEBUG] [transaction.ZigBeeTransactionManager] - transactionListenerRemoved: 1 outstanding
2021-04-28 22:16:12.943 [DEBUG] [e.transaction.ZigBeeTransactionQueue] - Broadcast: transactionComplete, state=COMPLETE, outstanding=0
2021-04-28 22:16:12.943 [DEBUG] [transaction.ZigBeeTransactionManager] - Transaction Manager: Send Next transaction. outstandingTransactions=1, outstandingQueues=1, sleepy=0/3
2021-04-28 22:16:12.944 [DEBUG] [zigbee.transaction.ZigBeeTransaction] - Transaction state changed: nwk=FFFC, TID=30, event=TX_ACK, state=COMPLETE
2021-04-28 22:16:12.959 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-  AF_DATA_SRSP (FE 01 64 01 00 64)
2021-04-28 22:16:12.960 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-- ZToolPacket (FE 01 45 CB 3C B3)
2021-04-28 22:16:12.960 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX STA: msgTag=31 state=TX_ACK
2021-04-28 22:16:12.962 [DEBUG] [31.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=1, apiId=45 CB, data=FE 01 45 CB 3C B3, checksum=B3, error=false
2021-04-28 22:16:12.963 [DEBUG] [e.dongle.cc2531.ZigBeeDongleTiCc2531] - Unhandled ZToolPacket type 0x45cb
[... cut ...] nwkAddrAssocDevList=[]]
2021-04-28 22:16:54.425 [DEBUG] [31.network.impl.CommandInterfaceImpl] - <-- ZB_FIND_DEVICE_CONFIRM (FE 0B 46 85 01 52 00 00 80 00 35 00 00 84 00 AA)
2021-04-28 22:16:54.426 [DEBUG] [31.network.impl.CommandInterfaceImpl] - Received Async Cmd: Packet: subsystem=null, length=11, apiId=46 85, data=FE 0B 46 85 01 52 00 00 80 00 35 00 00 84 00 AA, checksum=AA, error=false
2021-04-28 22:16:54.426 [DEBUG] [e.dongle.cc2531.ZigBeeDongleTiCc2531] - Unhandled ZToolPacket type 0x4685

cdjackson, 2021-08-02:
```
Endpoint          : 1
Device Profile    : 0x0104, ZIGBEE_HOME_AUTOMATION
Device Type       : 0x0302, TEMPERATURE_SENSOR
Device Version    : 0
Input Clusters    : (Server)
   0000 Basic
     - APS Security disabled
        S       0 r-- UNSIGNED_8_BIT_INTEGER    ZCL Version                               
        S       1 r-- UNSIGNED_8_BIT_INTEGER    Application Version                       
        S       2 r-- UNSIGNED_8_BIT_INTEGER    Stack Version                             
        S       3 r-- UNSIGNED_8_BIT_INTEGER    HW Version                                
        S       4 r-- CHARACTER_STRING          Manufacturer Name                         
        S       5 r-- CHARACTER_STRING          Model Identifier                          
        S       6 r-- CHARACTER_STRING          Date Code                                 
        S       7 r-- ENUMERATION_8_BIT         Power Source                              
        S       8 r-- ENUMERATION_8_BIT         Generic Device Class                      
        S       9 r-- ENUMERATION_8_BIT         Generic Device Type                       
        S      10 r-- CHARACTER_STRING          Product Code                              
        S      11 r-- CHARACTER_STRING          Product URL                               
        S      16 rw- CHARACTER_STRING          Location Description                      
        S      17 rw- ENUMERATION_8_BIT         Physical Environment                      
        S      18 rw- BOOLEAN                   Device Enabled                            
        S      19 rw- BITMAP_8_BIT              Alarm Mask                                
        S      20 rw- BITMAP_8_BIT              Disable Local Config                      
        S   16384 r-- CHARACTER_STRING          SW Build ID                               
   0001 Power Configuration
     - APS Security disabled
        S       0 r-- UNSIGNED_16_BIT_INTEGER   Mains Voltage                             
        S       1 r-- UNSIGNED_16_BIT_INTEGER   Mains Frequency                           
        S      16 rw- BITMAP_8_BIT              Mains Alarm Mask                          
        S      17 rw- UNSIGNED_16_BIT_INTEGER   Mains Voltage Min Threshold               
        S      18 rw- UNSIGNED_16_BIT_INTEGER   Mains Voltage Max Threshold               
        S      19 rw- UNSIGNED_16_BIT_INTEGER   Mains Voltage Dwell Trip Point            
        S      32 r-- UNSIGNED_8_BIT_INTEGER    Battery Voltage                           
        S      33 r-s UNSIGNED_8_BIT_INTEGER    Battery Percentage Remaining              
        S      48 rw- CHARACTER_STRING          Battery Manufacturer                      
        S      49 rw- ENUMERATION_8_BIT         Battery Size                              
        S      50 rw- UNSIGNED_16_BIT_INTEGER   Battery A Hr Rating                       
        S      51 rw- UNSIGNED_8_BIT_INTEGER    Battery Quantity                          
        S      52 rw- UNSIGNED_8_BIT_INTEGER    Battery Rated Voltage                     
        S      53 rw- BITMAP_8_BIT              Battery Alarm Mask                        
        S      54 rw- UNSIGNED_8_BIT_INTEGER    Battery Voltage Min Threshold             
        S      55 rw- UNSIGNED_8_BIT_INTEGER    Battery Voltage Threshold 1               
        S      56 rw- UNSIGNED_8_BIT_INTEGER    Battery Voltage Threshold 2               
        S      57 rw- UNSIGNED_8_BIT_INTEGER    Battery Voltage Threshold 3               
        S      58 rw- UNSIGNED_8_BIT_INTEGER    Battery Percentage Min Threshold          
        S      59 rw- UNSIGNED_8_BIT_INTEGER    Battery Percentage Threshold 1            
        S      60 rw- UNSIGNED_8_BIT_INTEGER    Battery Percentage Threshold 2            
        S      61 rw- UNSIGNED_8_BIT_INTEGER    Battery Percentage Threshold 3            
        S      62 r-- BITMAP_32_BIT             Battery Alarm State                       
   0003 Identify
     - APS Security disabled
        S       0 rw- UNSIGNED_16_BIT_INTEGER   Identify Time                             
   0402 Temperature Measurement
     - APS Security disabled
        S       0 r-s SIGNED_16_BIT_INTEGER     Measured Value                            
        S       1 r-- SIGNED_16_BIT_INTEGER     Min Measured Value                        
        S       2 r-- SIGNED_16_BIT_INTEGER     Max Measured Value                        
        S       3 r-s UNSIGNED_16_BIT_INTEGER   Tolerance                                 
   0405 Relative Humidity Measurement
     - APS Security disabled
        S       0 r-s UNSIGNED_16_BIT_INTEGER   Measured Value                            
        S       1 r-- UNSIGNED_16_BIT_INTEGER   Min Measured Value                        
        S       2 r-- UNSIGNED_16_BIT_INTEGER   Max Measured Value                        
        S       3 r-s UNSIGNED_16_BIT_INTEGER   Tolerance                                 
Output Clusters   : (Client)
   0003 Identify
     - APS Security disabled
```

joerg1985, 2021-08-10:
I was able to pair it after multiple attemps, but there seens to be more wrong with these sensors or the bindings. After some time (20h - 60h) all of the sensors stop updating values and go offline. After reinitialization of the gateway the sensors are back online, but they will still not update values.

I will try a fresh install, after the debian 11 release and could provide logs, if someone could tell what kind of logs are needed.

cdjackson, 2021-08-11:
Probably if it stops receiving data, there will just be nothing in the logs, but please feel free to provide the debug logs and I will take a look.

FranzSchi, 2022-01-13:
Hi,

I can confirm that the device is detected as eWeLink TH01.
Just one more small wish on this issue:

`  22843  593B  00124B00251333B3  END_DEVICE    ONLINE      1  ZIGBEE_HOME_AUTOMATION     TEMPERATURE_SENSOR         eWeLink          TH01`

The device would also have a cluster for the Battery Lifetime:

   0001 Power Configuration
     - APS Security disabled
        S       0 r-- UNSIGNED_16_BIT_INTEGER   Mains Voltage
        S       1 r-- UNSIGNED_16_BIT_INTEGER   Mains Frequency
        S      16 rw- BITMAP_8_BIT              Mains Alarm Mask
        S      17 rw- UNSIGNED_16_BIT_INTEGER   Mains Voltage Min Threshold
        S      18 rw- UNSIGNED_16_BIT_INTEGER   Mains Voltage Max Threshold
        S      19 rw- UNSIGNED_16_BIT_INTEGER   Mains Voltage Dwell Trip Point
        S      32 r-- UNSIGNED_8_BIT_INTEGER    Battery Voltage
        S      33 r-s UNSIGNED_8_BIT_INTEGER    Battery Percentage Remaining
        S      48 rw- CHARACTER_STRING          Battery Manufacturer
        S      49 rw- ENUMERATION_8_BIT         Battery Size
        S      50 rw- UNSIGNED_16_BIT_INTEGER   Battery A Hr Rating
        S      51 rw- UNSIGNED_8_BIT_INTEGER    Battery Quantity
        S      52 rw- UNSIGNED_8_BIT_INTEGER    Battery Rated Voltage
        S      53 rw- BITMAP_8_BIT              Battery Alarm Mask
        S      54 rw- UNSIGNED_8_BIT_INTEGER    Battery Voltage Min Threshold
        S      55 rw- UNSIGNED_8_BIT_INTEGER    Battery Voltage Threshold 1
        S      56 rw- UNSIGNED_8_BIT_INTEGER    Battery Voltage Threshold 2
        S      57 rw- UNSIGNED_8_BIT_INTEGER    Battery Voltage Threshold 3
        S      58 rw- UNSIGNED_8_BIT_INTEGER    Battery Percentage Min Threshold
        S      59 rw- UNSIGNED_8_BIT_INTEGER    Battery Percentage Threshold 1
        S      60 rw- UNSIGNED_8_BIT_INTEGER    Battery Percentage Threshold 2
        S      61 rw- UNSIGNED_8_BIT_INTEGER    Battery Percentage Threshold 3
        S      62 r-- BITMAP_32_BIT             Battery Alarm State

Could you support here and return the battery Level also?
Here would be the XML file from OH.

[00124B00251333B3.zip](https://github.com/openhab/org.openhab.binding.zigbee/files/7862433/00124B00251333B3.zip)

cdjackson, 2022-01-13:
> The device would also have a cluster for the Battery Lifetime:

I'm not sure what you mean by battery lifetime, but the battery level as a voltage and percentage are already supported in the binding. If the device supports these attributes it should already be supported.

FranzSchi, 2022-01-13:
Hi Chris, 

yes the device has a Cluster 1
The only problem is it is not bound, but I can bind it manually.
Now I have the chicken Egg Problem. If I delete the device it's gone again, so I can never see it in OpenHab.
Could It be that this device has a specific configuration missing Cluster 1?

[image]

BR
/Franz

FranzSchi, 2022-01-14:
Dear valued friends,

and it seems even to report something.

openhab> zigbee nodes | grep 00124B00251333B3
  46535  B5C7  00124B00251333B3  END_DEVICE    ONLINE      1  ZIGBEE_HOME_AUTOMATION     TEMPERATURE_SENSOR         eWeLink          TH01
openhab> zigbee endpoint 46535/1
IEEE Address      : 00124B00251333B3
Network Address   : 46535
Endpoint          : 1
Device Profile    : 0x0104, ZIGBEE_HOME_AUTOMATION
Device Type       : 0x0302, TEMPERATURE_SENSOR
Device Version    : 0
Input Clusters    : (Server)
   0000 Basic
     - APS Security disabled
        S       0 r-- UNSIGNED_8_BIT_INTEGER    ZCL Version                              Thu Jan 13 21:48:06 CET 2022 1
        S       1 r-- UNSIGNED_8_BIT_INTEGER    Application Version                      Thu Jan 13 21:48:06 CET 2022 5
        S       2 r-- UNSIGNED_8_BIT_INTEGER    Stack Version
        S       3 r-- UNSIGNED_8_BIT_INTEGER    HW Version                               Thu Jan 13 21:48:06 CET 2022 1
        S       4 r-- CHARACTER_STRING          Manufacturer Name                        Thu Jan 13 21:48:07 CET 2022 eWeLink
        S       5 r-- CHARACTER_STRING          Model Identifier                         Thu Jan 13 21:48:06 CET 2022 TH01
        S       6 r-- CHARACTER_STRING          Date Code                                Thu Jan 13 21:48:06 CET 2022 20211103
        S       7 r-- ENUMERATION_8_BIT         Power Source
        S       8 r-- ENUMERATION_8_BIT         Generic Device Class
        S       9 r-- ENUMERATION_8_BIT         Generic Device Type
        S      10 r-- CHARACTER_STRING          Product Code
        S      11 r-- CHARACTER_STRING          Product URL
        S      16 rw- CHARACTER_STRING          Location Description
        S      17 rw- ENUMERATION_8_BIT         Physical Environment
        S      18 rw- BOOLEAN                   Device Enabled
        S      19 rw- BITMAP_8_BIT              Alarm Mask
        S      20 rw- BITMAP_8_BIT              Disable Local Config
        S   16384 r-- CHARACTER_STRING          SW Build ID
   0001 Power Configuration
     - APS Security disabled
        S       0 r-- UNSIGNED_16_BIT_INTEGER   Mains Voltage
        S       1 r-- UNSIGNED_16_BIT_INTEGER   Mains Frequency
        S      16 rw- BITMAP_8_BIT              Mains Alarm Mask
        S      17 rw- UNSIGNED_16_BIT_INTEGER   Mains Voltage Min Threshold
        S      18 rw- UNSIGNED_16_BIT_INTEGER   Mains Voltage Max Threshold
        S      19 rw- UNSIGNED_16_BIT_INTEGER   Mains Voltage Dwell Trip Point
        **S      32 r-- UNSIGNED_8_BIT_INTEGER    Battery Voltage                          Fri Jan 14 08:20:54 CET 2022 31
        S      33 r-s UNSIGNED_8_BIT_INTEGER    Battery Percentage Remaining             Fri Jan 14 08:21:16 CET 2022 200**
        S      48 rw- CHARACTER_STRING          Battery Manufacturer
        S      49 rw- ENUMERATION_8_BIT         Battery Size
        S      50 rw- UNSIGNED_16_BIT_INTEGER   Battery A Hr Rating
        S      51 rw- UNSIGNED_8_BIT_INTEGER    Battery Quantity
        S      52 rw- UNSIGNED_8_BIT_INTEGER    Battery Rated Voltage
        S      53 rw- BITMAP_8_BIT              Battery Alarm Mask
        S      54 rw- UNSIGNED_8_BIT_INTEGER    Battery Voltage Min Threshold
        S      55 rw- UNSIGNED_8_BIT_INTEGER    Battery Voltage Threshold 1
        S      56 rw- UNSIGNED_8_BIT_INTEGER    Battery Voltage Threshold 2
        S      57 rw- UNSIGNED_8_BIT_INTEGER    Battery Voltage Threshold 3
        S      58 rw- UNSIGNED_8_BIT_INTEGER    Battery Percentage Min Threshold
        S      59 rw- UNSIGNED_8_BIT_INTEGER    Battery Percentage Threshold 1
        S      60 rw- UNSIGNED_8_BIT_INTEGER    Battery Percentage Threshold 2
        S      61 rw- UNSIGNED_8_BIT_INTEGER    Battery Percentage Threshold 3
        S      62 r-- BITMAP_32_BIT             Battery Alarm State
   0003 Identify
     - APS Security disabled
        S       0 rw- UNSIGNED_16_BIT_INTEGER   Identify Time
   0402 Temperature Measurement
     - APS Security disabled
        S       0 r-s SIGNED_16_BIT_INTEGER     Measured Value                           Fri Jan 14 08:23:54 CET 2022 2428
        S       1 r-- SIGNED_16_BIT_INTEGER     Min Measured Value
        S       2 r-- SIGNED_16_BIT_INTEGER     Max Measured Value
        S       3 r-s UNSIGNED_16_BIT_INTEGER   Tolerance
   0405 Relative Humidity Measurement
     - APS Security disabled
        S       0 r-s UNSIGNED_16_BIT_INTEGER   Measured Value                           Fri Jan 14 08:23:44 CET 2022 2267
        S       1 r-- UNSIGNED_16_BIT_INTEGER   Min Measured Value
        S       2 r-- UNSIGNED_16_BIT_INTEGER   Max Measured Value
        S       3 r-s UNSIGNED_16_BIT_INTEGER   Tolerance
Output Clusters   : (Client)
   0003 Identify
     - APS Security disabled

FranzSchi, 2022-02-04:
Hi @cdjackson,

I have 2 more questions on that behavior.

1. What is the criterial for the Binding to create the Binding Table entry? 
Because that one is missing and even if i create it for the power 0001 Power Configuration cluster the Battery Level does not appear.

2. I've figured out that the device is reporting every 10 Seconds. If you are not using a gas driven glass melting oven then I guess this is to often.
For sure I can adjust it on my own.... for Temperature and Humidity.

zigbee subscribe 19620/1 0x0402 0x0000 120 900 20
zigbee subscribe 19620/1 0x0405 0x0000 120 900 50

But I've wanted to ask you if we could add this into the Binding so that I could edit it in the channel detail settings like shown below.

Thanks a lot
/Franz

[image]

cdjackson, 2022-02-04:
> The only problem is it is not bound, but I can bind it manually.

It's not about if it has this cluster - the question is what attributes it supports - that is the problem. Binding to the cluster will do nothing in itself. The zigbee binding will try and detect what attributes the device supports, and if it supports the attributes the binding knows, then it will provide the channel.

> and it seems even to report something.

I don't know what you're trying to indicate with this dump. This information doesn't mean that the device supports a particular feature - other than the clusters.

> What is the criterial for the Binding to create the Binding Table entry?
> Because that one is missing and even if i create it for the power 0001 Power Configuration cluster the Battery Level does not appear.

So I guess this is my point above - binding is clearly important, but it won't help if the device doesn't support the attribute you want to report.

> But I've wanted to ask you if we could add this into the Binding so that I could edit it in the channel detail settings like shown below.

I can look at adding this, but you can also reduce the minimum reporting period if it's reporting too fast. If you want to only report every 60 seconds, then change this in the configuration.

FranzSchi, 2022-02-15:
Hi Chris, 

thanks so far

> I can look at adding this, but you can also reduce the minimum reporting period if it's reporting too fast. If you want to only report every 60 seconds, then change this in the configuration.

Which configuration? If I add it like this:
zigbee subscribe 19620/1 0x0402 0x0000 120 900 20
zigbee subscribe 19620/1 0x0405 0x0000 120 900 50
Then it's gone after restart of OpenHab.
Is there somewhere a config file?

I guess it would be good in general to have this reporting settings in all Channels (why not)?

> So I guess this is my point above - binding is clearly important, but it won't help if the device doesn't support the attribute you want to report.

Could I config this Power behavior also somewhere in a config?
What's the meaning of "doesn't support"? 
The cluster is there, I can read the config, I see that a subscription is there, is it the command attsupported  that report the supported? Then I would be confused because it does not return anything for any cluster (see below)

```
openhab> zigbee attsupported 19620/1 0x0402
> zigbee attsupported 19620/1 0x0405
> zigbee attsupported 19620/1 0x001
Failed to retrieve supported attributes

Failed to retrieve supported attributes

Failed to retrieve supported attributes

openhab>
```

Also if I take a look to other projects it looks like there is nothing special.

[image]

Source (Line 105 ff from) https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/sonoff.js

Sorry for my unskilled questions
/Franz

cdjackson, 2022-02-19:
> Then it's gone after restart of OpenHab.

I would need to see a log to find out why the device is being reprogrammed as you state here. The configuration is stored in the DEVICE, not in the binding.

> I guess it would be good in general to have this reporting settings in all Channels (why not)?

It should be in most channels, but maybe some are missing. If so, please raise a specific issue for the channel and it can be looked at.

> Could I config this Power behavior also somewhere in a config?

You should be able to configure it using the command line if it is using a standard zigbee configuration (ie attributes). I don't really know what is used here.

> What's the meaning of "doesn't support"?

Most attributes are optional - a device doesn't have to support them, and if it's not supported by the device then the binding can't do anything.

> Also if I take a look to other projects it looks like there is nothing special.

I'm not familiar with this code and do not know what attributes these point to "under the hood" so I think without this information it's impossible to say if this is using standard zigbee functionality (which I guess is what you mean by "nothing special") or if it is using something custom in the other project to support non-standard features. Maybe you already know this? If so, please can we talk about attributes, rather than code from another project that is not really relevant.

FranzSchi, 2022-03-12:
Hello @cdjackson,

sorry for the very late reply I was out of order for a few days.

> > Then it's gone after restart of OpenHab.
> 
> I would need to see a log to find out why the device is being reprogrammed as you state here. The configuration is stored in the DEVICE, not in the binding.

I just tried it out by rebooting OH. 
But was no longer able to reproduce this issue (the reporting remains as it was before)
Maybe its related to the fact that I exchanged the Osram Smart Plug+ Plugs that also confused my Motion sensors.

> > I guess it would be good in general to have this reporting settings in all Channels (why not)?
> 
> It should be in most channels, but maybe some are missing. If so, please raise a specific issue for the channel and it can be looked at.

As recommended I created this Ticket:
https://github.com/openhab/org.openhab.binding.zigbee/issues/743

> I'm not familiar with this code and do not know what attributes these point to "under the hood" so I think without this information it's impossible to say if this is using standard zigbee [...]

About the power configuration:
So it looks like this:
Cluster 0x0001 Attribute: 33
+ Cluster 0x0001 Attribute: 32

openhab> zigbee endpoint 19620/1
IEEE Address      : 00124B00251333B3
Network Address   : 19620
Endpoint          : 1
Device Profile    : 0x0104, ZIGBEE_HOME_AUTOMATION
Device Type       : 0x0302, TEMPERATURE_SENSOR
Device Version    : 0
Input Clusters    : (Server)
   0000 Basic
     - APS Security disabled
        S       0 r-- UNSIGNED_8_BIT_INTEGER    ZCL Version                              Sat Mar 12 13:25:07 CET 2022 1
        S       1 r-- UNSIGNED_8_BIT_INTEGER    Application Version                      Sat Mar 12 13:25:07 CET 2022 5
        S       2 r-- UNSIGNED_8_BIT_INTEGER    Stack Version
        S       3 r-- UNSIGNED_8_BIT_INTEGER    HW Version                               Sat Mar 12 13:25:07 CET 2022 1
        S       4 r-- CHARACTER_STRING          Manufacturer Name                        Sat Mar 12 13:25:07 CET 2022 eWeLink
        S       5 r-- CHARACTER_STRING          Model Identifier                         Sat Mar 12 13:25:07 CET 2022 TH01
        S       6 r-- CHARACTER_STRING          Date Code                                Sat Mar 12 13:25:07 CET 2022 20211103
        S       7 r-- ENUMERATION_8_BIT         Power Source
        S       8 r-- ENUMERATION_8_BIT         Generic Device Class
        S       9 r-- ENUMERATION_8_BIT         Generic Device Type
        S      10 r-- CHARACTER_STRING          Product Code
        S      11 r-- CHARACTER_STRING          Product URL
        S      16 rw- CHARACTER_STRING          Location Description
        S      17 rw- ENUMERATION_8_BIT         Physical Environment
        S      18 rw- BOOLEAN                   Device Enabled
        S      19 rw- BITMAP_8_BIT              Alarm Mask
        S      20 rw- BITMAP_8_BIT              Disable Local Config
        S   16384 r-- CHARACTER_STRING          SW Build ID
   0001 Power Configuration
     - APS Security disabled
        S       0 r-- UNSIGNED_16_BIT_INTEGER   Mains Voltage
        S       1 r-- UNSIGNED_16_BIT_INTEGER   Mains Frequency
        S      16 rw- BITMAP_8_BIT              Mains Alarm Mask
        S      17 rw- UNSIGNED_16_BIT_INTEGER   Mains Voltage Min Threshold
        S      18 rw- UNSIGNED_16_BIT_INTEGER   Mains Voltage Max Threshold
        S      19 rw- UNSIGNED_16_BIT_INTEGER   Mains Voltage Dwell Trip Point
        **S      32 r-- UNSIGNED_8_BIT_INTEGER    Battery Voltage                          Sat Mar 12 13:58:00 CET 2022 30
        S      33 r-s UNSIGNED_8_BIT_INTEGER    Battery Percentage Remaining             Sat Mar 12 13:58:00 CET 2022 200**
        S      48 rw- CHARACTER_STRING          Battery Manufacturer
        S      49 rw- ENUMERATION_8_BIT         Battery Size
        S      50 rw- UNSIGNED_16_BIT_INTEGER   Battery A Hr Rating

The attribute is there, as the date shows its frequently updated.

I can read the attribute:

openhab> zigbee read 19620/1 0x0001 33
Reading endpoint 4CA4/1, cluster server cluster Power Configuration (0001), attributes Battery Percentage Remaining
Response for cluster 0x0001
Attribute   33  Battery Percentage Remaining                        UNSIGNED_8_BIT_INTEGER          200

I can bind it to the device. and read it back.
openhab> zigbee bindtable 19620
Binding table for node 19620 [00124B00251333B3]
Src Address          | Dest Address         | Group | Mode    | Cluster
00124B00251333B3/1   | 04CD15FFFEBB3E7B/1   |       | Address | 0405:RELATIVE_HUMIDITY_MEASUREMENT
00124B00251333B3/1   | 04CD15FFFEBB3E7B/1   |       | Address | 0001:POWER_CONFIGURATION
00124B00251333B3/1   | 04CD15FFFEBB3E7B/1   |       | Address | 0402:TEMPERATURE_MEASUREMENT

cdjackson, 2022-03-12:
The battery percent remaining channel is already supported. If it is not showing in your device, then probably it was not detected during the initialisation of your device. Try reinitialising and keeping the device awake by pressing the button every second or so and it should be detected.

cdjackson, 2022-03-12:
> About the power configuration:
> So it looks like this:

Just to note that this printout doesn't show what the device supports - it shows the complete configuration of this cluster in the zigbee standards. What you list here is exactly what I listed earlier in this thread and doesn't show that the device supports any specific attribute.

Clearly attribute 33 is supported though as you see the reports. As above, it is therefore likely an issue with initialisation.

FranzSchi, 2022-03-15:
Hi @cdjackson,

very good that we are clear with the meaning of supporting now.

> Just to note that this printout doesn't show what the device supports - it shows the complete configuration of this cluster in the ZigBee standards.
It thought by printing a date and a value close to the attribute means that the value could be read and is therefore supported. Is there any difference?

> Try reinitialising and keeping the device awake by pressing the button every second or so and it should be detected.

What I tried is to use the "reinitialize device" slider and press every second the button -> did not succeed
Next I tried to remove the device and added it again while pressing the button every second -> unlucky also not.

Here is what the log of that procedure. I wonder why it might not appear Cluster 0001 is very often mentioned.
[openhab.zip](https://github.com/openhab/org.openhab.binding.zigbee/files/8254039/openhab.zip)

@nailyk-fr: I've seen you've added the sensor to the list of supported devices here: https://github.com/openhab/org.openhab.binding.zigbee/pull/742/commits/3d90740b30714a15ed5da18c5cead4acffc1f42c can you see the Battery Remaining Channel in  OpenHab?

BR
/Franz

cdjackson, 2022-03-17:
I've simplified the way the converter detects the channel - this might solve the problem as the device doesn't support some of the standard zigbee commands used to detect what attributes it supports.

FranzSchi, 2022-03-18:
Hello @cdjackson ,

I've just installed the Snapshot 
`bundle:update -r org.openhab.binding.zigbee https://openhab.jfrog.io/artifactory/libs-snapshot-local/org/openhab/addons/bundles/org.openhab.binding.zigbee/3.3.0-SNAPSHOT/org.openhab.binding.zigbee-3.3.0-20220318.033610-15.jar`

I can confirm that the Battery Channel is now visible and returns a value. 

[image]

The bindtable contains the POWER_CONFIGURATION cluster.

```
openhab> zigbee bindtable 00124B0025128716
Binding table for node 22338 [00124B0025128716]
Src Address          | Dest Address         | Group | Mode    | Cluster
00124B0025128716/1   | 04CD15FFFEBB3E7B/1   |       | Address | 0001:POWER_CONFIGURATION
00124B0025128716/1   | 04CD15FFFEBB3E7B/1   |       | Address | 0405:RELATIVE_HUMIDITY_MEASUREMENT
00124B0025128716/1   | 04CD15FFFEBB3E7B/1   |       | Address | 0402:TEMPERATURE_MEASUREMENT
```

Good Job Chris! Thank you so much and have a great weekend!

/Franz

daccy2001, 2022-03-25:
I've been watching this issue for a while before deciding to buy this Sonoff sensor... so a big thanks to Chris from me, too, for improving support for this sensor, and to Franz and others for collecting all the debug info. I can also report that the Battery Level works with the current snapshot.

I'm also keen to understand more about if/how the reporting interval can be configured (nicely via the gui would be great... but I'm learning more about what I can do via the console all the time). I've only had this sensor for a couple of days, so I've no idea if changing the reporting rate will make the battery last substantially longer (so no idea if this is really a big deal), but the updates do seem to be far more frequent than I need them!

//Andrew

cdjackson, 2022-03-25:
You shouldn't need to play around too much with the reporting. Reporting is not done on time (it is, but it's relatively long) - it's done on change.

The binding sets temperatures to report every 0.1 degree of change - if the temperature doesn't change very fast, then the binding sets the device to report once every 2 hours - just so that it knows it's still alive and ONLINE.

I might look to make this configurable, but for me the battery on these small devices lasts a long time (I've had a similar one running for 18 months and it's still going strong).

FranzSchi, 2022-03-27:
Hi @daccy2001 ,

thanks for your feedback.
I've already created an Issue here to set the Reporting in the GUI. 
https://github.com/openhab/org.openhab.binding.zigbee/issues/743
It's also interesting for other devices like my smart plugs with Metering because then I would be able to poll them from the Plugin and not from the console.

@cdjackson, yes the Sensor supports the "On Changed Value"
The problem is more the 1 for Min reporting because the Sensor reports by default every 10 seconds.

attribute.setReporting(**1**, REPORTING_PERIOD_DEFAULT_MAX, **0.1**).get();

I'm also not sure about 0.1 as changed value, If I modify the reporting on the console I use (10) because the value is stored in a 4 Digit format. like 2514 means 25.14 °
zigbee subscribe 19620/1 0x0402 0x0000 120 900 **10**

```
            if (bindResponse.isSuccess()) {
                // Configure reporting
                ZclAttribute attribute = serverCluster
                        .getAttribute(ZclTemperatureMeasurementCluster.ATTR_MEASUREDVALUE);
                CommandResult reportingResponse = attribute.setReporting(1, REPORTING_PERIOD_DEFAULT_MAX, 0.1).get();
                handleReportingResponse(reportingResponse, POLLING_PERIOD_DEFAULT, REPORTING_PERIOD_DEFAULT_MAX);
            } else {
```

Have a good Sunday
/Franz

cdjackson, 2022-03-27:
> The problem is more the 1 for Min reporting because the Sensor reports by default every 10 seconds.

I don't understand what you mean here? If the sensor always reports every 10 seconds, then that's not something that I can influence in the binding if it will always report at 10 seconds (is that what you mean?). The binding sets the reporting to a value between 1 second and 2 hours - if the data is changing quickly, then it will report as fast as 1 second - if the data is not changing quickly, then it will report every 2 hours. I don't set anything to 10 seconds.

> I'm also not sure about 0.1 as changed value, If I modify the reporting on the console I use (10) because the value is stored in a 4 Digit format. like 2514 means 25.14 °

Hmmm - now this is a very (very!) good point. I'm of course aware that the temperature is presented with a calibration that is " / 100", but mistakes happen and I think that you are correct here that this should be 10. This may be the root cause of your issue with the 10 second reporting.

> Have a good Sunday

Sadly it's Monday already here.

FranzSchi, 2022-04-02:
Hello fellows,

@cdjackson: I'm not sure about Monday / Sunday. Are you located now in NZ or in the UK? Because Github Still sails UK but Monday is Sunday would also point to NZ.

testing Time :-)
[image]

4 Brand new Sensors.

Updated the Plugin:
`bundle:update -r org.openhab.binding.zigbee https://openhab.jfrog.io/artifactory/libs-snapshot-local/org/openhab/addons/bundles/org.openhab.binding.zigbee/3.3.0-SNAPSHOT/org.openhab.binding.zigbee-3.3.0-20220402.033612-17.jar
`

Checked the Configuration ....

```
openhab> zigbee reportcfg 30810/1 0x0402 0
Cluster 1026, Attribute 0, type SIGNED_16_BIT_INTEGER, minimum reporting interval: 1, maximum reporting interval: 7200, timeout: 0

openhab> zigbee reportcfg 30810/1 0x0405 0
Cluster 1029, Attribute 0, type UNSIGNED_16_BIT_INTEGER, minimum reporting interval: 1, maximum reporting interval: 7200, timeout: 0
```

And the result is much better:

[image]

THX to all contributors and have a nice Weekend

/Franz

cdjackson, 2022-04-02:
> Are you located now in NZ or in the UK?

I'm in New Zealand - we moved here about a year ago now :)

I'm pleased it's all working well. Shall we close this issue or is there still something that's not quite right?

closed 2022-05-21 by cdjackson