openhab/org.openhab.binding.zigbee #641: [zigbee] timeout issue after 1830s

Stefan300381, 2021-04-09:
Hi,

I posted this [in the community forum](https://community.openhab.org/t/zigbee-devices-offline-after-polling-timeout-of-1830s/119023) but didn't get a reply. 

I'm testing the zigbee binding with an Ember Coordinator and see my two routers and 1 end device getting flagged as offline after the timeout of 1830s has reached (no clue where this value is defined). However, everything seems to work fine.

Previously, in a different setup with zigbee2mqtt, I could see that the same routers (CC2530+CC2591 with koenkk firmware) send data about every minute. So how come, the current setup marks them as offline after 1830s?

This is what I see in the logs:

```
2021-02-23 19:04:59.547 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'zigbee:device:3a4f49d56f:00158d0003121b0b' changed from OFFLINE to ONLINE
2021-02-23 19:04:59.549 [DEBUG] [.zigbee.handler.ZigBeeIsAliveTracker] - IsAlive Tracker scheduled task for thingUID=zigbee:device:3a4f49d56f:00158d0003121b0b in 1830 seconds
2021-02-23 19:20:34.988 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00158D0003121B0B: Polling [zigbee:device:3a4f49d56f:00158d0003121b0b:00158D0003121B0B_1_switch] channels...
2021-02-23 19:35:29.553 [DEBUG] [.zigbee.handler.ZigBeeIsAliveTracker] - IsAlive Tracker LastChance Timeout has been reached for thingUID=zigbee:device:3a4f49d56f:00158d0003121b0b
2021-02-23 19:35:29.558 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00158D0003121B0B: Polling stopped
2021-02-23 19:35:29.563 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00158D0003121B0B: Polling initialised at 1871140ms
2021-02-23 19:35:59.567 [DEBUG] [.zigbee.handler.ZigBeeIsAliveTracker] - IsAlive Tracker Timeout has been reached for thingUID=zigbee:device:3a4f49d56f:00158d0003121b0b
2021-02-23 19:35:59.573 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'zigbee:device:3a4f49d56f:00158d0003121b0b' changed from ONLINE to OFFLINE
```

thanks in advance

Stefan

cdjackson, 2021-04-10:
Presumably there has not been any information received from the devices, so they get marked offline.

For support questions it’s best to discuss on the forum. This is really for bugs.

Stefan300381, 2021-04-10:
thanks. understood, but in the forum I didnt receive a single response. would love to understand why the devices send data every minute with z2mqtt but not with zigbee binding. also how to configure the timeout, e.g. for my xiaomi sensors who send data only once per hour. all in all it looks like the binding flags devices as offline although they work fine. so it seems to be a "UI" issue.

cdjackson, 2021-04-10:
Ok, I understand, but just because the forum didn’t receive a response doesn’t mean that opening a bug report is the right approach. Unfortunately I’m moving house/job/country at the moment so have been less active on the forum recently but asking here doesn’t really mean I have more time.

Stefan300381, 2021-04-10:
Makes total sense. I assumed that like in the forum multiple people will have a look and might be able to help. At least now I know who the author of the binding is ;-) totally respect that you don't have time right now. Feel free to close the ticket. Would highly appreciate if you come back to me in the forum once you are free again. Take care

closed 2021-04-24 by cdjackson