openhab/org.openhab.binding.zigbee #649: Support for Z-Stack 3 (Z-Stack 3.0.x and Z-Stack 3.x.x) - Texas Instruments Zigbee 3.0 stack on adapters like CC2652 and CC1352

Hedda, 2021-04-29:
### Outline

**Is your feature request related to a problem? Please describe.**

It looks like there is currently no support for Z-Stack 3 (Z-Stack 3.0.x and Z-Stack 3.x.x) - Texas Instruments Zigbee 3.0 stack on adapters like CC2652/CC2652P/CC2652R/CC2652RB and CC1352/CC1352P?

https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/README.md

Personally, I can currently recommend Electrolama zzh (zig-a-zig-ah) CC2652R USB stick/dongle which is open-source hardware:

https://electrolama.com/projects/zig-a-zig-ah/

https://github.com/electrolama/zig-a-zig-ah

**Describe the solution you'd like**

Full support Texas Instruments USB adapter running Z-Stack 3 based firmware (Z-Stack 3.0.x and Z-Stack 3.x.x).

https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator

The primary goal would be to use the newer and more powerful USB adapter based on Texas Instruments CC2652/CC2652P/CC2652R/CC2652RB and CC1352/CC1352P.

Hopefully, that will later also lead to full Zigbee 3.0 compliance but that is only a secondary goal and probably not really a priority.

http://www.ti.com/lit/an/swra615a/swra615a.pdf

**Describe alternatives you've considered**

The primary alternative today, if you like to like a Zigbee 3.0 compliant Zogbee coordinator USB adapter for creating a Zigbee 3 network, is to go with a newer Silicon Labs EmberZNet based adapter like EFR32 (EFR32MG12 or EFR32MG21)

**Additional context**

While not recommended for a "production" Zigbee network it is possible for testing and development purposes to flash CC2538, CC2531 and CC2530 based USB adapters with Z-Stack 3.0.x firmware:

https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.0.x/bin

If you today have a CC2531 USB adapter that is already flashed with Z-Stack Home 1.2 firmware then you can now actually upgrade to a Z-Stack 3.0.x firmware yourself quite easily via USB using zigpy-znp command line tools (as a stand-alone tool):

https://github.com/zigpy/zigpy-znp/blob/dev/TOOLS.md

Note! Be sure to do a backup before you upgrade from Z-Stack Home 1.2 firmware to Z-Stack 3.0.x firmware as you will need to restore that backup afterwards as the upgrade procedure will perform an initialization of the adapter.

https://github.com/zigpy/zigpy-znp/blob/dev/TOOLS.md#backup-and-restore

https://github.com/zigpy/zigpy-znp/blob/dev/TOOLS.md#nvram-backup

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used |  Electrolama zzh (zig-a-zig-ah) CC2652R USB stick           |
| openHAB version  | OpenHAB 3.0                                                   |
| Hardware         |    Raspberry Pi 4 Model B 4GB                 |
| Memory           |                                                                                 |
| Java version     |                                                                                |
| Devices          | Provide all device types and their addresses                                   |

### Logs

NA

cdjackson, 2021-05-12:
Isn't this a duplicate of #605?

Hedda, 2021-05-12:
> Isn't this a duplicate of #605?

Probably in spirit yes but technically no since the MCU hardware is not bound to Z-Stack version and vice versa.

I believe that you could probably get Z-Stack Home 1.2.x working on CC2562 and then it would work with openHAB as it is. 

You can also already run Z-Stack 3.0.x on CC253x (CC2530, CC2531 and CC2538).

https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator

If support is added for Z-Stack 3.x API to openHAB then you will gain support for all CC25x2 and CC13x2 as it default on them.

cdjackson, 2021-05-12:
Again though, there will be not any support added directly to the binding - this needs to be done through the zigbee libraries at per https://github.com/zsmartsystems/com.zsmartsystems.zigbee/issues/1226. I think we should close this - if the support is added to the library, then it will also be supported here.

cdjackson, 2021-05-12:
I will close this as a duplicate of the other mentioined above.

closed 2021-05-12 by cdjackson