openhab/org.openhab.binding.zigbee #653: [Zigbee] ZigbeeThingHandler exception, possibly causing my Aqara things to always appear offline

bjerregaardp, 2021-06-08:
I select Zigbee binding and click Scan. Then I press the button on the Aqara device till its LED starts blinking.

## Expected Behavior
No exceptions in the log.
Zigbee Node updates should make it all the way from discovery to UI.

## Current Behavior
Aqara devices are always listed as either GONE or OFFLINE.
 ```
2021-05-23 12:00:49.894 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - 00158D00067C1BA7: Node discovery not complete
2021-05-23 12:01:08.518 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00158D00067C1BA7: Node updated - ZigBeeNode [state=ONLINE, IEEE=00158D00067C1BA7, NWK=5C66, endpoints=[]]
2021-05-23 12:01:08.519 [WARN ] [zsmartsystems.zigbee.ZigBeeExecutors] - Uncaught exception in thread NotificationService-thread-9
java.lang.NullPointerException: null
	at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.nodeUpdated(ZigBeeThingHandler.java:879) ~[?:?]
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager$14.run(ZigBeeNetworkManager.java:1735) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) ~[?:?]
	at java.lang.Thread.run(Thread.java:829) [?:?]
 ```

The IEEE value in the logline matches the last part of the zigbee device id of a Thing.

## Possible Solution

## Steps to Reproduce (for Bugs)

1. I select Zigbee binding and click Scan. 
2. Then I press the button on the Aqara device till its LED starts blinking.

## Context

## Your Environment

* Version used: OpenHAB 3.1.0.M4
* Environment name and version (e.g. Chrome 76, Java 8, Node.js 12.9, ...):  Edge, Zulu 11
* Operating System and version (desktop or mobile, Windows 10, Raspbian Buster, ...): Windows 10
* Device: Aqara button WXKG11LM

cdjackson, 2021-06-08:
Please provide the full debug log. Please read the binding docs to learn how to enable this.

cdjackson, 2021-08-09:
Closing this due to lack of information. Please feel free to reopen if you can provide more information, and preferably the debug log showing the issue, as requested previously.

closed 2021-08-09 by cdjackson

joni1993, 2021-12-16:
I am experiencing a similar issue with my Smabit Contact. The device is seen when Scanning but upon adding it is stuck at "Node has not completed discovery" + OFFLINE.

```
17:41:09.623 [DEBUG] [ing.zigbee.handler.ZigBeeThingHandler] - 00124B000EE99EED: Node updated - ZigBeeNode [state=ONLINE, IEEE=00124B000EE99EED, NWK=0847, endpoints=[]]
17:41:09.624 [WARN ] [.zsmartsystems.zigbee.ZigBeeExecutors] - Uncaught exception in thread NotificationService-thread-27
java.lang.NullPointerException: null
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.nodeUpdated(ZigBeeThingHandler.java:862) ~[?:?]
        at com.zsmartsystems.zigbee.ZigBeeNetworkManager$14.run(ZigBeeNetworkManager.java:1815) ~[?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) ~[?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) ~[?:?]
        at java.lang.Thread.run(Thread.java:829) [?:?]
17:41:09.876 [DEBUG] [nding.zigbee.internal.ZigBeeDataStore] - 00124B000EE99EED: ZigBee saving network state complete.
17:41:20.515 [DEBUG] [gbee.discovery.ZigBeeDiscoveryService] - 00124B000EE99DC7: Discovery: Starting discovery for existing device
17:41:20.516 [DEBUG] [gbee.discovery.ZigBeeDiscoveryService] - 00124B000EE99EED: Discovery: Starting discovery for existing device
17:41:20.517 [DEBUG] [gbee.discovery.ZigBeeDiscoveryService] - Starting ZigBee scan for zigbee:coordinator_cc2531:91ce26ffb4
17:41:20.526 [INFO ] [gbee.discovery.ZigBeeDiscoveryService] - 00124B000EE99DC7: Starting ZigBee device discovery
17:41:20.527 [INFO ] [gbee.discovery.ZigBeeDiscoveryService] - 00124B000EE99EED: Starting ZigBee device discovery
17:41:20.528 [DEBUG] [gbee.discovery.ZigBeeDiscoveryService] - 00124B000EE99DC7: Creating ZigBee device zigbee:device with bridge zigbee:coordinator_cc2531:91ce26ffb4
17:41:20.528 [DEBUG] [gbee.discovery.ZigBeeDiscoveryService] - 00124B000EE99EED: Creating ZigBee device zigbee:device with bridge zigbee:coordinator_cc2531:91ce26ffb4
17:41:20.528 [DEBUG] [gbee.discovery.ZigBeeDiscoveryService] - 00124B000EE99DC7: Node discovery not complete
17:41:20.530 [DEBUG] [gbee.discovery.ZigBeeDiscoveryService] - 00124B000EE99EED: Node discovery not complete
17:42:31.953 [DEBUG] [.transaction.ZigBeeTransactionManager] - Transaction complete: ZigBeeTransaction [ieeeAddress=00124B000EE99DC7 queueTime=8013, state=FAILED, sendCnt=1, command=PowerDescriptorRequest [0000/0 -> 435D/0, cluster=0003, TID=8B, nwkAddrOfInterest=435D]]
17:42:31.954 [DEBUG] [.transaction.ZigBeeTransactionManager] - transactionListenerRemoved: 0 outstanding
17:42:31.954 [DEBUG] [ee.transaction.ZigBeeTransactionQueue] - 00124B000EE99DC7: transactionComplete, state=FAILED, outstanding=0
17:42:31.954 [DEBUG] [ee.transaction.ZigBeeTransactionQueue] - 00124B000EE99DC7: Added transaction to queue, len=1, transaction=ZigBeeTransaction [ieeeAddress=00124B000EE99DC7 queueTime=8014, state=WAITING, sendCnt=1, command=PowerDescriptorRequest [0000/0 -> 435D/0, cluster=0003, TID=8B, nwkAddrOfInterest=435D]]
17:42:31.954 [DEBUG] [.transaction.ZigBeeTransactionManager] - Transaction Manager: Send Next transaction. outstandingTransactions=0, outstandingQueues=2, sleepy=0/3
17:42:31.955 [DEBUG] [.transaction.ZigBeeTransactionManager] - 435D/0: Sending ZigBeeTransaction [ieeeAddress=00124B000EE99DC7 queueTime=8015, state=WAITING, sendCnt=1, command=PowerDescriptorRequest [0000/0 -> 435D/0, cluster=0003, TID=8B, nwkAddrOfInterest=435D]]
17:42:31.955 [DEBUG] [.transaction.ZigBeeTransactionManager] - transactionListenerAdded: 1 outstanding
17:42:31.955 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - TX CMD: PowerDescriptorRequest [0000/0 -> 435D/0, cluster=0003, TID=8B, nwkAddrOfInterest=435D]
17:42:31.956 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0000/0, destinationAddress=435D/0, profile=0000, cluster=0003, addressMode=DEVICE, radius=8, apsSecurity=false, ackRequest=true, apsCounter=DD, rssi=--, lqi=--, payload=8B 5D 43]
17:42:31.956 [DEBUG] [531.network.impl.CommandInterfaceImpl] - ->  AF_DATA_REQUEST (Packet: subsystem=null, length=13, apiId=24 01, data=FE 0D 24 01 5D 43 00 00 03 00 DD 30 08 03 8B 5D 43 46, checksum=46, error=false)
17:42:31.975 [DEBUG] [531.network.impl.CommandInterfaceImpl] - <-  AF_DATA_SRSP (FE 01 64 01 00 64)
17:42:31.976 [DEBUG] [rtsystems.zigbee.ZigBeeNetworkManager] - RX STA: msgTag=8B state=TX_ACK
17:42:31.976 [DEBUG] [.transaction.ZigBeeTransactionManager] - notifyTransactionProgress: TID=8B, state=TX_ACK, outstanding=1
17:42:31.977 [DEBUG] [.zigbee.transaction.ZigBeeTransaction] - Transaction state changed: nwk=435D, TID=8B, event=TX_ACK, state=TRANSMITTED
17:42:36.478 [DEBUG] [discovery.ZigBeeNodeServiceDiscoverer] - 00124B000EE99EED: Node SVC Discovery: running NWK_ADDRESS
17:42:36.479 [DEBUG] [ee.transaction.ZigBeeTransactionQueue] - 00124B000EE99EED: Added transaction to queue, len=1, transaction=ZigBeeTransaction [ieeeAddress=00124B000EE99EED queueTime=0, state=WAITING, sendCnt=0, command=NetworkAddressRequest [0000/0 -> 0847/0, cluster=0000, TID=--, ieeeAddr=00124B000EE99EED, requestType=0, startIndex=0]]
17:42:36.480 [DEBUG] [.transaction.ZigBeeTransactionManager] - Transaction Manager: Send Next transaction. outstandingTransactions=1, outstandingQueues=2, sleepy=0/3
[... cut ...]
17:43:38.695 [DEBUG] [.transaction.ZigBeeTransactionManager] - transactionListenerRemoved: 1 outstanding
17:43:38.696 [DEBUG] [ee.transaction.ZigBeeTransactionQueue] - 00124B000EE99DC7: transactionComplete but not outstanding, state=CANCELLED, outstanding=0
17:43:38.696 [DEBUG] [.transaction.ZigBeeTransactionManager] - Transaction Manager: Send Next transaction. outstandingTransactions=1, outstandingQueues=1, sleepy=0/3
17:43:38.696 [DEBUG] [.transaction.ZigBeeTransactionManager] - Transaction Manager: Send Next transaction. outstandingTransactions=1, outstandingQueues=1, sleepy=0/3
```

cdjackson, 2021-12-16:
I'd suggest to open a new issue and provide all the information requested when opening an issue.