openhab/org.openhab.binding.zigbee #655: Error in openhab.log

kpanchen, 2021-06-26:
After 3.1.0-snapshot or 3.1.0 RC update zigbee binding logs error in the log file. Was Ok in original 3 released version.

Configuration

Coordinator used - Nortek Security & Control HUSBZB-1
openHAB version - 3.1.0 RC 
Hardware - Intel
Memory - 16gb
Java version - Zulu 11
Devices - no devices

`[openhab.log](https://github.com/openhab/org.openhab.binding.zigbee/files/6719749/openhab.log)`

kaikreuzer, 2021-06-26:
I can confirm this on a fresh and clean 3.1.RC1 installation. 

@cdjackson I consider this a critical issue, could you please have a look?

kaikreuzer, 2021-06-26:
FTR, here's the log details:
```
2021-06-26 10:25:03.396 [ERROR] [al.EmberZigBeeConsoleCommandProvider] - bundle org.openhab.binding.zigbee.console.ember:3.1.0.RC1 (252)[org.openhab.binding.zigbee.console.ember.internal.EmberZigBeeConsoleCommandProvider(307)] : Error during instantiation of the implementation object
java.lang.reflect.InvocationTargetException: null
	at jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method) ~[?:?]
	at jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62) ~[?:?]
	at jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45) ~[?:?]
	at java.lang.reflect.Constructor.newInstance(Constructor.java:490) ~[?:?]
	at org.apache.felix.scr.impl.inject.internal.ComponentConstructorImpl.newInstance(ComponentConstructorImpl.java:316) ~[?:?]
	at org.apache.felix.scr.impl.manager.SingleComponentManager.createImplementationObject(SingleComponentManager.java:286) ~[?:?]
	at org.apache.felix.scr.impl.manager.SingleComponentManager.createComponent(SingleComponentManager.java:115) ~[?:?]
	at org.apache.felix.scr.impl.manager.SingleComponentManager.getService(SingleComponentManager.java:1000) ~[?:?]
	at org.apache.felix.scr.impl.manager.SingleComponentManager.getServiceInternal(SingleComponentManager.java:973) ~[?:?]
	at org.apache.felix.scr.impl.manager.SingleComponentManager.getService(SingleComponentManager.java:918) ~[?:?]
	at org.eclipse.osgi.internal.serviceregistry.ServiceFactoryUse$1.run(ServiceFactoryUse.java:216) ~[org.eclipse.osgi-3.16.200.jar:?]
	at java.security.AccessController.doPrivileged(Native Method) ~[?:?]
	at org.eclipse.osgi.internal.serviceregistry.ServiceFactoryUse.factoryGetService(ServiceFactoryUse.java:213) ~[org.eclipse.osgi-3.16.200.jar:?]
	at org.eclipse.osgi.internal.serviceregistry.ServiceFactoryUse.getService(ServiceFactoryUse.java:114) ~[org.eclipse.osgi-3.16.200.jar:?]
	at org.eclipse.osgi.internal.serviceregistry.ServiceConsumer$2.getService(ServiceConsumer.java:48) ~[org.eclipse.osgi-3.16.200.jar:?]
	at org.eclipse.osgi.internal.serviceregistry.ServiceRegistrationImpl.getService(ServiceRegistrationImpl.java:547) ~[org.eclipse.osgi-3.16.200.jar:?]
	at org.eclipse.osgi.internal.serviceregistry.ServiceRegistry.getService(ServiceRegistry.java:533) ~[org.eclipse.osgi-3.16.200.jar:?]
	at org.eclipse.osgi.internal.framework.BundleContextImpl.getService(BundleContextImpl.java:656) ~[org.eclipse.osgi-3.16.200.jar:?]
	at org.apache.felix.scr.impl.manager.SingleRefPair.getServiceObject(SingleRefPair.java:88) ~[?:?]
	at org.apache.felix.scr.impl.inject.methods.BindMethod.getServiceObject(BindMethod.java:675) ~[?:?]
	at org.apache.felix.scr.impl.manager.DependencyManager.getServiceObject(DependencyManager.java:2556) ~[?:?]
	at org.apache.felix.scr.impl.manager.DependencyManager.doInvokeBindMethod(DependencyManager.java:2075) ~[?:?]
	at org.apache.felix.scr.impl.manager.DependencyManager.invokeBindMethod(DependencyManager.java:2058) ~[?:?]
	at org.apache.felix.scr.impl.manager.SingleComponentManager.invokeBindMethod(SingleComponentManager.java:443) ~[?:?]
	at org.apache.felix.scr.impl.manager.DependencyManager$MultipleDynamicCustomizer.addedService(DependencyManager.java:333) ~[?:?]
	at org.apache.felix.scr.impl.manager.DependencyManager$MultipleDynamicCustomizer.addedService(DependencyManager.java:301) ~[?:?]
	at org.apache.felix.scr.impl.manager.ServiceTracker$Tracked.customizerAdded(ServiceTracker.java:1200) ~[?:?]
	at org.apache.felix.scr.impl.manager.ServiceTracker$Tracked.customizerAdded(ServiceTracker.java:1121) ~[?:?]
	at org.apache.felix.scr.impl.manager.ServiceTracker$AbstractTracked.trackAdding(ServiceTracker.java:928) ~[?:?]
	at org.apache.felix.scr.impl.manager.ServiceTracker$AbstractTracked.track(ServiceTracker.java:864) ~[?:?]
	at org.apache.felix.scr.impl.manager.ServiceTracker$Tracked.serviceChanged(ServiceTracker.java:1152) ~[?:?]
	at org.apache.felix.scr.impl.BundleComponentActivator$ListenerInfo.serviceChanged(BundleComponentActivator.java:114) ~[?:?]
	at org.eclipse.osgi.internal.serviceregistry.FilteredServiceListener.serviceChanged(FilteredServiceListener.java:120) ~[org.eclipse.osgi-3.16.200.jar:?]
	at org.eclipse.osgi.internal.framework.BundleContextImpl.dispatchEvent(BundleContextImpl.java:957) ~[org.eclipse.osgi-3.16.200.jar:?]
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:234) ~[org.eclipse.osgi-3.16.200.jar:?]
	at org.eclipse.osgi.framework.eventmgr.ListenerQueue.dispatchEventSynchronous(ListenerQueue.java:151) ~[org.eclipse.osgi-3.16.200.jar:?]
	at org.eclipse.osgi.internal.serviceregistry.ServiceRegistry.publishServiceEventPrivileged(ServiceRegistry.java:936) ~[org.eclipse.osgi-3.16.200.jar:?]
	at org.eclipse.osgi.internal.serviceregistry.ServiceRegistry.publishServiceEvent(ServiceRegistry.java:873) ~[org.eclipse.osgi-3.16.200.jar:?]
	at org.eclipse.osgi.internal.serviceregistry.ServiceRegistrationImpl.register(ServiceRegistrationImpl.java:141) ~[org.eclipse.osgi-3.16.200.jar:?]
	at org.eclipse.osgi.internal.serviceregistry.ServiceRegistry.registerService(ServiceRegistry.java:261) ~[org.eclipse.osgi-3.16.200.jar:?]
	at org.eclipse.osgi.internal.framework.BundleContextImpl.registerService(BundleContextImpl.java:496) ~[org.eclipse.osgi-3.16.200.jar:?]
	at org.apache.felix.scr.impl.manager.AbstractComponentManager$3.register(AbstractComponentManager.java:929) ~[?:?]
[... cut ...]
	at java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:484) ~[?:?]
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474) ~[?:?]
	at java.util.stream.ReduceOps$ReduceOp.evaluateSequential(ReduceOps.java:913) ~[?:?]
	at java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234) ~[?:?]
	at java.util.stream.ReferencePipeline.collect(ReferencePipeline.java:578) ~[?:?]
	at org.openhab.binding.zigbee.console.ember.internal.EmberZigBeeConsoleCommandProvider.<init>(EmberZigBeeConsoleCommandProvider.java:63) ~[?:?]
	... 44 more
```

cdjackson, 2021-06-26:
From what I see, I don’t think anything has changed in this area, so probably this is a corner case that we need to understand what’s happening. Please always provide debug log files - as per the binding documentation. Otherwise it’s really not possible to work out what is happening.

cdjackson, 2021-06-26:
> Caused by: java.lang.IllegalStateException: Duplicate key ncpscan (attempted merging values ***@***.*** and ***@***.***)

Hmm - on another look at this error I think I know what might cause this...

kaikreuzer, 2021-06-26:
Thanks for the quick fix, @cdjackson. I have built a new snapshot and can confirm that the issue is fixed in that.

closed 2021-06-26 by kaikreuzer

cdjackson, 2021-06-26:
Thanks for testing Kai. I’m trying to start getting my new system installed over the weekend so now have M1 running (just!) but still got to install some devices...

kpanchen, 2021-06-26:
> Thanks for testing Kai. I’m trying to start getting my new system installed over the weekend so now have M1 running (just!) but still got to install some devices...
> […](#)

Thanks Chris, I'll test it too. If needed debug log can be provided!