openhab/org.openhab.binding.zigbee #660: Unhandled exceptions during bunding init (OH3)

kpanchen, 2021-06-29:
Updated to OH+binding 3.1.0. Unhandled exceptions on binding startup / things discovery:

`2021-06-29 02:24:36.255 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - B0CE18140013DFBC: Starting ZigBee device discovery
2021-06-29 02:24:36.289 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - B0CE18140013C94E: Starting ZigBee device discovery
2021-06-29 02:24:36.291 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - B0CE18140013CDD5: Starting ZigBee device discovery
2021-06-29 02:24:36.292 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - B0CE18140013C0CE: Starting ZigBee device discovery
2021-06-29 02:24:36.292 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - B0CE18140013C2E8: Starting ZigBee device discovery
2021-06-29 02:24:37.198 [WARN ] [zsmartsystems.zigbee.ZigBeeExecutors] - Uncaught exception in thread NotificationService-thread-16
java.lang.ClassCastException: class java.lang.Integer cannot be cast to class java.lang.Long (java.lang.Integer and java.lang.Long are in module java.base of loader 'bootstrap')
	at com.zsmartsystems.zigbee.serialization.DefaultSerializer.appendZigBeeType(DefaultSerializer.java:209) ~[?:?]
	at com.zsmartsystems.zigbee.zcl.field.AttributeReportingConfigurationRecord.serialize(AttributeReportingConfigurationRecord.java:361) ~[?:?]
	at com.zsmartsystems.zigbee.zcl.ZclFieldSerializer.serialize(ZclFieldSerializer.java:38) ~[?:?]
	at com.zsmartsystems.zigbee.zcl.clusters.general.ConfigureReportingCommand.serialize(ConfigureReportingCommand.java:107) ~[?:?]
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.sendCommand(ZigBeeNetworkManager.java:905) ~[?:?]
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.send(ZigBeeTransactionManager.java:443) ~[?:?]
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.sendNextTransaction(ZigBeeTransactionManager.java:681) ~[?:?]
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.transactionComplete(ZigBeeTransactionManager.java:540) ~[?:?]
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransaction.completeTransaction(ZigBeeTransaction.java:435) ~[?:?]
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransaction.transactionStatusReceived(ZigBeeTransaction.java:500) ~[?:?]
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager$3.run(ZigBeeTransactionManager.java:579) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) ~[?:?]
	at java.lang.Thread.run(Thread.java:829) [?:?]
2021-06-29 02:24:37.216 [WARN ] [zsmartsystems.zigbee.ZigBeeExecutors] - Uncaught exception in thread NotificationService-thread-4
java.lang.ClassCastException: class java.lang.Integer cannot be cast to class java.lang.Long (java.lang.Integer and java.lang.Long are in module java.base of loader 'bootstrap')
	at com.zsmartsystems.zigbee.serialization.DefaultSerializer.appendZigBeeType(DefaultSerializer.java:209) ~[?:?]
	at com.zsmartsystems.zigbee.zcl.field.AttributeReportingConfigurationRecord.serialize(AttributeReportingConfigurationRecord.java:361) ~[?:?]
	at com.zsmartsystems.zigbee.zcl.ZclFieldSerializer.serialize(ZclFieldSerializer.java:38) ~[?:?]
	at com.zsmartsystems.zigbee.zcl.clusters.general.ConfigureReportingCommand.serialize(ConfigureReportingCommand.java:107) ~[?:?]
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.sendCommand(ZigBeeNetworkManager.java:905) ~[?:?]
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.send(ZigBeeTransactionManager.java:443) ~[?:?]
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.sendNextTransaction(ZigBeeTransactionManager.java:681) ~[?:?]
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.transactionComplete(ZigBeeTransactionManager.java:540) ~[?:?]
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransaction.completeTransaction(ZigBeeTransaction.java:435) ~[?:?]
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransaction.commandReceived(ZigBeeTransaction.java:387) ~[?:?]
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager$2.run(ZigBeeTransactionManager.java:557) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) ~[?:?]
	at java.lang.Thread.run(Thread.java:829) [?:?]
2021-06-29 02:24:37.433 [ERROR] [iscovery.ZigBeeNodeServiceDiscoverer] - B0CE18140013C94E: Node SVC Discovery: exception: 
java.lang.ClassCastException: class java.lang.Integer cannot be cast to class java.lang.Long (java.lang.Integer and java.lang.Long are in module java.base of loader 'bootstrap')
	at com.zsmartsystems.zigbee.serialization.DefaultSerializer.appendZigBeeType(DefaultSerializer.java:209) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.zcl.field.AttributeReportingConfigurationRecord.serialize(AttributeReportingConfigurationRecord.java:361) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.zcl.ZclFieldSerializer.serialize(ZclFieldSerializer.java:38) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.zcl.clusters.general.ConfigureReportingCommand.serialize(ConfigureReportingCommand.java:107) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.sendCommand(ZigBeeNetworkManager.java:905) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.send(ZigBeeTransactionManager.java:443) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.sendNextTransaction(ZigBeeTransactionManager.java:681) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.queueTransaction(ZigBeeTransactionManager.java:372) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager.sendTransaction(ZigBeeTransactionManager.java:351) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.sendTransaction(ZigBeeNetworkManager.java:2032) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer.requestNetworkAddress(ZigBeeNodeServiceDiscoverer.java:330) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer.requestNetworkAddress(ZigBeeNodeServiceDiscoverer.java:309) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer.access$800(ZigBeeNodeServiceDiscoverer.java:75) ~[bundleFile:?]
	at com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer$NodeServiceDiscoveryTask.run(ZigBeeNodeServiceDiscoverer.java:644) [bundleFile:?]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515) [?:?]
	at java.util.concurrent.FutureTask.run(FutureTask.java:264) [?:?]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304) [?:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) [?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) [?:?]
	at java.lang.Thread.run(Thread.java:829) [?:?]`

Seems fine after, no crashes.

cdjackson, 2021-06-29:
Please provide debug logging for this, and also advise what the devices are that might be causing this.

kpanchen, 2021-06-29:
> Please provide debug logging for this, and also advise what the devices are that might be causing this.

[openhab.log](https://github.com/openhab/org.openhab.binding.zigbee/files/6731185/openhab.log)

Devices are just five sengled E21-N13A bulbs, all the same in the same location. Not added as a thing at the moment.
[image]

Thanks!

cdjackson, 2021-06-29:
Unfortunately I can't really see what is happening with this. There must be an attribute that is incorrectly cast. Normally this happens if a device has an attribute configured incorrectly. The library tries to normalise such things, but there's always the possibility that there's one I've not covered. Unfortunately the log doesn't show this as there's no communications with the device.

I'd suggest to remove the XML for the device (in the "userdata / bridge name" folder) and then start discovery again and see if we can log the issue.

kpanchen, 2021-06-29:
> Unfortunately I can't really see what is happening with this. There must be an attribute that is incorrectly cast. Normally this happens if a device has an attribute configured incorrectly. The [...]

[openhab.log](https://github.com/openhab/org.openhab.binding.zigbee/files/6731955/openhab.log)
This time all five detected fine:

[image]

cdjackson, 2021-06-30:
I really need to see the full debug logs to understand what is happening. Please take a look at the binding documentation on how to configure this - currently there is no zigbee information being logged, and as the error is in the zigbee framework it's not possible to see what's happening if this isn't enabled.

cdjackson, 2021-07-23:
This should be fixed with #662

closed 2021-07-23 by cdjackson