openhab/org.openhab.binding.zigbee #664: NullPointerException when running `openhab:zigbee nodes` in console

scaph0, 2021-07-27:
### Outline

If I issue a `openhab:zigbee nodes` from the console, I receive an unexpected error during execution.

Tried with several firmwares on the coordinator (6.0.3, 6.7.0 or 6.9.1.0).

Tried with or without devices linked to coordinator (TRADFRI bulbs or Aqara TVOC Air Quality Monitor or TS0121 generic plug).

Exception at 17:59:02 below.

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used | POPP ZB-Stick (Zigbee) [rebranded Elecard ELR023]                                                                                 |
| openHAB version  | 3.1.0     |
| Hardware         | Raspberry Pi 3 Model B Rev 1.2                                                                               |
| Memory           | 1GB                                                                               |
| Java version     | zulu11.48.21-ca-jdk11.0.11-linux_aarch32hf                                                                               |
| Devices          | Various or none                                   |

### Logs

```
openhab> openhab:zigbee ncpversion
Ember NCP version 6.9.1.0, EZSP version 8

openhab> openhab:zigbee nodes
An unexpected error occurred during execution.
openhab> log:display
17:58:56.595 [DEBUG] [le.osgi.LoggingCommandSessionListener] - Command: 'log:clear' returned 'null'
17:58:56.843 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue size: 1
17:58:56.845 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=2, ackNum=6, reTx=false, data=32 00 01 18 00]
17:58:56.853 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=6, ackNum=3, reTx=false, data=32 80 01 18 00 02]
17:58:56.854 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - ASH: Frame acked and removed AshFrameData [frmNum=2, ackNum=6, reTx=false, data=32 00 01 18 00]
17:58:56.855 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - ASH took EZSP frame to receive queue. Queue length 0
17:58:56.856 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - ASH added EZSP frame to receive queue. Queue length 1
17:58:56.857 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=7, notRdy=false]
17:58:56.900 [DEBUG] [org.eclipse.jetty.io.IdleTimeout     ] - SocketChannelEndPoint@c2ab86{l=/192.168.32.222:8080,r=/192.168.32.218:63388,OPEN,fill=-,flush=-,to=30000/30000}{io=0/0,kio=0,kro=1}->HttpConnection@1c71b49[p=HttpParser{s=CONTENT,0 of -1},g=HttpGenerator@3afd4e{s=COMMITTED}]=>HttpChannelOverHttp@bc4644{s=HttpChannelState@f42031{s=WAITING rs=ASYNC os=COMMITTED is=IDLE awp=false se=false i=false al=2},r=1,c=false/false,a=WAITING,uri=//openhabian:8080/rest/events/states,age=1651014} idle timeout check, elapsed: 30000 ms, remaining: 0 ms
17:58:56.902 [DEBUG] [org.eclipse.jetty.io.IdleTimeout     ] - SocketChannelEndPoint@c2ab86{l=/192.168.32.222:8080,r=/192.168.32.218:63388,OPEN,fill=-,flush=-,to=30002/30000}{io=0/0,kio=0,kro=1}->HttpConnection@1c71b49[p=HttpParser{s=CONTENT,0 of -1},g=HttpGenerator@3afd4e{s=COMMITTED}]=>HttpChannelOverHttp@bc4644{s=HttpChannelState@f42031{s=WAITING rs=ASYNC os=COMMITTED is=IDLE awp=false se=false i=false al=2},r=1,c=false/false,a=WAITING,uri=//openhabian:8080/rest/events/states,age=1651016} idle timeout expired
17:58:56.904 [DEBUG] [org.eclipse.jetty.io.FillInterest    ] - onFail FillInterest@1dca750{null}
java.util.concurrent.TimeoutException: Idle timeout expired: 30000/30000 ms
        at org.eclipse.jetty.io.IdleTimeout.checkIdleTimeout(IdleTimeout.java:171) ~[?:?]
        at org.eclipse.jetty.io.IdleTimeout.idleCheck(IdleTimeout.java:113) ~[?:?]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515) [?:?]
        at java.util.concurrent.FutureTask.run(FutureTask.java:264) [?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304) [?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) [?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) [?:?]
        at java.lang.Thread.run(Thread.java:829) [?:?]
17:58:56.908 [DEBUG] [org.eclipse.jetty.io.WriteFlusher    ] - ignored: WriteFlusher@d95f55{IDLE}->null
java.util.concurrent.TimeoutException: Idle timeout expired: 30000/30000 ms
        at org.eclipse.jetty.io.IdleTimeout.checkIdleTimeout(IdleTimeout.java:171) ~[?:?]
        at org.eclipse.jetty.io.IdleTimeout.idleCheck(IdleTimeout.java:113) ~[?:?]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515) [?:?]
        at java.util.concurrent.FutureTask.run(FutureTask.java:264) [?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304) [?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) [?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) [?:?]
        at java.lang.Thread.run(Thread.java:829) [?:?]
17:58:56.913 [DEBUG] [org.eclipse.jetty.io.AbstractEndPoint] - Ignored idle endpoint SocketChannelEndPoint@c2ab86{l=/192.168.32.222:8080,r=/192.168.32.218:63388,OPEN,fill=-,flush=-,to=30013/30000}{io=0/0,kio=0,kro=1}->HttpConnection@1c71b49[p=HttpParser{s=CONTENT,0 of -1},g=HttpGenerator@3afd4e{s=COMMITTED}]=>HttpChannelOverHttp@bc4644{s=HttpChannelState@f42031{s=WAITING rs=ASYNC os=COMMITTED is=IDLE awp=false se=false i=false al=2},r=1,c=false/false,a=WAITING,uri=//openhabian:8080/rest/events/states,age=1651027}
17:58:57.847 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue size: 1
17:58:57.848 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=3, ackNum=7, reTx=false, data=33 00 01 18 00]
17:58:57.855 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=7, ackNum=4, reTx=false, data=33 80 01 18 00 02]
17:58:57.857 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - ASH: Frame acked and removed AshFrameData [frmNum=3, ackNum=7, reTx=false, data=33 00 01 18 00]
17:58:57.858 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - ASH added EZSP frame to receive queue. Queue length 1
17:58:57.858 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - ASH took EZSP frame to receive queue. Queue length 0
17:58:57.859 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=0, notRdy=false]
17:58:58.850 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue size: 1
17:58:58.851 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=4, ackNum=0, reTx=false, data=34 00 01 18 00]
[... 94 lines cut ...]
17:59:05.875 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - ASH: TX EZSP queue size: 1
17:59:05.876 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=3, ackNum=7, reTx=false, data=3B 00 01 18 00]
17:59:05.882 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=7, ackNum=4, reTx=false, data=3B 80 01 18 00 02]
17:59:05.883 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - ASH: Frame acked and removed AshFrameData [frmNum=3, ackNum=7, reTx=false, data=3B 00 01 18 00]
17:59:05.884 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - ASH took EZSP frame to receive queue. Queue length 0
17:59:05.884 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - ASH added EZSP frame to receive queue. Queue length 1
17:59:05.885 [DEBUG] [le.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=0, notRdy=false]
17:59:06.333 [DEBUG] [le.osgi.LoggingCommandSessionListener] - Executing command: 'log:display'
17:59:06.337 [DEBUG] [org.apache.felix.configadmin         ] - getConfiguration(pid=org.ops4j.pax.logging, location=null)
17:59:06.339 [DEBUG] [org.apache.felix.configadmin         ] - Found cached configuration org.ops4j.pax.logging bound to ?
17:59:06.357 [DEBUG] [org.apache.felix.configadmin         ] - getConfiguration(pid=org.ops4j.pax.logging, location=null)
17:59:06.358 [DEBUG] [org.apache.felix.configadmin         ] - Found cached configuration org.ops4j.pax.logging bound to ?
[... cut ...]
17:59:06.387 [DEBUG] [org.apache.felix.configadmin         ] - Scheduling task Update: pid=org.ops4j.pax.logging
17:59:06.388 [DEBUG] [org.apache.felix.configadmin         ] - Running task Update: pid=org.ops4j.pax.logging
17:59:06.388 [DEBUG] [org.apache.felix.configadmin         ] - UpdateConfiguration(org.ops4j.pax.logging) scheduled
17:59:06.388 [DEBUG] [ROOT                                 ] - bundle org.apache.felix.scr:2.1.26 (72) configurationEvent: Handling UPDATED of Configuration PID=org.ops4j.pax.logging for component holders []

openhab>
```

cdjackson, 2021-07-27:
I'd need to understand the devices etc in your network as the error is caused by something wrong elsewhere. Possibly there is a corrupt device - it's hard to say without seeing the devices in the XML files.

cdjackson, 2021-07-28:
So from the XML files you've sent, the issue seems to be the following -:

```
      <profileId>41440</profileId>
      <deviceId>97</deviceId>
```

There is a bug in the definition of the ZGP profile ID that I will fix. I'll also prevent this throwing an exception if the device type is not known.

I'll close this as it is not an issue in the OH binding - it's in the underlying libraries.

closed 2021-07-28 by cdjackson