openhab/org.openhab.binding.zigbee #665: Devices added -> All offline

Hoerli1337, 2021-08-01:
Have already described the problem in detail in the OpenHAB forum.
If I add devices with the ZigBee binding, it works, but the devices are completely offline and are not read correctly.

It seems that the XML data is not created correctly, so OpenHAB does not know what it is dealing with.

XML-Example:
```
<ZigBeeNode>
  <ieeeAddress>0011223344556677</ieeeAddress>
  <networkAddress>21917</networkAddress>
  <endpoints/>
  <bindingTable/>
</ZigBeeNode>
```

- Thing + Binding deleted
- OpenHAB stopped
- XML data deleted
- Cache cleared
- OpenHAB restarted
=> problem still exists

```
2021-08-01 14:11:44.415 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 7CB03EAA0A0B9D5D: Starting ZigBee device discovery
2021-08-01 14:11:44.438 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 804B50FFFE5834DA: Starting ZigBee device discovery
2021-08-01 14:11:44.505 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 7CB03EAA0A0C29F7: Starting ZigBee device discovery
2021-08-01 14:11:44.508 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 7CB03EAA0A0B9C86: Starting ZigBee device discovery
2021-08-01 14:11:44.519 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 60A423FFFE93C5C8: Starting ZigBee device discovery
2021-08-01 14:11:44.539 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 086BD7FFFE405E1A: Starting ZigBee device discovery
2021-08-01 14:12:43.851 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 7CB03EAA0A0B9C86: Starting ZigBee device discovery
2021-08-01 14:12:43.852 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 7CB03EAA0A0B9D5D: Starting ZigBee device discovery
2021-08-01 14:12:43.854 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 7CB03EAA0A0C29F7: Starting ZigBee device discovery
2021-08-01 14:12:43.855 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 60A423FFFE93C5C8: Starting ZigBee device discovery
2021-08-01 14:12:43.857 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 804B50FFFE5834DA: Starting ZigBee device discovery
2021-08-01 14:12:47.665 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 7CB03EAA0A0C3E05: Starting ZigBee device discovery
2021-08-01 14:12:47.674 [INFO ] [g.discovery.internal.Persistentbox] - Added new thing 'zigbee:device:ZigBeeUsbStick:7cb03eaa0a0c3e05' to inbox.
2021-08-01 14:12:50.030 [ERROR] [converter.ZigBeeConverterSwitchLevel] - 086BD7FFFE405E1A: Error initialising device
2021-08-01 14:12:50.033 [INFO ] [ng.zigbee.handler.ZigBeeThingHandler] - 086BD7FFFE405E1A: Channel zigbee:device:ZigBeeUsbStick:086bd7fffe405e1a:086BD7FFFE405E1A_1_dimmer failed to initialise device
2021-08-01 14:13:22.086 [ERROR] [converter.ZigBeeConverterSwitchLevel] - 086BD7FFFE405E1A: Error initialising device
2021-08-01 14:13:22.087 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 086BD7FFFE405E1A: Channel zigbee:device:ZigBeeUsbStick:086bd7fffe405e1a:086BD7FFFE405E1A_1_dimmer failed to initialise device
```

https://community.openhab.org/t/openhab-zigbee-binding-funktioniert-nicht-korrekt-openhab-zigbee-binding-does-not-work-correctly/124900/4

- ZigBee coordinator cc2531
- Tested with different devices
- Raspberry Pi 3B+ tested with fresh system
- Tested with my Rasperry Pi 4B (main system running OpenHAB 3 so far)
- The stick is new because my old one broke down

Tested the whole thing with a competitor product (HA) to rule out that the stick is broken. There it works.

cdjackson, 2021-08-01:
> Have already described the problem in detail in the OpenHAB forum.

Please clearly describe the problem here - you may have discussed this on the forum, but it would be good to summarise it here rather than reading many posts that chop and change as you learn new things. This issue should be as self contained as possible.

I would also request a debug log please.

Please also state the devices.

Please also provide the firmware you're using.

> It seems that the XML data is not created correctly, so OpenHAB does not know what it is dealing with.

It's probably the other way around. Since openHAB can't communicate with the devices, it can't generate the XML!

Maybe this is an issue with compatability between the coordinator and the device. Many new devices now require Z3.0 firmware - I've no idea if you have this or not and I'm not familiar with the 2531 firmware. If the device doesn't see the correct security handshake, then it will leave the network. Maybe this is your problem, but I really have very little information on which to help - sorry.

Hoerli1337, 2021-08-02:
Here is the detailed version.

Firmware:
I'm guessing it's firmware v2.12 or v2.50. Bought the stick ready to go so can't say for sure.
The seller provides the download here: https://www.nanocul.de/selfsite.php?aktion=Firmware

I had so far a stick, which has the same firmware.
This has also worked so far with the QS-Zigbee-S05-LN switches and the 3A dimmer.
With Home Assistant, the Ledvance plugs also work.

Devices:
- 1x 3A Smart Inline Dimmer (https://www.amazon.de/Inline-Schalter-intelligente-Lichtdimmer-Hausautomation-Alexa-Sprachsteuerung/dp/B089YPZMBN)
- 4x LEDVANCE Smart+ Plug (https://www.amazon.de/gp/product/B07VCG5G7H)
- 1x Philips Hue Motion Sensor (https://www.amazon.de/Philips-Hue-Bewegungssensor-Aussenbereich-Tageslichtsensor/dp/B07KMP8P1Q)
- 3x QS-Zigbee-S05-LN (https://www.ebay.de/itm/333735471804)

Debug-log:
[openhab-zigbee-debug.log](https://github.com/openhab/org.openhab.binding.zigbee/files/6918688/openhab-zigbee-debug.log)

With a new stick, I now run into the same problems again.

- Stick is recognized and mounted
- After several changes of the config and about 20 restarts of the Raspberry Pi, the binding finally finds the devices again.
=> The setup is easy, scanning for devices is probably already a first big hurdle (Here the problem from 8 months ago: ZigBee (CC2531) + Tuya Switch - OH3)
- Found devices can be created as Thing

The problem is that OpenHAB has no idea what kind of Zigbee-stick I’m using.
- what kind of stick I am using
- the connected devices are at all

Why?
OpenHAB creates XML configurations here: /var/lib/openhab/zigbee/zigbee_coordinator_cc2531_NAME.
I guess these are not filled in correctly.
Luckily I made a config backup of the last working state with my broken stick and have the old XML data on hand.

The new config it USB sticks looks like this:

```
<ZigBeeNode>
  <ieeeAddress>0011223344556677</ieeeAddress>
  <networkAddress>0</networkAddress>
  <endpoints/>
  <bindingTable/>
</ZigBeeNode>
```
(MAC intentionally faked)

The original config actually looks like this:
```
<ZigBeeNode>
  <ieeeAddress>0011223344556677</ieeeAddress>
  <networkAddress>0</networkAddress>
  <nodeDescriptor>
    <apsFlags>0</apsFlags>
    <bufferSize>80</bufferSize>
    <complexDescriptorAvailable>false</complexDescriptorAvailable>
    <manufacturerCode>0</manufacturerCode>
    <logicalType>COORDINATOR</logicalType>
    <serverCapabilities class="sorted-set">
      <ServerCapabilitiesType>PRIMARY_TRUST_CENTER</ServerCapabilitiesType>
    </serverCapabilities>
    <incomingTransferSize>160</incomingTransferSize>
    <outgoingTransferSize>160</outgoingTransferSize>
    <userDescriptorAvailable>false</userDescriptorAvailable>
    <frequencyBands>
      <FrequencyBandType>FREQ_2400_MHZ</FrequencyBandType>
    </frequencyBands>
    <macCapabilities class="sorted-set">
      <MacCapabilitiesType>ALTERNATIVE_PAN</MacCapabilitiesType>
      <MacCapabilitiesType>FULL_FUNCTION_DEVICE</MacCapabilitiesType>
[... 931 lines cut ...]
Have a few log entries here from connecting new devices to the ZigBee stick.
```
2021-08-01 14:11:44.415 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 7CB03EAA0A0B9D5D: Starting ZigBee device discovery
2021-08-01 14:11:44.438 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 804B50FFFE5834DA: Starting ZigBee device discovery
2021-08-01 14:11:44.505 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 7CB03EAA0A0C29F7: Starting ZigBee device discovery
2021-08-01 14:11:44.508 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 7CB03EAA0A0B9C86: Starting ZigBee device discovery
2021-08-01 14:11:44.519 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 60A423FFFE93C5C8: Starting ZigBee device discovery
2021-08-01 14:11:44.539 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 086BD7FFFE405E1A: Starting ZigBee device discovery
2021-08-01 14:12:43.851 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 7CB03EAA0A0B9C86: Starting ZigBee device discovery
2021-08-01 14:12:43.852 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 7CB03EAA0A0B9D5D: Starting ZigBee device discovery
2021-08-01 14:12:43.854 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 7CB03EAA0A0C29F7: Starting ZigBee device discovery
2021-08-01 14:12:43.855 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 60A423FFFE93C5C8: Starting ZigBee device discovery
2021-08-01 14:12:43.857 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 804B50FFFE5834DA: Starting ZigBee device discovery
2021-08-01 14:12:47.665 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 7CB03EAA0A0C3E05: Starting ZigBee device discovery
2021-08-01 14:12:47.674 [INFO ] [g.discovery.internal.Persistentbox] - Added new thing 'zigbee:device:ZigBeeUsbStick:7cb03eaa0a0c3e05' to inbox.
2021-08-01 14:12:50.030 [ERROR] [converter.ZigBeeConverterSwitchLevel] - 086BD7FFFE405E1A: Error initialising device
2021-08-01 14:12:50.033 [INFO ] [ng.zigbee.handler.ZigBeeThingHandler] - 086BD7FFFE405E1A: Channel zigbee:device:ZigBeeUsbStick:086bd7fffe405e1a:086BD7FFFE405E1A_1_dimmer failed to initialise device
2021-08-01 14:13:22.086 [ERROR] [converter.ZigBeeConverterSwitchLevel] - 086BD7FFFE405E1A: Error initialising device
2021-08-01 14:13:22.087 [INFO ] [bee.discovery.ZigBeeDiscoveryService] - 086BD7FFFE405E1A: Channel zigbee:device:ZigBeeUsbStick:086bd7fffe405e1a:086BD7FFFE405E1A_1_dimmer failed to initialise device
```

How can I fix the whole thing again?
It’s slightly humble like that.
I’d like to keep it simple, which the ZigBee binding had managed quite well so far.
Zigbee2Mqtt I therefore do not want to do to myself.

cdjackson, 2021-08-02:
> OpenHAB creates XML configurations here: /var/lib/openhab/zigbee/zigbee_coordinator_cc2531_NAME.
I guess these are not filled in correctly.

The fact that the files are incorrect is an effect, not the cause. The cause seems to be that the devices are not responding (as can be seen in the log). I'm not sure why this is though.

> However, the devices still do not work.

Again - you can't just copy over the XML from your old system and expect this to make any difference. You need to address the underlying cause which appears to be that the devices are not communicating. We can see in the log that commands are being sent, and occasionally some responses are received, but most of the commands do not receive a response.

[image]

Until the communications issue is resolved, the system won't work. If you have a sniffer, you could try sniffing the network to see what is happening. This might provide some more information depending on the root cause of the issue.

Personally I do not recommend the TI sticks and do not use them myself, and neither do any of the commercial users I support so I'm not very familiar with the firmware. Maybe this is a ZB3 issue - I'm not sure

Hoerli1337, 2021-08-03:
The XML data was also only a first test, because I noticed this after several hours of search work.
Could have been by chance the solution.

Unfortunately I have no way to sniff the traffic.
If you follow these instructions (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html), you have to flash the stick again.
But since I do not own anything for this, I can not do that.
Here I have to rely on the OH3 log.

Since it had already worked with another stick and I could operate 7 months without problem two ZigBee devices,
I have the feeling that the problem lies somewhere in OpenHAB.
With Home Assistant it worked with the plugs on the first try.

Which transmitter would you recommend?
I do not want to spend a lot of money for a new device again.
The stick costs between 15 and 20 €, which I still find ok.
The advantage is that the stick is plugged directly into the RPi and I can separate my IoT devices from the normal network.

cdjackson, 2021-08-03:
> I have the feeling that the problem lies somewhere in OpenHAB.

I don't see how? At the moment, the log clearly shows that the devices are not responding - do you know why this is?

Unfortunately the binding does not see all traffic - so we can't see if there is some sort of communications going on - only a network sniffer can see this low level activity.

> Which transmitter would you recommend?

All the companies I support use an Ember chipset. There are a few different ones around - I'd suggest to ask on the forum but I think a few people use the Elelabs stick. The cost is still around the same I believe - €20 approximately I think.

Hoerli1337, 2021-08-03:
Have created here once again a log.
This time also a few times RX are present.
But the stupid is currently, the binding finds a plug now no longer at all.
I have deleted the device from OpenHAB, cleared the cache and also rebooted several times.
The device is still not found again.

My assumption is still that by an update of OpenHAB something went wrong here and the binding no longer works correctly.
I still exclude a hardware problem.
Otherwise I would not have been able to find the devices.
[openhab.log](https://github.com/openhab/org.openhab.binding.zigbee/files/6924336/openhab.log)

cdjackson, 2021-08-03:
This log looks fundamentally the same - there is no response from most devices. Only one device is responding.  I'm not really sure that I can help a lot here - I don't believe there is a bug in the binding, and it is likely to be something to do with your configuration, or possibly the hardware. I know you are "excluding hard problem" but as I've said there is still the possibility that the firmware you have is not compatible with the devices.  As I said earlier though, without a sniffer, we can't see what is happening on the network - the binding is simply passing the commands to the dongle to handle, and these always fail, so the problem is at the network layer - not the application layer that is handled by the binding.

I think we probably need to close this here as I really don't see any bug which is the purpose of github issues. It is better to discuss support issues on the support forum.

Hoerli1337, 2021-08-03:
It can be closed as soon as the problem is solved.
I still don't rule out a binding problem, as very similar problems have existed in the OpenHAB forum for some time.
There is no solution there either.
Maybe someone reads this here who knows the problem and already has a suitable solution in stock.

It's nice of you to help, but I can't give you the information you're looking for, because unfortunately I can't get it.

cdjackson, 2021-08-03:
> It can be closed as soon as the problem is solved.

Ok, then please clearly describe the bug since at the moment I do not see any bug with the binding. Again, this system here is not a general support forum - it's for reporting and resolving software bugs and I can't see that there is a bug here at this point in time.

> I still don't rule out a binding problem

So, as I've said, the binding is sending commands to the device, and the device is not responding. Unless you have further information I'm unable to provide more help - as much as I would like to.

> Maybe someone reads this here who knows the problem 

The problem is that it's impossible to answer this issue since all we know is the device is not responding. Why that is is impossible for me, or others to say. I've made some suggestions as to why this is, but as it seems to be at the network level it's impossible to say without network level information and you can't provide that.

> It's nice of you to help, but I can't give you the information you're looking for, because unfortunately I can't get it.

So what do you propose? You say we can't close this, but you can't provide information on the bug you're reporting. I've suggested that you discuss this on the support forum if you can't clearly define the bug, but you also don't want to do that.

I really want to be able to help, but I have no information on which to help you, so I'm unfortunately at a loss. Without more information it seems pointless to leave this open forever so I propose to close unless you can provide further information to allow me to help you out further.

cdjackson, 2021-08-04:
> Since I already own two of the devices from the beginning and they also worked with the broken stick, I don't see the problem here with the firmware either.

Just one other point here - if it used to work in the past, then this also means it is not a binding issue and it should be something else in the system - firmware, environment (eg RF), devices, distance, etc.

cdjackson, 2021-08-05:
Closing this as there's simply insufficient information to proceed. If you can provide additional information as requested above then please re-open with this information.

closed 2021-08-05 by cdjackson