openhab/org.openhab.binding.zigbee #709: console - zigbee nodes - error

Summer88, 2021-11-22:
### Guidelines

console was showing this output: 

```
openhab-cli console
zigbee nodes 

Network  Addr  IEEE Address      Logical Type  State      EP   Profile                    Device Type                Manufacturer     Model          
      0  0000  000D6F000DD02839  COORDINATOR   ONLINE   
  16068  3EC4  847127FFFE0323E1  END_DEVICE    ONLINE      1  ZIGBEE_HOME_AUTOMATION     TEMPERATURE_SENSOR         _TZ3000_rusu2vzb  TS0201  
```

Then I've added a Legrand Light switch with neutral K4003C from WebUI scan, and started getting errors. 
Device isa Legrand Light switch with neutral K4003C.
Now the zigbee nodes output:

```
openhab> zigbee nodes
An unexpected error occurred during execution.
```

Logs:

```

2021-11-22 11:26:09.051 [ERROR] [b.core.io.console.ConsoleInterpreter] - An error occurred while executing the console command.
java.lang.NullPointerException: null
	at com.zsmartsystems.zigbee.ZigBeeDeviceType.getByValue(ZigBeeDeviceType.java:648) ~[?:?]
	at com.zsmartsystems.zigbee.console.ZigBeeConsoleNodeListCommand.printNode(ZigBeeConsoleNodeListCommand.java:96) ~[?:?]
	at com.zsmartsystems.zigbee.console.ZigBeeConsoleNodeListCommand.process(ZigBeeConsoleNodeListCommand.java:71) ~[?:?]
	at org.openhab.binding.zigbee.console.internal.ZigBeeConsoleCommandExtension.handleZigbeeCommand(ZigBeeConsoleCommandExtension.java:149) ~[?:?]
	at org.openhab.binding.zigbee.console.internal.ZigBeeConsoleCommandExtension.handleCommand(ZigBeeConsoleCommandExtension.java:117) ~[?:?]
	at org.openhab.binding.zigbee.console.internal.ZigBeeConsoleCommandExtension.execute(ZigBeeConsoleCommandExtension.java:89) ~[?:?]
	at org.openhab.core.io.console.ConsoleInterpreter.execute(ConsoleInterpreter.java:55) [bundleFile:?]
	at org.openhab.core.io.console.karaf.internal.CommandWrapper.execute(CommandWrapper.java:78) [bundleFile:?]
	at org.apache.karaf.shell.impl.console.osgi.secured.SecuredCommand.execute(SecuredCommand.java:68) [bundleFile:4.3.2]
	at org.apache.karaf.shell.impl.console.osgi.secured.SecuredCommand.execute(SecuredCommand.java:86) [bundleFile:4.3.2]
	at org.apache.felix.gogo.runtime.Closure.executeCmd(Closure.java:599) [bundleFile:4.3.2]
	at org.apache.felix.gogo.runtime.Closure.executeStatement(Closure.java:526) [bundleFile:4.3.2]
	at org.apache.felix.gogo.runtime.Closure.execute(Closure.java:415) [bundleFile:4.3.2]
	at org.apache.felix.gogo.runtime.Pipe.doCall(Pipe.java:416) [bundleFile:4.3.2]
	at org.apache.felix.gogo.runtime.Pipe.call(Pipe.java:229) [bundleFile:4.3.2]
	at org.apache.felix.gogo.runtime.Pipe.call(Pipe.java:59) [bundleFile:4.3.2]
	at java.util.concurrent.FutureTask.run(FutureTask.java:264) [?:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) [?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) [?:?]
	at java.lang.Thread.run(Thread.java:829) [?:?]
```

```
log:set debug org.openhab.binding.zigbee
log:set debug com.zsmartsystems.zigbee
```

---

### Outline

Describe your issue. Try and be clear and avoid jumping to conclusions about the cause - instead provide full information about the issue, what you did and what happened. Cover what happened, and what you think should have happened.

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used |                 Ember EM35x NCP - BV 2010/10                                                               |
| openHAB version  | OpenHabian v.3.1.0 Release build    |
| Hardware         |                    Raspberry Pi 4 Model B Rev 1.2                                                         |
| Memory           |                   4GB                                                             |
| Java version     |                           openjdk 11.0.13 2021-10-19 LTS
[... 98 lines cut ...]
2021-11-22 11:35:01.024 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX STA: msgTag=2A state=RX_NAK
2021-11-22 11:35:01.024 [DEBUG] [transaction.ZigBeeTransactionManager] - notifyTransactionProgress: TID=2A, state=RX_NAK, outstanding=1
2021-11-22 11:35:01.335 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingRouteErrorHandler [networkId=0, status=EMBER_MAC_INDIRECT_TIMEOUT, target=3EC4]
2021-11-22 11:35:03.812 [DEBUG] [transaction.ZigBeeTransactionManager] - Transaction complete: ZigBeeTransaction [ieeeAddress=847127FFFE0323E1 queueTime=60043, state=FAILED, sendCnt=2, command=BindRequest [0000/0 -> 3EC4/0, cluster=0021, TID=2B, srcAddress=847127FFFE0323E1, srcEndpoint=1, bindCluster=1026, dstAddrMode=3, dstAddress=000D6F000DD02839, dstEndpoint=1]]
2021-11-22 11:35:03.813 [DEBUG] [transaction.ZigBeeTransactionManager] - transactionListenerRemoved: 0 outstanding
2021-11-22 11:35:03.814 [DEBUG] [e.transaction.ZigBeeTransactionQueue] - 847127FFFE0323E1: transactionComplete, state=FAILED, outstanding=0
2021-11-22 11:35:03.815 [DEBUG] [e.transaction.ZigBeeTransactionQueue] - 847127FFFE0323E1: transactionComplete exceeded max retries 2
2021-11-22 11:35:03.816 [DEBUG] [zigbee.transaction.ZigBeeTransaction] - Transaction cancelled: ZigBeeTransaction [ieeeAddress=847127FFFE0323E1 queueTime=60046, state=CANCELLED, sendCnt=2, command=BindRequest [0000/0 -> 3EC4/0, cluster=0021, TID=2B, srcAddress=847127FFFE0323E1, srcEndpoint=1, bindCluster=1026, dstAddrMode=3, dstAddress=000D6F000DD02839, dstEndpoint=1]]
2021-11-22 11:35:03.817 [DEBUG] [converter.ZigBeeConverterTemperature] - 847127FFFE0323E1: Failed to bind temperature measurement cluster
2021-11-22 11:35:03.817 [DEBUG] [transaction.ZigBeeTransactionManager] - Transaction complete: ZigBeeTransaction [ieeeAddress=847127FFFE0323E1 queueTime=60048, state=CANCELLED, sendCnt=2, command=BindRequest [0000/0 -> 3EC4/0, cluster=0021, TID=2B, srcAddress=847127FFFE0323E1, srcEndpoint=1, bindCluster=1026, dstAddrMode=3, dstAddress=000D6F000DD02839, dstEndpoint=1]]
2021-11-22 11:35:03.818 [DEBUG] [transaction.ZigBeeTransactionManager] - transactionListenerRemoved: 0 outstanding
2021-11-22 11:35:03.818 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - 847127FFFE0323E1: Initializing channel zigbee:device:01380FB1:847127fffe0323e1:847127FFFE0323E1_1_batteryalarm with org.openhab.binding.zigbee.internal.converter.ZigBeeConverterBatteryAlarm@969fe4
2021-11-22 11:35:03.819 [DEBUG] [e.transaction.ZigBeeTransactionQueue] - 847127FFFE0323E1: transactionComplete but not outstanding, state=CANCELLED, outstanding=0
2021-11-22 11:35:03.819 [DEBUG] [onverter.ZigBeeConverterBatteryAlarm] - 847127FFFE0323E1: Initialising device battery alarm converter
2021-11-22 11:35:03.819 [DEBUG] [transaction.ZigBeeTransactionManager] - Transaction Manager: Send Next transaction. outstandingTransactions=0, outstandingQueues=1, sleepy=0/3
2021-11-22 11:35:03.821 [DEBUG] [transaction.ZigBeeTransactionManager] - 3EC4/0: Sending ZigBeeTransaction [ieeeAddress=847127FFFE0323E1 queueTime=40028, state=WAITING, sendCnt=0, command=BindRequest [0000/0 -> 3EC4/0, cluster=0021, TID=2C, srcAddress=847127FFFE0323E1, srcEndpoint=1, bindCluster=1029, dstAddrMode=3, dstAddress=000D6F000DD02839, dstEndpoint=1]]
2021-11-22 11:35:03.821 [DEBUG] [transaction.ZigBeeTransactionManager] - transactionListenerAdded: 1 outstanding
2021-11-22 11:35:03.822 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: BindRequest [0000/0 -> 3EC4/0, cluster=0021, TID=2C, srcAddress=847127FFFE0323E1, srcEndpoint=1, bindCluster=1029, dstAddrMode=3, dstAddress=000D6F000DD02839, dstEndpoint=1]
2021-11-22 11:35:03.824 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0000/0, destinationAddress=3EC4/0, profile=0000, cluster=0021, addressMode=DEVICE, radius=8, apsSecurity=false, ackRequest=true, apsCounter=39, rssi=--, lqi=--, payload=2C E1 23 03 FE FF 27 71 84 01 05 04 03 39 28 D0 0D 00 6F 0D 00 01]
2021-11-22 11:35:03.824 [DEBUG] [transaction.ZigBeeTransactionManager] - Transaction Manager: Send Next transaction. outstandingTransactions=1, outstandingQueues=1, sleepy=0/3
2021-11-22 11:35:03.826 [DEBUG] [e.transaction.ZigBeeTransactionQueue] - 847127FFFE0323E1: Added transaction to queue, len=2, transaction=ZigBeeTransaction [ieeeAddress=847127FFFE0323E1 queueTime=0, state=WAITING, sendCnt=0, command=BindRequest [0000/0 -> 3EC4/0, cluster=0021, TID=--, srcAddress=847127FFFE0323E1, srcEndpoint=1, bindCluster=1, dstAddrMod [... cut ...]
2021-11-22 11:35:03.827 [DEBUG] [transaction.ZigBeeTransactionManager] - Transaction Manager: Send Next transaction. outstandingTransactions=1, outstandingQueues=1, sleepy=0/3
2021-11-22 11:35:03.839 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspSendUnicastResponse [networkId=0, status=EMBER_SUCCESS, sequence=99]
2021-11-22 11:35:03.870 [DEBUG] [transaction.ZigBeeTransactionManager] - Transaction Manager: Send Next transaction. outstandingTransactions=1, outstandingQueues=1, sleepy=0/3
```

cdjackson, 2021-11-22:
Please ensure you are using the latest binding as this is already resolved.

closed 2021-11-22 by cdjackson

Summer88, 2021-11-23:
I've upgraded to Milestone 3.2.0.M4 and it's working.
Thank you, BR