openhab/org.openhab.binding.zigbee #721: Firmware upgrade the Bitronvideo BV 2010/10 ZigBee USB dongle

jogethome, 2021-12-29:
I'm experiencing similar issues like  
https://community.openhab.org/t/firmware-upgrade-the-bitronvideo-bv-2010-10-zigbee-usb-dongle/128879

After successful upgrading the firmware of the stick to 6.7.8 those things which were already bound before
were discovered after a while. 
It was unpossible to add new things. Having the stick running on 115200baud the stick switched back to 57600baud during scan.
Afterwards it got in unresponsive error state. Switching it back to 115200baud brings it back online.
This was the case for stable version. 
Afterwards I switched to Test / Snapshot versions in the hope getting rid of this. Unfortunately the discovered things are now in state "gone" and the scan has the same behaviour as above.
I tried now the last two snapshots.
What I also unsuccessfully tried is putting the baudrate fix into a things definition file.

In the log file of openhab I get a Java stack trace like 
2021-12-28 21:57:55.792 [WARN ] [mmon.WrappedScheduledExecutorService] - Scheduled runnable ended with an exception:
java.lang.NullPointerException: null
        at org.openhab.binding.zigbee.ember.handler.EmberHandler.setGroupRegistration(EmberHandler.java:365) ~[?:?]
        at org.openhab.binding.zigbee.ember.handler.EmberHandler.initializeDongleSpecific(EmberHandler.java:120) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler.initialiseZigBee(ZigBeeCoordinatorHandler.java:551) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler.lambda$2(ZigBeeCoordinatorHandler.java:559) ~[?:?]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515) ~[?:?]
        at java.util.concurrent.FutureTask.run(FutureTask.java:264) ~[?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304) ~[?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) [?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) [?:?]
        at java.lang.Thread.run(Thread.java:829) [?:?]

Thanks & Regards
Georg

## Steps to Reproduce (for Bugs)
1. Set Zigbee Config to 115200baud & Save Setting
2. Start Scan for new devices

## Your Environment

Openhab Version:     3.3.0-SNAPSHOT (#2662) (same/similar issue with snapshot before and latest test/stable versions)
Zigbee BV 2010/10 Stick with walthowd/husbzb ember Firmware 6.7.8 from
https://github.com/grobasoz/zigbee-firmware/tree/af7c35ea8d580152eb9853af1d3fab91bef3b5d4/EM3587

jogethome, 2021-12-29:
Hi,
it seems to be that I found a workaround.
I did a couple of steps meanwhile (reflashing, deletion, cache-clean ....).
However the last step seems to solve the issue.
I did set up the ZIGBEE Controller manually and ignored the automatically discovered one.
Afterwards Zigbee network scanning works, also with 115200baud (for now).
I'll post later if any further error pops up.

porst17, 2021-12-29:
I am getting the exact same error message in the log for my BV 2010/10 dongle. However, I did not upgrade the firmware. It is still the one that it was sold with: 5.8.0.0

The dongle was working well a couple of days ago, but suddenly, I am unable to discover new things. Things that I added previously kept working, but scanning for new devices never yields a result. The above error message appears every time I restart OpenHab or when I change one of the dongle's settings (like e.g. Power Mode).

I am running a clean install of OpenHab 3.2.0 with just the ZigBee binding 3.2.0 installed. 

What bothers me is that discovery was working a couple of days ago and now stopped working for no apparent reason. Are there any persistent settings in the dongle that survive a power cycle and a re-installation of OpenHab?

@jogethome Does "cache-clean" mean cleaning the dongles cache or the OpenHab cache? It tried the latter, but it did not change anything. What's the procedure for a "cache-clean"?

RayMYP, 2021-12-29:
Just to confirm the exception is not 6.7.8 firmware specific. I'm running stable Openhab 3.2 with a non patched Zigbee BV 2010/10 Stick on original firmware 5.8.0 (currently reviewing upgrade options but it's my only stick). The `Thing` is setup in the UI and has a 57600 baud.

During Openhab start up I always see the same:
```
2021-12-29 14:12:48.254 [WARN ] [mmon.WrappedScheduledExecutorService] - Scheduled runnable ended with an exception: 
java.lang.NullPointerException: null
	at org.openhab.binding.zigbee.ember.handler.EmberHandler.setGroupRegistration(EmberHandler.java:365) ~[?:?]
	at org.openhab.binding.zigbee.ember.handler.EmberHandler.initializeDongleSpecific(EmberHandler.java:120) ~[?:?]
	at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler.initialiseZigBee(ZigBeeCoordinatorHandler.java:551) ~[?:?]
	at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler.lambda$2(ZigBeeCoordinatorHandler.java:559) ~[?:?]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515) ~[?:?]
	at java.util.concurrent.FutureTask.run(FutureTask.java:264) ~[?:?]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304) ~[?:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) [?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) [?:?]
	at java.lang.Thread.run(Thread.java:829) [?:?]
```

Not sure about discovery of new devices as don't have any to search for at the moment.

jogethome, 2021-12-29:
to answer the question about cache clean completely, here is the procedure at a raspberry pi:
sudo systemctl stop openhab
sudo openhab-cli clean-cache
sudo systemctl start openhab

jogethome, 2021-12-29:
The java stacktrace is still there 
2021-12-29 17:28:35.367 [WARN ] [mmon.WrappedScheduledExecutorService] - Scheduled runnable ended with an exception:
java.lang.NullPointerException: null
        at org.openhab.binding.zigbee.ember.handler.EmberHandler.setGroupRegistration(EmberHandler.java:365) ~[?:?]
        at org.openhab.binding.zigbee.ember.handler.EmberHandler.initializeDongleSpecific(EmberHandler.java:120) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler.initialiseZigBee(ZigBeeCoordinatorHandler.java:551) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler.lambda$2(ZigBeeCoordinatorHandler.java:559) ~[?:?]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515) ~[?:?]
        at java.util.concurrent.FutureTask.run(FutureTask.java:264) ~[?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304) ~[?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) [?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) [?:?]
        at java.lang.Thread.run(Thread.java:829) [?:?]

obviously unimportant relating to my problem.

porst17, 2021-12-29:
After cleaning the OH cache, nothing changed. The error is still in the log and discovery still fails. I can't tell if the two things are related. The NPE should be fixed regardless.

cdjackson, 2021-12-29:
I will provide a fix for this NPE shortly.

cdjackson, 2021-12-29:
This should now be resolved in #722

closed 2021-12-29 by cdjackson

jogethome, 2021-12-29:
Thanks Chris, from what I see you are making a great job here!