openhab/org.openhab.binding.zigbee #729: Add special cluster attributes for Tuya TS130f cover switch

FranzSchi, 2022-01-13:
### Guidelines

Delete this section in your issue...

Please add the special cluster attributes for Tuya TS130f cover switch (And Rollershutter) as it is discussed here to allow the calibratoin mode:

https://community.openhab.org/t/ts130f-rollershutter-controller-via-zigbee-binding/107687/5

Details should be specified here:
[[[I added special Tuya attributes to cluster 'genOnOff' and 'closuresWindowCovering' for support of the Tuya TS130f RollerShutter switch. Discussed at Koenkk/zigbee2mqtt#4257](https://github.com/Koenkk/zigbee-herdsman/pull/240)](https://github.com/Koenkk/zigbee-herdsman/pull/240)](https://github.com/Koenkk/zigbee-herdsman/pull/240)

```
openhab> zigbee endpoint 20561/1
IEEE Address      : A4C1382E683CB5F8
Network Address   : 20561
Endpoint          : 1
Device Profile    : 0x0104, ZIGBEE_HOME_AUTOMATION
Device Type       : 0x0202, WINDOW_COVERING_DEVICE
Device Version    : 1
Input Clusters    : (Server)
   0000 Basic
     - APS Security disabled
        S       0 r-- UNSIGNED_8_BIT_INTEGER    ZCL Version                              Fri Jan 07 12:29:57 CET 2022 3
        S       1 r-- UNSIGNED_8_BIT_INTEGER    Application Version                      Fri Jan 07 12:29:57 CET 2022 67
        S       2 r-- UNSIGNED_8_BIT_INTEGER    Stack Version                            Fri Jan 07 12:29:57 CET 2022 0
        S       3 r-- UNSIGNED_8_BIT_INTEGER    HW Version                               Fri Jan 07 12:29:57 CET 2022 1
        S       4 r-- CHARACTER_STRING          Manufacturer Name                        Fri Jan 07 12:29:57 CET 2022 _TZ3000_dbpmpco1
        S       5 r-- CHARACTER_STRING          Model Identifier                         Fri Jan 07 12:29:57 CET 2022 TS130F
        S       6 r-- CHARACTER_STRING          Date Code                                Fri Jan 07 12:29:57 CET 2022
        S       7 r-- ENUMERATION_8_BIT         Power Source                             Fri Jan 07 18:44:18 CET 2022 1
        S       8 r-- ENUMERATION_8_BIT         Generic Device Class
        S       9 r-- ENUMERATION_8_BIT         Generic Device Type
        S      10 r-- CHARACTER_STRING          Product Code
        S      11 r-- CHARACTER_STRING          Product URL
        S      16 rw- CHARACTER_STRING          Location Description
        S      17 rw- ENUMERATION_8_BIT         Physical Environment
        S      18 rw- BOOLEAN                   Device Enabled
        S      19 rw- BITMAP_8_BIT              Alarm Mask
        S      20 rw- BITMAP_8_BIT              Disable Local Config
        S   16384 r-- CHARACTER_STRING          SW Build ID
   0004 Groups
     - APS Security disabled
        S       0 r-- BITMAP_8_BIT              Name Support
   0005 Scenes
     - APS Security disabled
        S       0 r-- UNSIGNED_8_BIT_INTEGER    Scene Count
        S       1 r-- UNSIGNED_8_BIT_INTEGER    Current Scene
        S       2 r-- UNSIGNED_16_BIT_INTEGER   Current Group
        S       3 r-- BOOLEAN                   Scene Valid
        S       4 r-- BITMAP_8_BIT              Name Support
        S       5 r-- IEEE_ADDRESS              Last Configured By
   0006 On/Off
     - APS Security disabled
        U       0 r-s BOOLEAN                   On Off
        U   16384 r-- BOOLEAN                   Global Scene Control
        S   16385 rw- UNSIGNED_16_BIT_INTEGER   On Time
        S   16386 rw- UNSIGNED_16_BIT_INTEGER   Off Wait Time
        U   16387 rw- ENUMERATION_8_BIT         Start Up On Off
   0102 Window Covering
     - APS Security disabled
        S       0 r-- ENUMERATION_8_BIT         Window Covering Type
        S       1 r-- UNSIGNED_16_BIT_INTEGER   Physical Closed Limit - Lift
        S       2 r-- UNSIGNED_16_BIT_INTEGER   Physical Closed Limit - Tilt
        S       3 r-- UNSIGNED_16_BIT_INTEGER   Current Position - Lift
        S       4 r-- UNSIGNED_16_BIT_INTEGER   Current Position - Tilt
        S       5 r-- UNSIGNED_16_BIT_INTEGER   Number Of Actuations - Lift
        S       6 r-- UNSIGNED_16_BIT_INTEGER   Number Of Actuations - Tilt
        S       7 r-- BITMAP_8_BIT              Config Status
        S       8 r-- UNSIGNED_8_BIT_INTEGER    Current Position Lift Percentage         Fri Jan 07 19:17:53 CET 2022 100
        S       9 r-- UNSIGNED_8_BIT_INTEGER    Current Position Tilt Percentage
        S      16 r-- UNSIGNED_16_BIT_INTEGER   Installed Open Limit - Lift
        S      17 r-- UNSIGNED_16_BIT_INTEGER   Installed Closed Limit - Lift
        S      18 r-- UNSIGNED_16_BIT_INTEGER   Installed Open Limit - Tilt
        S      19 r-- UNSIGNED_16_BIT_INTEGER   Installed Closed Limit - Tilt
        S      20 rws UNSIGNED_16_BIT_INTEGER   Velocity - Lift
        S      21 rws UNSIGNED_16_BIT_INTEGER   Acceleration Time - Lift
        S      22 rws UNSIGNED_16_BIT_INTEGER   Deceleration Time - Lift
        S      23 rws BITMAP_8_BIT              Mode
        S      24 rws OCTET_STRING              Intermediate Setpoints - Lift
        S      25 rws OCTET_STRING              Intermediate Setpoints - Tilt
Output Clusters   : (Client)
   000A Time
     - APS Security disabled
   0019 Ota Upgrade
     - APS Security disabled
        S       0 r-- IEEE_ADDRESS              Upgrade Server ID
        S       1 r-- UNSIGNED_32_BIT_INTEGER   File Offset
        S       2 r-- UNSIGNED_32_BIT_INTEGER   Current File Version                     Fri Jan 07 12:29:57 CET 2022 67
        S       3 r-- UNSIGNED_16_BIT_INTEGER   Current ZigBee Stack Version
        S       4 r-- UNSIGNED_32_BIT_INTEGER   Downloaded File Version
        S       5 r-- UNSIGNED_16_BIT_INTEGER   Downloaded ZigBee Stack Version
        S       6 r-- ENUMERATION_8_BIT         Image Upgrade Status
        S       7 r-- UNSIGNED_16_BIT_INTEGER   Manufacturer ID
        S       8 r-- UNSIGNED_16_BIT_INTEGER   Image Type ID
        S       9 r-- UNSIGNED_16_BIT_INTEGER   Minimum Block Request Period
        S      10 r-- UNSIGNED_32_BIT_INTEGER   Image Stamp
Groups Supported  : 32
Groups Configured : 0
```

### Outline

We are not able to run the calibration mode.
[image]

### Configuration

Ember Coordinator

### Logs

Here is the XML of the device detection
[A4C1382E683CB5F8.zip](https://github.com/openhab/org.openhab.binding.zigbee/files/7863598/A4C1382E683CB5F8.zip)

cdjackson, 2022-01-13:
Please can you define the attributes required to be supported.

FranzSchi, 2022-01-13:
Hi Chris,

thanks for your fast reply, I really appreciate that.

I guess we need this properties:

			tuyaMovingState: { ID: 0xf000, type: dataType_1.default.enum8 }, - Not so critical
 			tuyaCalibration: { ID: 0xf001, type: dataType_1.default.enum8 }, - very much needed
 			tuyaMotorReversal: { ID: 0xf002, type: dataType_1.default.enum8 },     - would be nice

Cluster | CMD | Attributes | Data Type | Value | Description

(0x0102) | Write Attributes (0x02) | 0xf002 | 8-Bit (0x30) | 1 (0x01) | Motor reversal on
(0x0102) | Write Attributes (0x02) | 0xf002 | 8-Bit (0x30) | 0 (0x00) | Motor reversal off
(0x0102) | Write Attributes (0x02) | 0xf001 | 8-Bit (0x30) | 0 (0x00) | Start Calibration
(0x0102) | Write Attributes (0x02) | 0xf001 | 8-Bit (0x30) | 1 (0x01) | End Calibration
(0x0006) | Write Attributes (0x02) | 0x8001 | 8-Bit (0x30) | 0 (0x00) | Light Mode  1
(0x0006) | Write Attributes (0x02) | 0x8001 | 8-Bit (0x30) | 1 (0x01) | Light Mode  2
(0x0006) | Write Attributes (0x02) | 0x8001 | 8-Bit (0x30) | 2 (0x02) | Light Mode  3

(0x0102) | Command (0x00) |   |   |   | open
(0x0102) | Command (0x01) |   |   |   | close
(0x0102) | Command (0x02) |   |   |   | stopp

(0x0102) | Report Attributes (0x0a) | 0x0008 | 8-Bit (0x20) | 0 (0x00) | Position 0%
(0x0102) | Report Attributes (0x0a) | 0x0008 | 8-Bit (0x20) | 100 (0x64) | Position 100%
(0x0102) | Report Attributes (0x0a) | 0xf000 | 8-Bit (0x30) | 0 (0x00) | Moving up
(0x0102) | Report Attributes (0x0a) | 0xf000 | 8-Bit (0x30) | 1 (0x01) | Stopp
(0x0102) | Report Attributes (0x0a) | 0xf000 | 8-Bit (0x30) | 2 (0x02) | Moving down

May Greetings
/Franz

FranzSchi, 2022-01-14:
Hi,

just as an info for the others, I've just calibrated my RollerShutter.
Replace "20561/1" with your node ID

Start Calibration:
openhab> zigbee write 20561/1 258 61441 0 ENUMERATION_8_BIT
Attribute value write success.

End Calibration:
openhab> zigbee write 20561/1 258 61441 1 ENUMERATION_8_BIT
Attribute value write success.

Follow this Procedure:
https://www.zigbee2mqtt.io/devices/TS130F.html

BR
/Franz

FranzSchi, 2026-01-18:
This one can be closed as its just for the calibration process

closed 2026-01-18 by FranzSchi