openhab/org.openhab.binding.zigbee #730: Color Temperature channel not created on Tuya TS0505B

Rosen01, 2022-01-21:
### Outline

The channel to control the color temperature (cold/warm white) is not created.
[image]
The color channel is working well.

Info on the device from [Zigbee2MQTT](https://www.zigbee2mqtt.io/devices/TS0505B.html).

### Configuration

OpenHAB: 3.2.0
Coordinator: Ember Coordinator (Elelabs Zigbee adapter ELR023 Firmware 6.10.3.0)
Hardware: Raspberry pi 3 B

### Logs

`Binding table for node 750 [A4C138A4B750E0DC]
Src Address          | Dest Address         | Group | Mode    | Cluster
A4C138A4B750E0DC/1   | 84FD27FFFE0E4497/1   |       | Address | 0300:COLOR_CONTROL
A4C138A4B750E0DC/1   | 84FD27FFFE0E4497/1   |       | Address | 0006:ON_OFF
A4C138A4B750E0DC/1   | 84FD27FFFE0E4497/1   |       | Address | 0008:LEVEL_CONTROL

|>| Node Descriptor
| |> Logical Type               ROUTER
| |> MAC Capabilities           [FULL_FUNCTION_DEVICE, MAINS_POWER, RECEIVER_ON_WHEN_IDLE]
| |> Stack Compliance           21
| |> Server Capabilities        []
| |> Buffer Size                66
| |> Incoming Transfer Size     66
| |> Outgoing Transfer Size     66
|
|>| Power Descriptor
| |> Available Power Sources    [MAINS]
| |> Current Power Source       MAINS
| |> Current Power Mode         RECEIVER_ON_IDLE
| |> Power Level                FULL
|
|>| ZDO
| |> ManagementBindRequest      SUCCESS
| |> IeeeAddressRequest         SUCCESS
| |> ManagementLqiRequest       SUCCESS
| |> ManagementRoutingRequest   TIMEOUT
|
|>| Basic Information
| |> Generic Device Class
| |> Generic Device Type
| |> Manufacturer Name          _TZ3210_k1pe6ibm
| |> Model Indentifier          TS0505B
| |> Product Code
| |> Product URL
| |> Date Code
| |> Application Version        65
| |> Software Build ID
| |> Hardware Version           1
| |> Zcl Version                3
| |> Stack Version              0
| |
| |>| Endpoint 1
| | |> Profile                  0104  ZIGBEE_HOME_AUTOMATION
| | |> Device Type              010D  EXTENDED_COLOR_LIGHT
| | |> Device Version           1
| | |
| | |>| Input Clusters
| | | |
| | | |>| Cluster 0000 Basic
| | | | |> Type                 Server [Input]
| | | | |> Manufacturer Spec.   No
| | | | |
| | | | |>| Commands Generated
| | | | | |> FAILURE
| | | | |
| | | | |>| Commands Received
| | | | |> FAILURE
| | | | |
| | | | |>| Attributes Supported
| | | | | |> 0000 ZCL Version
| | | | | |> 0001 Application Version
| | | | | |> 0002 Stack Version
| | | | | |> 0003 HW Version
| | | | | |> 0004 Manufacturer Name
| | | | | |> 0005 Model Identifier
| | | | | |> 0006 Date Code
| | | | | |> 0007 Power Source
| | | | | |> FFDE Unknown
| | | | | |> FFFD Unknown
| | | | | |> FFFE Unknown
| | | |
| | | |>| Cluster 0003 Identify
| | | | |> Type                 Server [Input]
| | | | |> Manufacturer Spec.   No
| | | | |> Unsupported locally
| | | |
| | | |>| Cluster 0004 Groups
| | | | |> Type                 Server [Input]
| | | | |> Manufacturer Spec.   No
| | | | |> Unsupported locally
| | | |
| | | |>| Cluster 0005 Scenes
| | | | |> Type                 Server [Input]
| | | | |> Manufacturer Spec.   No
| | | | |
| | | | |>| Commands Generated
| | | | | |> FAILURE
| | | | |
| | | | |>| Commands Received
| | | | |> FAILURE
| | | | |
| | | | |>| Attributes Supported
| | | | | |> 0000 Scene Count
| | | | | |> 0001 Current Scene
| | | | | |> 0002 Current Group
| | | | | |> 0003 Scene Valid
| | | | | |> 0004 Name Support
| | | | | |> FFFD Unknown
| | | |
| | | |>| Cluster 0006 On/Off
| | | | |> Type                 Server [Input]
| | | | |> Manufacturer Spec.   No
| | | | |
| | | | |>| Commands Generated
| | | | | |> FAILURE
| | | | |
| | | | |>| Commands Received
| | | | |> FAILURE
| | | | |
| | | | |>| Attributes Supported
| | | | | |> 0000 On Off
| | | | | |> F000 Unknown
| | | | | |> FFFD Unknown
| | | |
| | | |>| Cluster 0008 Level Control
| | | | |> Type                 Server [Input]
| | | | |> Manufacturer Spec.   No
| | | | |
| | | | |>| Commands Generated
| | | | | |> FAILURE
| | | | |
| | | | |>| Commands Received
| | | | |> FAILURE
| | | | |
| | | | |>| Attributes Supported
| | | | | |> 0000 Current Level
| | | | | |> 000F Options
| | | | | |> F000 Unknown
| | | | | |> FFFD Unknown
| | | |
| | | |>| Cluster 0300 Color Control
| | | | |> Type                 Server [Input]
| | | | |> Manufacturer Spec.   No
| | | | |> Unsupported locally
| | | |
| | | |>| Cluster 1000
| | | | |> Type                 Server [Input]
| | | | |> Manufacturer Spec.   No
| | | | |> Unsupported locally
| | | |
| | | |>| Cluster EF00
| | | | |> Type                 Server [Input]
| | | | |> Manufacturer Spec.   No
| | | | |> Unsupported locally
| | |
| | |>| Output Clusters
| | | |
| | | |>| Cluster 000A Time
| | | | |> Type                 Client [Output]
| | | | |> Manufacturer Spec.   No
| | | | |> Unsupported locally
| | | |
| | | |>| Cluster 0019 Ota Upgrade
| | | | |> Type                 Client [Output]
| | | | |> Manufacturer Spec.   No
| | | | |
| | | | |>| Commands Generated
| | | | | |> FAILURE
| | | | |
| | | | |>| Commands Received
| | | | |> FAILURE
| | | | |
| | | | |>| Attributes Supported
| | | | | |> 0000 Upgrade Server ID
| | | | | |> 0001 File Offset
| | | | | |> 0002 Current File Version
| | | | | |> 0003 Current ZigBee Stack Version
| | | | | |> 0004 Downloaded File Version
| | | | | |> 0005 Downloaded ZigBee Stack Version
| | | | | |> 0006 Image Upgrade Status
| | | | | |> 0007 Manufacturer ID
| | | | | |> 0008 Image Type ID
| | | | | |> 0009 Minimum Block Request Period
| | | | | |> FFFD Unknown
| |
| |>| Endpoint 242
| | |> Profile                  A1E0  ZIGBEE_GREEN_POWER
| | |> Device Type              0061
| | |> Device Version           0`

[openhab.log](https://github.com/openhab/org.openhab.binding.zigbee/files/7916285/openhab.log)
[A4C138A4B750E0DC.zip](https://github.com/openhab/org.openhab.binding.zigbee/files/7916292/A4C138A4B750E0DC.zip)

cdjackson, 2022-01-21:
The problem is likely caused by the following response from the device -:

```
DiscoverAttributesResponse [Color Control: C211/1 -> 0000/1, cluster=0300, TID=9E, discoveryComplete=true, attributeInformation=[AttributeInformation [dataType=UNSIGNED_16_BIT_INTEGER, identifier=65533]]]
```

Here the device lists the attributes it supports, and it doesn't include the required info for the channels you're looking for.

Rosen01, 2022-01-21:
Thanks for the fast answer !
If I've correctly anderstood, there is a problem in the response of the capabilities by the controller/lamp.
It's a RGB+CCT controller, so I know it's possible to control the cold/warm white leds.
Is there a way we can force the creation of this "color temperature" channel ? (ex. by creating XML thing file)

cdjackson, 2022-01-22:
The device is incorrectly reporting what attributes is supports, and this causes the binding not to support this channel.

I've changed the way the binding works now so that it doesn't ask the device what attributes it supports - it just tries to read the attribute. This should work for most devices.

PR #731 (2022-01-25): Simplify colour channel initialisation to improve compatibility with non-standard devices
This removes the request to ask devices what attributes they support, and just reads the attributes. This should make the binding more tolerant to non-standard devices.

Closes #730
closed 2022-01-25 by cdjackson