openhab/org.openhab.binding.zigbee #737: [Zigbee] Aqara Cube fails cluster/channel discovery

einalex, 2022-02-08:


cdjackson, 2022-05-21:
I'd need to see a log showing the initialisation of the device to understand why it's not initialising. Please can you provide this.

richardklingler, 2022-10-10:
This error persists also with regular Aqara rocker switches (single/double) on OH3.3 and RPi4.

Seems to be an issue with Ember as on Homeassistant with the same Sonoff USB stick all Aqara switches work flawlessly...

einalex, 2022-10-10:
[old-cube.patch.txt](https://github.com/openhab/org.openhab.binding.zigbee/files/9744897/old-cube.patch.txt)
This was my (working) attempt to fix the problem back then.
The issue were afaik missing channel types for ANALOGINPUT and MULTISTATEINPUT.
I'm not sure this is the correct way to solve it, but it worked for me.

cdjackson, 2022-10-10:
> This was my (working) attempt to fix the problem back then.

If you have a fix to add this support, I'd really suggest to provide a PR so it can be reviewed.

> This error persists also with regular Aqara rocker switches (single/double) on OH3.3 and RPi4.

Probably this is a completely different issue if it's a different device.  I would suggest to raise a new issue.

> Seems to be an issue with Ember as on Homeassistant with the same Sonoff USB stick all Aqara switches work flawlessly.

It's unlikely, and Homeassistant also uses the Sonoff switches which use the Ember chipset.

Moytora, 2023-01-18:
Hi Alex,
is this fix still woking for you? I tried to compile the Addon with the changes from your patch, however it is not compiling. I guess its missing files like: "ZigBeeConverterAnalogInput.java" and "ZigBeeConverterMultiStateInput.java"

I get these compilation errors:
`[ERROR] /home/.../org.openhab.binding.zigbee/src/main/java/org/openhab/binding/zigbee/internal/converter/ZigBeeDefaultChannelConverterProvider.java:[46,20] The method put(org.openhab.core.thing.type.ChannelTypeUID, java.lang.Class<? extends org.openhab.binding.zigbee.converter.ZigBeeBaseChannelConverter>) in the type java.util.Map<org.openhab.core.thing.type.ChannelTypeUID,java.lang.Class<? extends org.openhab.binding.zigbee.converter.ZigBeeBaseChannelConverter>> is not applicable for the arguments (org.openhab.core.thing.type.ChannelTypeUID, java.lang.Class<ZigBeeConverterAnalogInput>)`

`[ERROR] /home/.../org.openhab.binding.zigbee/src/main/java/org/openhab/binding/zigbee/internal/converter/ZigBeeDefaultChannelConverterProvider.java:[46,68] ZigBeeConverterAnalogInput cannot be resolved to a type`

`[ERROR] /home/.../org.openhab.binding.zigbee/src/main/java/org/openhab/binding/zigbee/internal/converter/ZigBeeDefaultChannelConverterProvider.java:[47,20] The method put(org.openhab.core.thing.type.ChannelTypeUID, java.lang.Class<? extends org.openhab.binding.zigbee.converter.ZigBeeBaseChannelConverter>) in the type java.util.Map<org.openhab.core.thing.type.ChannelTypeUID,java.lang.Class<? extends org.openhab.binding.zigbee.converter.ZigBeeBaseChannelConverter>> is not applicable for the arguments (org.openhab.core.thing.type.ChannelTypeUID, java.lang.Class<ZigBeeConverterMultistateInput>)`

`[ERROR] /home/.../org.openhab.binding.zigbee/src/main/java/org/openhab/binding/zigbee/internal/converter/ZigBeeDefaultChannelConverterProvider.java:[47,72] ZigBeeConverterMultistateInput cannot be resolved to a type
`

Do you have a git repository with these changes, where I could pull them from?

closed 2026-02-14 by github-actions[bot]