openhab/org.openhab.binding.zigbee #743: Allow configuration of reporting parameters in Channel Details for all Attributes that support reporting

FranzSchi, 2022-03-12:
### Outline

As a OpenHab user I want to be empowered to configure in the Channel Details the Reporting properties similar to.

zigbee subscribe 19620/1 0x001 0x0020 120 900 0
zigbee subscribe 19620/1 0x001 0x0021 120 900 0

like show here:
[image]

### Configuration

All device attributes that support (or has setup a reporting)

cdjackson, 2022-03-12:
As requested in the other thread, please provide a list of the converters that require this enhancement.

FranzSchi, 2022-03-15:
Hi @cdjackson,

thanks a lot for your kind support I really appreciate this.
It sometimes takes some times for me to understand the technical terms, I hope this answers the question.

As mentioned in the Post https://community.openhab.org/t/zigbee-consecutive-polling-of-attributes/134166/2?u=franzs
in the OH Forum. I would suggest to put in.

- Minimum Reporting Period
- Maximum Reporting Period
- Polling Period
- Reportable Change

Into this Converters:
ZigBeeConverterBatteryPercent
ZigBeeConverterBatteryVoltage

ZigBeeConverterMeasurementPower
ZigBeeConverterMeasurementRmsCurrent
ZigBeeConverterMeasurementRmsVoltage

ZigBeeConverterMeteringSummationDelivered
ZigBeeConverterMeteringSummationReceived

ZigBeeConverterRelativeHumidity
ZigBeeConverterTemperature

ZigBeeConverterWindowCoveringLift

Thanks a lot!
/Franz

FranzSchi, 2025-02-25:
Hi @cdjackson ,
I've seen that this was partly done for Plugs.
But as i remember it is still not applied to to the device unless a known init error is given is not resolved I do not remember the issue number anymore.
right?