openhab/org.openhab.binding.zigbee #797: The XML document '/OH-INF/addon/addon.xml' in module 'org.openhab.binding.zigbee' could not be parsed

rkoshak, 2023-02-14:
### Outline

This is a new behavior I've seen since upgrading from OH 4.0 SNAPSHOT #3304 to #3316.

Except for the alarming error in the logs, this does not appear to prevent the binding from operating but I have other problems which might be masking problems caused by this. 

The full exception logged:

```
2023-02-14 08:00:14.168 [WARN ] [ig.xml.osgi.XmlDocumentBundleTracker] - The XML document '/OH-INF/addon/addon.xml' in module 'org.openhab.binding.zigbee' could not be parsed:
---- Debugging information ----
cause-exception     : com.thoughtworks.xstream.mapper.CannotResolveClassException
cause-message       : author
class               : java.util.ArrayList
required-type       : java.util.ArrayList
converter-type      : com.thoughtworks.xstream.converters.collections.CollectionConverter
path                : /addon/author
line number         : 10
class[1]            : org.openhab.core.addon.xml.internal.AddonInfoXmlResult
required-type[1]    : org.openhab.core.addon.xml.internal.AddonInfoXmlResult
converter-type[1]   : org.openhab.core.addon.xml.internal.AddonInfoConverter
version             : 1.4.19
-------------------------------
com.thoughtworks.xstream.converters.ConversionException:
---- Debugging information ----
cause-exception     : com.thoughtworks.xstream.mapper.CannotResolveClassException
cause-message       : author
class               : java.util.ArrayList
required-type       : java.util.ArrayList
converter-type      : com.thoughtworks.xstream.converters.collections.CollectionConverter
path                : /addon/author
line number         : 10
class[1]            : org.openhab.core.addon.xml.internal.AddonInfoXmlResult
required-type[1]    : org.openhab.core.addon.xml.internal.AddonInfoXmlResult
converter-type[1]   : org.openhab.core.addon.xml.internal.AddonInfoConverter
version             : 1.4.19
-------------------------------
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:81) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.AbstractReferenceUnmarshaller.convert(AbstractReferenceUnmarshaller.java:72) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:68) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:52) ~[bundleFile:1.4.19]
        at org.openhab.core.addon.xml.internal.AddonInfoConverter.unmarshal(AddonInfoConverter.java:77) ~[?:?]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:74) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.AbstractReferenceUnmarshaller.convert(AbstractReferenceUnmarshaller.java:72) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:68) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:52) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.start(TreeUnmarshaller.java:136) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.AbstractTreeMarshallingStrategy.unmarshal(AbstractTreeMarshallingStrategy.java:32) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1421) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.XStream.unmarshal(XStream.java:1399) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.XStream.fromXML(XStream.java:1350) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.XStream.fromXML(XStream.java:1304) ~[bundleFile:1.4.19]
        at org.openhab.core.config.xml.util.XmlDocumentReader.readFromXML(XmlDocumentReader.java:105) ~[bundleFile:?]
        at org.openhab.core.config.xml.osgi.XmlDocumentBundleTracker.parseDocuments(XmlDocumentBundleTracker.java:396) [bundleFile:?]
        at org.openhab.core.config.xml.osgi.XmlDocumentBundleTracker.processBundle(XmlDocumentBundleTracker.java:382) [bundleFile:?]
        at org.openhab.core.config.xml.osgi.XmlDocumentBundleTracker$2.run(XmlDocumentBundleTracker.java:347) [bundleFile:?]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539) [?:?]
        at java.util.concurrent.FutureTask.run(FutureTask.java:264) [?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304) [?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136) [?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635) [?:?]
        at java.lang.Thread.run(Thread.java:833) [?:?]
Caused by: com.thoughtworks.xstream.mapper.CannotResolveClassException: author
        at com.thoughtworks.xstream.mapper.DefaultMapper.realClass(DefaultMapper.java:81) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:125) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.mapper.DynamicProxyMapper.realClass(DynamicProxyMapper.java:55) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:125) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.mapper.PackageAliasingMapper.realClass(PackageAliasingMapper.java:88) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:125) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.mapper.ClassAliasingMapper.realClass(ClassAliasingMapper.java:79) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:125) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.mapper.ArrayMapper.realClass(ArrayMapper.java:74) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:125) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.mapper.SecurityMapper.realClass(SecurityMapper.java:71) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.mapper.MapperWrapper.realClass(MapperWrapper.java:125) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.mapper.CachingMapper.realClass(CachingMapper.java:47) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.util.HierarchicalStreams.readClassType(HierarchicalStreams.java:29) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.converters.collections.AbstractCollectionConverter.readBareItem(AbstractCollectionConverter.java:131) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.converters.collections.AbstractCollectionConverter.readItem(AbstractCollectionConverter.java:117) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.converters.collections.CollectionConverter.addCurrentElementToCollection(CollectionConverter.java:99) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:92) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:86) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.converters.collections.CollectionConverter.unmarshal(CollectionConverter.java:81) ~[bundleFile:1.4.19]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:74) ~[bundleFile:1.4.19]
        ... 24 more                                                                                                             │
```

This might be related to https://github.com/openhab/openhab-core/pull/3362

The Zwave binding generates the same error.

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used | HUSZB1                                                                               |
| openHAB version  | 4.0 SNAPSHOT 3304 to 4.0 SNAPSHOT 3316    |
| Hardware         |   VM with 4 CPUs on an Intel based server class machine running VMWare ESXi, running in Docker    |
| Memory           |      6 GB                                                                          |
| Java version     |  OpenJDK 17.0.6 2023-01-17 Temurin-17.0.6+10                                                                              |
| Devices          | Not relevant to this issue but I can provide if required |

rkoshak, 2023-02-14:
I'm going to go ahead and close this. It seems likely that the problem is already fixed but not apparent because of the failing builds for the Zigbee and Zwave bindings. Once the builds work again, if the problem persists I'll come reopen it.

closed 2023-02-14 by rkoshak

kaikreuzer, 2023-02-14:
Right, this has been fixed by https://github.com/openhab/org.openhab.binding.zigbee/pull/795