openhab/org.openhab.binding.zigbee #803: Where support SONOFF ZBDongle-E USB Dongle Plus ZigBee in openHAB ZigBee Binding?!

frfstu, 2023-04-21:
Where is the support for working with openHAB that you declare in the description of your products ?! https://aliexpress.ru/item/1005004540962187.html?spm=a2g2w.orderdetail.0.0.3bbe4aa6byNPAU&sku_id=12000029537476470

Description

SONOFF ZB Dongle-E USB Dongle Plus ZigBee 3.0 Wireless Zigbee Gateway Analyzer ZHA Zigbee2MQTT Pre-Sparkling as a ZigBee Router

SONOFF Zigbee 3.0 USB Dongle Plus is a universal Zigbee USB flash drive. It can be used as a Zigbee gateway in home assistant, openHAB, Zigbee2MQTT or other open source platforms to manage all your Zigbee devices locally, so you don't need to invest in different brands of Zigbee hubs, all you need is a universal dongle gateway plus!

Dongle Plus has a wide range of supported devices such as ZBMINI-L, SNZB01, SNZB02, SNZB03, SNZB04, S26R2ZB, S40 LITE ZB. Dongle plus pre-firmware with coordinator, you can also flash router firmware to expand the network range.

Dongle Plus supports various common systems such as Linux, Ubuntu, Raspbian, Windows, macOS (Windows and macOS require appropriate driver installation).

2023-04-22 01:48:55.587 [ERROR] [zigbee.dongle.ember.ZigBeeDongleEzsp] - EZSP Dongle: NCP requires unsupported version of EZSP (required = V9, supported = V4)

Zigbee2MQTT version
1.30.3 commit: 24c6b2e2

coordinator type
EZSP v9

Coordinator revision
7.1.1.0 build 273

IEEE Address Coordinator
0x*********
front end version
0.6.127
Release = Raspbian GNU/Linux 11 (bullseye)
Kernel = Linux 6.1.21-v7+
Platform = Raspberry Pi 3 Model B Plus Rev 1.3
Uptime = 0 day(s). 0:0:32
CPU Usage = 57.61% avg over 4 cpu(s) (4 core(s) x 1 socket(s))
CPU Load = 1m: 2.39, 5m: 0.61, 15m: 0.21
Memory = Free: 0.46GB (49%), Used: 0.48GB (51%), Total: 0.94GB
Swap = Free: 1.80GB (100%), Used: 0.00GB (0%), Total: 1.80GB
Root = Free: 94.13GB (86%), Used: 15.24GB (14%), Total: 114.06GB
Updates = 0 apt updates available.
Sessions = 1 session(s)
Processes = 131 running processes of 32768 maximum processes

################################################### ##############################

                       _ _ _ ____ _

___ ___ ___ ___ | | | | / \ | __ ) () ____ ___
/ _ \ / _ \ / _ \ / _ \ | || | /\ | _ \ | | / _ \ / _
| () | () | /| | | || _ | / \ | |) )| || () || | | |
___/| / _/|| |||| ||// _|/ || _|||| | |
|| openHAB 3.4.3 - Release Build

Details here:
https://github.com/xsp1989/zigbeeFirmware/issues/42#issuecomment-1518388299

cdjackson, 2023-04-21:
As said elsewhere, this version of the firmware is not currently supported but will be added soon. You also mention Zigbee2MQTT - that is a different binding and not related to this binding.

closed 2023-04-21 by cdjackson

frfstu, 2023-04-21:
> As said elsewhere, this version of the firmware is not currently supported but will be added soon. You also mention Zigbee2MQTT - that is a different binding and not related to this binding.

Zigbee2MQTT works thankfully!

cdjackson, 2023-05-02:
I'm not really sure why you've posted this link? Please can you explain?