openhab/org.openhab.binding.zigbee #808: Selectable Baudrate: 460800 for Corrdinators

FranzSchi, 2023-09-01:
Currently 

38400 or 57600 or 115200 baud can be selected
[image]

newer Firmware and Sticks allow JW flow and 460800 baud.
See here:
https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP

Thanks
/Franz

cdjackson, 2023-09-01:
I would suggest to use one of the values already available. There is no advantage in a faster baud rate.

FranzSchi, 2023-09-01:
Hi Chris,

thanks for that but if the Firmware only supports 460800 in case of HW Flow control?

BR
/Franz

Hedda, 2023-11-24:
As per above; community firmware builders are now starting to provide firmware images with different baud rate speeds, and a few community testers are at least claiming that 230400 baud rate speed gives them better results in very large Zigbee networks.

FYI tip for devs, darkxst is providing prebuild cutting-edge firmware for most common EFR32MGxx adapters with different speeds:

https://github.com/darkxst/silabs-firmware-builder/tree/manifestjs-SL/firmware_builds

`ncp-uart-hw-` = EZSP mode (pure EmberZnet Zigbee NCP).
`rcp-uart-802154-` = RCP mode for MultiPAN/multiprotocol (requires Silicon Labs EZSP and/or OpenThread daemon on the host).
`ot-rcp-` = OpenThread Only (Thread Spinel mode).

Note! That `hw` in the files name does not currently mean hardware flow control as that is hardcoded in the patches today.

For example, check out firmware for Sonoff ZBDongle-E (by ITead) hardware which has latest NCP images in different buad rate:

https://github.com/darkxst/silabs-firmware-builder/tree/manifestjs-SL/firmware_builds/zbdonglee

Web Flasher - Flash newer Silabs EFR32 adapters directly from your browser (only Chrome and Edge supported) -> [SL Web Flasher](https://darkxst.github.io/silabs-firmware-builder/)

Read this [blog post](https://dialedin.com.au/blog/sonoff-zbdongle-e-rcp-firmware) for more details and instructions for using RCP Multi-PAN firmware (with Silabs Multiprotocol Addon for HA).

Tip; The `manifestjs-SL` branch in darkxst's silabs-firmware-builder repo also allow you to build same firmware images yourself:

https://github.com/darkxst/silabs-firmware-builder/tree/manifestjs-SL/firmware_builds

PS: Some other Zigbee gateway solutions have an "Auto" mode where it probe to test connecting, starting at higher speed & HW.