openhab/org.openhab.binding.zigbee #812: Eurotronic Comet Zigbee - missing config

thopiekar, 2023-10-21:
### Outline

I just bought an energy saving radiator "Comet Zigbee" by Eurotronic. Compared to a different Zigbee thing (ledvance power plug), this one is discovered as "Unknown ZigBee Device F4CE367948F1F447".

As I'm completely new to the Zigbee community at OpenHAB, I don't know whether there is a similar database to Z-Wave things, where I can explore and add new devices to the project.

https://eurotronic.org/wp-content/uploads/2023/04/BDA-Comet-Zigbee-EN-2023-04-06.pdf
https://eurotronic.org/wp-content/uploads/2023/04/BDA-Comet-Zigbee-DE-2023-04-06.pdf

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used |    Aeotec Zi-Stick   |
| openHAB version  |  4.0.x    |
| Hardware         |     amd64   |
| Memory           |    <enough>    |
| Java version     |     17.x    |

### Logs

F4CE367948F1F447.xml
```
<ZigBeeNode>
  <ieeeAddress>F4CE367948F1F447</ieeeAddress>
  <networkAddress>25599</networkAddress>
  <macCapabilities class="sorted-set"/>
  <endpoints/>
  <bindingTable/>
</ZigBeeNode>
```

Thank you for your help! :slightly_smiling_face:

cdjackson, 2023-10-22:
There is no database. Probably the device did not respond to all requests during the interview so the binding doesn't have the information to work out what to do (as shown by the fact that. the XML is basically empty).

thopiekar, 2023-10-22:
So if I understand correctly the binding is able to receive all needed info from the device during inclusion.
I removed the thing from the network and re-added it three times, but didn't get any further than before.

Hereby the most recent openhab.log attached.
[openhab.log](https://github.com/openhab/org.openhab.binding.zigbee/files/13064523/openhab.log)

cdjackson, 2023-10-22:
There is very little information from the device - it fails to respond to most of the messages sent to it. Possibly the device is leaving the network for some reason - when I've seen this sort of thing in the past it's sometimes caused by the coordinator having old firmware, but I'm not sure. It could also be that the RF environment is not good - eg it's too far away from the coordinator.

Below you see the device join, but then no responses to the messages from the coordinator...

[image]

thopiekar, 2023-10-23:
Is there a way to re-request the data? Do I need to simply rerun the scan?

cdjackson, 2023-10-23:
It’s unclear why it’s not responding - probably requesting it again will not work since you’ve already tried many times. You need to resolve the communications problem first I think.

thopiekar, 2023-10-24:
I tried it with different coordinator settings, double checked the settings recommended by the manufacturer (https://aeotec.freshdesk.com/support/solutions/articles/6000261822-setup-zi-stick-with-openhab), but I still couldn't get the device included and discovered.
Re-adding the "LEDVANCE Plug Value" switch was no issue.

I'm just wondering about one step in the manual:
[image]
According to them there should be a number displayed. However, at this point, the device shows "no" (which could stand for "number") and after that "0", which is the coordinator, no? Or is this number for a node in the mesh fine?

cdjackson, 2023-10-24:
My guess is it is not correctly joining the network - or to be maybe a bit more precise, it joins the network, but then decides to leave. This can happen if a device feels the network security is not good enough - it was common with Z3.0 devices that joined an older HA1.2 network. As mentioned above, this is normally a firmware issue, but I'm surprised as I thought the Aotec device would likely have reasonably new firmware. Can you advise what firmware it is running?

thopiekar, 2023-10-25:
Well, I can't find any website for available firmware. Maybe I've been too optimistic when buying the dongle.
Is there any other coordinator than the ones listed here: https://www.openhab.org/addons/bindings/zigbee/#supported-coordinators ?
From what I see the elelabs website is broken, so the company is dead? Is the SONOFF coordinator a good choice?

Maybe it is better to give this project another try with different hardware 🤔

cdjackson, 2023-10-25:
The firmware version number should be displayed in the properties of the coordinator - can you advise what that says please?

thopiekar, 2023-10-26:
Sure!  :slightly_smiling_face: 

[image]

thopiekar, 2023-10-26:
Same crap with the SONOFF dongle :(

[image]

The laptop with the dongle and the Eurotronic Comet are less than half a meter apart.

I don't get it. For me, the Eurotronic Comet is bugged :(

thopiekar, 2023-10-26:
Adding the LEDVANCE plug was no issue at all..

thopiekar, 2023-11-01:
I have returned all Zigbee devices. Will give my project another try with ZWave 😇

closed 2023-11-01 by thopiekar