openhab/org.openhab.binding.zigbee #820: Shall we run mvn spotless:apply ??

andrewfg, 2023-12-26:
While working on a PR for this repository, I noticed that running 'mvn spotless:apply' results in many tens of files being reformatted.

As it is now common practice in openhab-core and openhab-addons to keep all source files spotless compliant, I wonder if we should do the same here? => If there are no objections, I would myself be quite happy to create a PR with spotless applied.

kaikreuzer, 2025-08-17:
> If there are no objections, I would myself be quite happy to create a PR with spotless applied.

No comment from @cdjackson since 1.5 years on this. I would agree with you that we should apply the same formatting rules as for other add-ons, so I wouldn't mind to apply spotless. The only risk is that existing open PRs might become useless, but it seems to me as if there isn't any PR that is about to be merged.

wborn, 2025-08-20:
I can create a PR to apply Spotless on a Linux machine.

We can also add a `.git-blame-ignore-revs` file after reformatting the code, see:

* https://medium.com/@ramunarasinga/git-blame-ignore-revs-to-ignore-bulk-formatting-changes-f20ac23e6155
* https://docs.github.com/en/repositories/working-with-files/using-files/viewing-and-understanding-files#ignore-commits-in-the-blame-view