openhab/org.openhab.binding.zigbee #830: Zigbee 4-gang switch toggles all gangs at once

stvred, 2024-02-13:
Hi, i have a [TS0004 4-gang switch](https://www.zigbee2mqtt.io/devices/TS0004_switch_module.html) and I have problems. When I press the physical button, everything works good. When i press button from OpenHAB, ON  first channel they all 4 turn on, with turning off it’s the same
```
2024-02-13 13:17:13.599 [INFO ] [openhab.event.ItemCommandEvent      ] - Item 'Unknown_ZigBee_Device_A4C138AB43A12D54_A4C138AB43A12D54_1_Switch' received command ON
2024-02-13 13:17:13.651 [INFO ] [openhab.event.ItemStateChangedEvent ] - Item 'Unknown_ZigBee_Device_A4C138AB43A12D54_A4C138AB43A12D54_1_Switch' changed from OFF to ON
2024-02-13 13:17:13.842 [INFO ] [openhab.event.ItemStateChangedEvent ] - Item 'Unknown_ZigBee_Device_A4C138AB43A12D54_A4C138AB43A12D54_2_Switch' changed from OFF to ON
2024-02-13 13:17:14.043 [INFO ] [openhab.event.ItemStateChangedEvent ] - Item 'Unknown_ZigBee_Device_A4C138AB43A12D54_A4C138AB43A12D54_3_Switch' changed from OFF to ON
2024-02-13 13:17:14.243 [INFO ] [openhab.event.ItemStateChangedEvent ] - Item 'Unknown_ZigBee_Device_A4C138AB43A12D54_A4C138AB43A12D54_4_Switch' changed from OFF to ON
```
I tried to find a solution to this problem on the Internet, but without success.

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used |       Zigbee 3.0 USB Dongle Plus V2                                       |
| openHAB version  | openHAB 4.1.1 Release Build     |
| Hardware         |                                                                                |
| Memory           |                                                                                |
| Java version     |      openjdk version "17.0.10" 2024-01-16 LTS                                                                          |
| Devices          |                                    |

### Logs

openhab> openhab:zigbee ncpversion
`Ember NCP version 6.10.3.0, EZSP version 8`

openhab> openhab:zigbee  ncpstate
```
Ember NCP state     : EMBER_JOINED_NETWORK
Local Node Type     : EMBER_COORDINATOR
IEEE Address        : 6C5CB1FFFEFC71E4
Custom IEEE Address : FFFFFFFFFFFFFFFF
NWK Address         : 0000
Network PAN Id      : 1DCF
Network EPAN Id     : EFDF9CE60229DD43
Radio Channel       : 11
Network Manager Id  : 0000
Radio TX Power      : -1
Join Method         : EMBER_USE_MAC_ASSOCIATION
Board Name          :
Manufacturer Name   :
Stack Version       : 6.10.3.0
Custom Version      :
```
openhab:zigbee nodes
```
Total known nodes in network: 5
Network  Addr  IEEE Address      Logical Type  State      EP   Profile                    Device Type                Manufacturer     Model
      0  0000  6C5CB1FFFEFC71E4  COORDINATOR   ONLINE
  21126  5286  A4C13827AA7C9A44  ROUTER        ONLINE      1  ZIGBEE_HOME_AUTOMATION     ON_OFF_LIGHT               _TZ3000_lmlsduws  TS0002
                                                           2  ZIGBEE_HOME_AUTOMATION     ON_OFF_LIGHT
                                                         242  ZIGBEE_GREEN_POWER         0061
  32992  80E0  A4C138661985071E  ROUTER        ONLINE      1  ZIGBEE_HOME_AUTOMATION     ON_OFF_LIGHT               _TZ3000_lmlsduws  TS0002
                                                           2  ZIGBEE_HOME_AUTOMATION     ON_OFF_LIGHT
                                                         242  ZIGBEE_GREEN_POWER         0061
  35482  8A9A  A4C1381B4C2AE709  END_DEVICE    ONLINE      1  ZIGBEE_HOME_AUTOMATION     TEMPERATURE_SENSOR         _TZ3000_bjawzodf  TY0201
  53771  D20B  A4C138AB43A12D54  ROUTER        ONLINE      1  ZIGBEE_HOME_AUTOMATION     ON_OFF_LIGHT               _TZ3000_5ajpkyq6  TS0004
                                                           2  ZIGBEE_HOME_AUTOMATION     ON_OFF_LIGHT
                                                           3  ZIGBEE_HOME_AUTOMATION     ON_OFF_LIGHT
                                                           4  ZIGBEE_HOME_AUTOMATION     ON_OFF_LIGHT
                                                         242  ZIGBEE_GREEN_POWER         0061
```
log:set debug org.openhab.binding.zigbee
log:set debug com.zsmartsystems.zigbee
```
2024-02-12 12:08:57.973 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - A4C138AB43A12D54: Command for channel zigbee:device:bbd459ca36:a4c138ab43a12d54:A4C138AB43A12D54_1_switch --> ON [OnOffType]
2024-02-12 12:08:57.974 [DEBUG] [e.transaction.ZigBeeTransactionQueue] - A4C138AB43A12D54: Added transaction to queue, len=1, transaction=ZigBeeTransaction [ieeeAddress=A4C138AB43A12D54 queueTime=0, state=WAITING, sendCnt=0, command=OnCommand [On/Off: 0000/0 -> D20B/1, cluster=0006, TID=--]]
2024-02-12 12:08:57.974 [DEBUG] [transaction.ZigBeeTransactionManager] - Transaction Manager: Send Next transaction. outstandingTransactions=0, outstandingQueues=1, sleepy=0/3
2024-02-12 12:08:57.974 [DEBUG] [transaction.ZigBeeTransactionManager] - D20B/1: Sending ZigBeeTransaction [ieeeAddress=A4C138AB43A12D54 queueTime=1, state=WAITING, sendCnt=0, command=OnCommand [On/Off: 0000/0 -> D20B/1, cluster=0006, TID=F2]]
2024-02-12 12:08:57.974 [DEBUG] [transaction.ZigBeeTransactionManager] - transactionListenerAdded: 1 outstanding
[... 70 lines cut ...]
2024-02-12 12:08:58.422 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=SERVER_TO_CLIENT, disableDefaultResponse=true, manufacturerCode=0, sequenceNumber=09, commandId=10]
2024-02-12 12:08:58.422 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReportAttributesCommand [On/Off: D20B/3 -> 0000/1, cluster=0006, TID=09, reports=[AttributeReport [attributeDataType=BOOLEAN, attributeIdentifier=0, attributeValue=true]]]
2024-02-12 12:08:58.423 [DEBUG] [transaction.ZigBeeTransactionManager] - notifyTransactionCommand: ReportAttributesCommand [On/Off: D20B/3 -> 0000/1, cluster=0006, TID=09, reports=[AttributeReport [attributeDataType=BOOLEAN, attributeIdentifier=0, attributeValue=true]]] 
2024-02-12 12:08:58.423 [DEBUG] [converter.ZigBeeConverterSwitchOnoff] - A4C138AB43A12D54: ZigBee attribute reports ZclAttribute [cluster=On/Off, id=0, name=On Off, dataType=BOOLEAN, lastValue=true, lastReportTime=Mon Feb 12 12:08:58 MSK 2024, implemented=true]
2024-02-12 12:08:58.423 [DEBUG] [converter.ZigBeeBaseChannelConverter] - A4C138AB43A12D54: Channel zigbee:device:bbd459ca36:a4c138ab43a12d54:A4C138AB43A12D54_3_switch updated to ON
2024-02-12 12:08:58.423 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - A4C138AB43A12D54: Updating ZigBee channel state zigbee:device:bbd459ca36:a4c138ab43a12d54:A4C138AB43A12D54_3_switch to ON
2024-02-12 12:08:58.423 [DEBUG] [.zigbee.handler.ZigBeeIsAliveTracker] - IsAlive Tracker reset for handler with thingUID=zigbee:device:bbd459ca36:a4c138ab43a12d54
2024-02-12 12:08:58.423 [DEBUG] [.zigbee.handler.ZigBeeIsAliveTracker] - IsAlive Tracker cancelled task for thingUID=zigbee:device:bbd459ca36:a4c138ab43a12d54
2024-02-12 12:08:58.423 [DEBUG] [.zigbee.handler.ZigBeeIsAliveTracker] - IsAlive Tracker scheduled task for thingUID=zigbee:device:bbd459ca36:a4c138ab43a12d54 in 1830 seconds
2024-02-12 12:08:58.622 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=6, ackNum=0, reTx=false, data=5B 90 01 45 00 00 04 01 06 00 04 01 00 01 00 00 14 8C BF 0B D2 FF FF 07 18 0A 0A 00 00 10 01 02]
2024-02-12 12:08:58.623 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=7, notRdy=false]
2024-02-12 12:08:58.623 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingMessageHandler [networkId=0, type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=0104, clusterId=0006, sourceEndpoint=4, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY], groupId=0, sequence=14], lastHopLqi=140, lastHopRssi=-65, sender=D20B, bindingIndex=255, addressIndex=255, messageContents=18 0A 0A 00 00 10 01]
2024-02-12 12:08:58.623 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=D20B/4, destinationAddress=0000/1, profile=0104, cluster=0006, addressMode=DEVICE, radius=0, apsSecurity=false, ackRequest=false, apsCounter=14, rssi=-65, lqi=8C, payload=18 0A 0A 00 00 10 01]
2024-02-12 12:08:58.623 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - A4C138AB43A12D54: Node update. NWK Address=NULL
[... cut ...]
2024-02-12 12:08:58.624 [DEBUG] [.zigbee.handler.ZigBeeIsAliveTracker] - IsAlive Tracker reset for handler with thingUID=zigbee:device:bbd459ca36:a4c138ab43a12d54
2024-02-12 12:08:58.624 [DEBUG] [.zigbee.handler.ZigBeeIsAliveTracker] - IsAlive Tracker cancelled task for thingUID=zigbee:device:bbd459ca36:a4c138ab43a12d54
2024-02-12 12:08:58.624 [DEBUG] [.zigbee.handler.ZigBeeIsAliveTracker] - IsAlive Tracker scheduled task for thingUID=zigbee:device:bbd459ca36:a4c138ab43a12d54 in 1830 seconds
```

stvred, 2024-02-13:
I found a similar problem, but it was solved by abandoning Binding and switching to z2m
[link](https://community.openhab.org/t/tuya-ts0002-and-openhab/147808/9)

cdjackson, 2024-02-13:
If I look at the log you provided, it looks like the binding sends a single command to the device on endpoint 1, and the device turns on all switches -:

[image]

This seems not normal from the device, so I am not sure why it is doing that.

stvred, 2024-02-14:
Yes, at first time it also look to me that the problem was in the device, change it and you will be happy. But as it turned out, this is not actually the problem. I put together a circuit link to which I [published](https://community.openhab.org/t/tuya-ts0002-and-openhab/147808/9) above (OH-mqttBinding-mosquitto-zidbee2mqtt-coordinator-device), and the device works correctly, all channels switch independently and everything works correctly. So, as it turned out, the problem is not in the device, but in the fact that in the circuit (OH-zigbeeBinging-coordinator-device) it sends a command to the coordinator for the device, and the device activates all channels at the same time. Also, I looked for similar problems in similar projects and found them in [HA](https://community.home-assistant.io/t/tuya-zigbee-2-gang-switch-toggles-both-gangs-at-once/487622) and[ zigbee2mqtt](https://github.com/Koenkk/zigbee2mqtt/issues/13954) and judging by the messages, an update was released that solved problem.

cdjackson, 2024-02-14:
Maybe there's a config setting in the device to change it's functionality - I don't know. To me, what the binding is doing by sending a single command to a single endpoint should result in a single switch coming on. This all looks fine. If the device has special, or non-standard functions, then unfortunately I'm not aware of this. You'd need to find out what the non-standard feature is before anything can be done here.

stvred, 2024-02-15:
Unfortunately, I don’t have an additional device to sniff the transfer from the coordinator to the device, I only see logs. The logs that I see are the debug log between the binding and the coordinator. If I can get some additional data in this connection, then tell me how to do it. In conjunction with Z2M at the mqtt level, i see that depending on the channel, a different command is transmitted, here is an example:
`payload{"indicator_mode":null,"linkquality":152,"power_on_behavior":null,"state_l1":"ON","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","switch_type":"toggle"}'
payload{"indicator_mode":null,"linkquality":152,"power_on_behavior":null,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","switch_type":"toggle"}'
payload{"indicator_mode":null,"linkquality":152,"power_on_behavior":null,"state_l1":"OFF","state_l2":"ON","state_l3":"OFF","state_l4":"OFF","switch_type":"toggle"}'
payload{"indicator_mode":null,"linkquality":152,"power_on_behavior":null,"state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","switch_type":"toggle"}'`

It seems to me that the problem is that the binding sends a command to the device in the payload with one command for one channel, and the device waits for four and interprets this command for all channels. I think the device receives the state of all four channels at once in one command. But these are my thoughts ;) based on the Z2M log.

cdjackson, 2024-02-17:
> It seems to me that the problem is that the binding sends a command to the device in the payload with one command for one channel, and the device waits for four and interprets this command for all channels.

Are you sure? This would be very strange. You migth be right of course, but it's very non-standard and it's not really possible to support this.

cdjackson, 2024-02-18:
Again, this is completely non-standard. Every endpoint should be separate - as though it is a separate device. So maybe there is other configuration that you are not seeing that gets sent before. I don't know, but this change is not easy for the binding to implement since it now bridges across multiple channels, and within the binding, the implementation is separate.

Maybe you can implement this as a rule if you are sure this is the issue.

stvred, 2024-02-24:
Problem solved by using [z2m](https://www.zigbee2mqtt.io/). Thank you @cdjackson  for your involvement