openhab/org.openhab.binding.zigbee #831: SLZB-06M does not connect

madhacker3kxl, 2024-03-02:
Using SLZB-06M to connect to openhab does not work. It will say socket connection fail. Any way to support Ethernet connected devices?

Reply from vendor:

> I can see that the communication between the SLZB-06 and openhab is working and packets are being forwarded but the requests that openhab is sending are incorrect.
> SLZB-06 runs on koenk's z-stack firmware and it's designed to work with Z2M and ZHA, and it looks like some commands are different/missing in it so openhab won't work.
> I also tested SLZB-06M here the situation is the same, communication starts but it seems that openhab is designed for the old version of ezsp and does not work with the new ones that are flashed into our coordinators.

Openhab Version: openHAB 4.1.1

cdjackson, 2024-03-02:
It's an interesting device - I'd potentially be interested to look at supporting this, but I am not really sure what protocol it is using and don't have the device to test with so it's hard to support.

madhacker3kxl, 2024-03-03:
Currently is has EZSP firmware. It works well with Zigbee2MQTT using tcp://. I am willing to pay for purchase of the device.

I even tried to get https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds/slzb-06m to work. It does not connect.

cdjackson, 2024-03-03:
As I mentioned, this looks like an interesting device and I'm happy to look at implementing it... 

It's not the EZSP firmware specifically that Im concerned about, but more how it interfaces over ethernet as this is non-standard I guess. I saw from a look at their website today that they also have a configuration UI which could possibly also be implemented as an openHAB thing, but I'd need to get a device to test, and also find the interface definition, or possibly just read through the source which I found on GitHub would also work if there's no doc (I couldn't find any at least).

Tarik2142, 2024-03-05:
> As I mentioned, this looks like an interesting device and I'm happy to look at implementing it...
>
> It's not the EZSP firmware specifically that Im concerned about, but more how it interfaces over [...]

The device works as a TCP server and acts as a UART-network bridge between the Zigbee chip and Z2M/ZHA or other software.
We can discuss sending a sample to you. Please write to info@smartlight.me

madhacker3kxl, 2024-03-26:
Any update on this?

cdjackson, 2024-03-26:
The manufacturer have kindly sent me a unit and hopefully it will arrive in the next few days. Once it arrives I will take a look at implementing it for OH.

madhacker3kxl, 2024-04-25:
Following up to see how this is going.

cdjackson, 2024-04-25:
I have received the device and done some initial testing with it using the USB and the library and it seems to work fine here. I need to test with OH - can you advise exactly how you define the tcp connection - is it just `tcp://ip-address:port`?

Also, I note that you tested this on OH4.1 - did you try OH 4.2 yet since the update for the latest EZSP was merged a couple of months back, but only into the 4.2 branch.

madhacker3kxl, 2024-04-25:
Yes, just `tcp://ip-address:port`. No issues with USB connection.

I haven't tried with 4.2 yet with IP connection.

cdjackson, 2024-04-25:
Ok, thanks. It's interesting to note that it works ok for you on the USB as I understood from re-reading the comments above a week or so back that the issue was related to the version of EZSP which would mean that USB also doesn't work...

I will try and test with the TCP connection over the weekend.

madhacker3kxl, 2024-04-25:
Sorry about the confusion. From discussing with the developers I had impression, that the tcp did not work due to version issues.

Let me know if there is anything from my end I can test out.

cdjackson, 2024-04-30:
Just a quick update on this...

I've implemented a new handler for the SLZB06. This has the advantage of tighter integration, and a better user experience (hopefully :)).

The new handler supports auto discovery through MDNS, and I'm looking to see if I can add information from the device as OH channels (eg the SLZB uptime, temperatures, etc).

@madhacker3kxl are you in a position to test this if I provide a JAR? It would be compiled against OH4.2 snapshot. In any case I'll try and get this linked here in the coming days.

madhacker3kxl, 2024-04-30:
That's excellent news!

Yes, I can test it later in the week.

cdjackson, 2024-05-11:
@madhacker3kxl sorry - I've not had time over the past week or so to look at this too much. You can try [this branch](https://github.com/cdjackson/org.openhab.binding.zigbee/commit/da11d16e3725b7343477735b405d7f576d0a8220) though if you want (and are able to compile the binding).

I think this works ok - discovery should work, polling of some of the info from the SLZB06 such as temperature should work, but I don't think I hooked up all the channels.

The thing I know I need to look at is the reconnection - if the network goes down, it doesn't seem to reconnect at the moment, which obviously is not good. Still, if you want to test, your feedback would be welcome.

If you can't compile the binding, let me know and I'll compile it - although I'd need to look at how to distribute as it can't be attached to an issue I think.

madhacker3kxl, 2024-05-13:
I was able to compile and connect! It took few attempts to get slzb connected due to serial buffer overruns. 

[image]

I also added few devices and they seem to report just fine. One thing I noticed there wasn't a option to set up the baudrate.

cdjackson, 2024-05-13:
Thanks.

I've not seen any "serial buffer overruns" here. Note that this isn't really "serial" and I should change this logging - it's network overruns (overruns of the receive buffer in the binding - it's received data from the SLZB06 but the binding hasn't processed it fast enough)... What are you running this on - I wonder if it's a "slow" computer?  It might be worth enabling debug logging to better see what's happening there.

Regarding baud rate - why do you need to set this? The binding side shouldn't care - it just sends the data to the SLZB06 across the network - the baud rate from the ESP32 (the host) in the SLZB06 and the EFR (the zigbee processor) is set in the SLZB06 and doesn't really matter to the binding.

I can probably make this configurable in the binding as well, but it doesn't really matter to openhab.

cdjackson, 2024-05-13:
Thanks @madhacker3kxl. As a matter of interest, are you using the SLZB06 network connection with other software? I'm wondering if the device is buffering up a load of data after you disconnect one software, and then when OH connects, it blasts out it's buffer...  Does this happen every time when starting the binding, or was it a once off? Of if you leave it unconnected for (say) 5 minutes, and then start the binding does that cause this?

I've not seen any unsolicited data come in after connection, but I'm running this on a relatively small test network. I could imagine that if you had sensors sending out reports, then the SLZB06 might be buffering the data received from the EFR32 while the network is down, and then if this is all sent quickly when the network comes back up, it could possibly swamp the buffer in the binding. 

Or I might be completely wrong :) If this is repeatable, it would be useful to see a debug log if possible.

madhacker3kxl, 2024-05-13:
I am only using it with OH. It seems like it is one off. 

I reinstalled the binding and added the devices again to get a debug log.

```
openhab> log:display
15:54:29.940 [INFO ] [hab.event.ThingStatusInfoChangedEvent] - Thing 'zigbee:coordinator_slzb06:dbfbac350b' changed from UNINITIALIZED to INITIALIZING
15:54:30.009 [WARN ] [tty.util.ssl.SslContextFactory.config] - Trusting all certificates configured for Client@15a024e2[provider=null,keyStore=null,trustStore=null]
15:54:30.009 [WARN ] [tty.util.ssl.SslContextFactory.config] - No Client EndPointIdentificationAlgorithm configured for Client@15a024e2[provider=null,keyStore=null,trustStore=null]
15:54:51.234 [INFO ] [hab.event.ThingStatusInfoChangedEvent] - Thing 'zigbee:coordinator_slzb06:dbfbac350b' changed from INITIALIZING to ONLINE
15:55:20.717 [DEBUG] [lzb06.internal.api.Slzb06Communicator] - SLZB06 request: http://192.168.8.10/ha_sensors
15:55:20.776 [DEBUG] [lzb06.internal.api.Slzb06Communicator] - SLZB06 response: {"Sensors":{"esp32_temp":"54.44","zb_temp":"51.81","uptime":1099,"socket_uptime":49693,"ram_usage":94,"fs_used":244,"ethernet":true,"wifi_connected":false,"wifi_status":255,"disable_leds":false,"night_mode":false,"auto_zigbee":false}}
15:55:50.780 [DEBUG] [lzb06.internal.api.Slzb06Communicator] - SLZB06 request: http://192.168.8.10/ha_sensors
15:55:50.835 [DEBUG] [lzb06.internal.api.Slzb06Communicator] - SLZB06 response: {"Sensors":{"esp32_temp":"54.44","zb_temp":"51.32","uptime":1129,"socket_uptime":79754,"ram_usage":94,"fs_used":244,"ethernet":true,"wifi_connected":false,"wifi_status":255,"disable_leds":false,"night_mode":false,"auto_zigbee":false}}
15:56:20.836 [DEBUG] [lzb06.internal.api.Slzb06Communicator] - SLZB06 request: http://192.168.8.10/ha_sensors
15:56:20.888 [DEBUG] [lzb06.internal.api.Slzb06Communicator] - SLZB06 response: {"Sensors":{"esp32_temp":"55.00","zb_temp":"52.96","uptime":1159,"socket_uptime":109808,"ram_usage":95,"fs_used":244,"ethernet":true,"wifi_connected":false,"wifi_status":255,"disable_leds":false,"night_mode":false,"auto_zigbee":false}}
15:56:46.054 [DEBUG] [covery.Slzb06MDNSDiscoveryParticipant] - SLZB-06: Discovery getServiceType '_slzb-06._tcp.local.'
15:56:46.257 [DEBUG] [covery.Slzb06MDNSDiscoveryParticipant] - SLZB-06: Discovery getServiceType '_slzb-06._tcp.local.'
15:56:50.889 [DEBUG] [lzb06.internal.api.Slzb06Communicator] - SLZB06 request: http://192.168.8.10/ha_sensors
15:56:50.945 [DEBUG] [lzb06.internal.api.Slzb06Communicator] - SLZB06 response: {"Sensors":{"esp32_temp":"55.00","zb_temp":"53.23","uptime":1189,"socket_uptime":139864,"ram_usage":95,"fs_used":244,"ethernet":true,"wifi_connected":false,"wifi_status":255,"disable_leds":false,"night_mode":false,"auto_zigbee":false}}
15:57:04.600 [INFO ] [gbee.discovery.ZigBeeDiscoveryService] - 00124B0022417813: Starting ZigBee device discovery
15:57:04.609 [INFO ] [ig.discovery.internal.PersistentInbox] - Added new thing 'zigbee:device:dbfbac350b:00124b0022417813' to inbox.
15:57:04.609 [INFO ] [openhab.event.InboxAddedEvent        ] - Discovery Result with UID 'zigbee:device:dbfbac350b:00124b0022417813' has been added.
15:57:20.947 [DEBUG] [lzb06.internal.api.Slzb06Communicator] - SLZB06 request: http://192.168.8.10/ha_sensors
15:57:21.002 [DEBUG] [lzb06.internal.api.Slzb06Communicator] - SLZB06 response: {"Sensors":{"esp32_temp":"55.56","zb_temp":"54.08","uptime":1219,"socket_uptime":169920,"ram_usage":95,"fs_used":244,"ethernet":true,"wifi_connected":false,"wifi_status":255,"disable_leds":false,"night_mode":false,"auto_zigbee":false}}
15:57:38.086 [INFO ] [openhab.event.InboxRemovedEvent      ] - Discovery Result with UID 'zigbee:device:dbfbac350b:00124b0022417813' has been removed.
15:57:38.093 [INFO ] [hab.event.ThingStatusInfoChangedEvent] - Thing 'zigbee:device:dbfbac350b:00124b0022417813' changed from UNINITIALIZED to INITIALIZING
15:57:38.096 [INFO ] [hab.event.ThingStatusInfoChangedEvent] - Thing 'zigbee:device:dbfbac350b:00124b0022417813' changed from INITIALIZING to UNKNOWN
15:57:38.461 [INFO ] [openhab.event.FirmwareStatusInfoEvent] - Firmware status of thing zigbee:device:dbfbac350b:00124b0022417813 changed to UNKNOWN.
15:57:48.277 [INFO ] [hab.event.ThingStatusInfoChangedEvent] - Thing 'zigbee:device:dbfbac350b:00124b0022417813' changed from UNKNOWN to ONLINE
15:57:51.004 [DEBUG] [lzb06.internal.api.Slzb06Communicator] - SLZB06 request: http://192.168.8.10/ha_sensors
15:57:51.058 [DEBUG] [lzb06.internal.api.Slzb06Communicator] - SLZB06 response: {"Sensors":{"esp32_temp":"55.56","zb_temp":"53.76","uptime":1250,"socket_uptime":199977,"ram_usage":95,"fs_used":244,"ethernet":true,"wifi_connected":false,"wifi_status":255,"disable_leds":false,"night_mode":false,"auto_zigbee":false}}
15:58:21.060 [DEBUG] [lzb06.internal.api.Slzb06Communicator] - SLZB06 request: http://192.168.8.10/ha_sensors
15:58:21.117 [DEBUG] [lzb06.internal.api.Slzb06Communicator] - SLZB06 response: {"Sensors":{"esp32_temp":"55.56","zb_temp":"53.49","uptime":1280,"socket_uptime":230035,"ram_usage":94,"fs_used":244,"ethernet":true,"wifi_connected":false,"wifi_status":255,"disable_leds":false,"night_mode":false,"auto_zigbee":false}}
15:58:25.673 [INFO ] [openhab.event.ItemStateChangedEvent  ] - Item 'test_Temperature' changed from NULL to 83.264 °F
15:58:31.872 [INFO ] [openhab.event.ItemUpdatedEvent       ] - Item 'test_Temperature' has been updated.
[... cut ...]
16:12:22.085 [DEBUG] [lzb06.internal.api.Slzb06Communicator] - SLZB06 response: {"Sensors":{"esp32_temp":"57.22","zb_temp":"53.99","uptime":731,"socket_uptime":721600,"ram_usage":97,"fs_used":244,"ethernet":true,"wifi_connected":false,"wifi_status":255,"disable_leds":false,"night_mode":false,"auto_zigbee":false}}
```

The buffer overrun never appeared again. I dont believe there is any buffer in this device as my other setup is with Zigbee2mqtt and after the service is off for a while, it only gets the latest informaion.

cdjackson, 2024-05-13:
Is the port 2003 correct? It's (I think) not the standard. I'm just wondering why the error here is that there's no route - clearly the SLZB is working and there's a route since the sensor data is being received ok... That said, the `socket_uptime` is not zero which (I think) means there's already a connection to the socket, and this might be preventing another connection. I would assume that the SLZB closes connections after some time if it's broken.

madhacker3kxl, 2024-05-13:
I apologize for not telling my setup. Yes, I set the port to be 2003. The log above is from disconnect (16:07:42.104) to reconnect. 16:10:06.972 is the time when I connected OH back to the network.

OH connected moments after the network came back online.

cdjackson, 2024-05-13:
Ok, thanks. So if it's connecting after the network is back up, that all sounds like it's working ok. I also noted in your log that the periodic reconnection seems to be working ok which is something that I thought wasn't working properly for me. I'll take another look at that when I get a chance (probably next week), but it's encouraging that it looks to be working in your system.

Anyway, please let me know if you have any problems, queries, or ideas for improvement.

cdjackson, 2024-06-03:
@madhacker3kxl I just wanted to know how this is working for you? I've been using this now on my system for the past few weeks and have had no problems at all.

madhacker3kxl, 2024-06-03:
So far, I have not had any issues, that I noticed anyways. I am still running the branch I compiled 3 weeks ago. Can't wait till Openhab v4.2 to come out! As always, many thanks!

cdjackson, 2024-06-03:
Great - thanks for the feedback. I've not made any significant changes to the SLZB-06 branch since you compiled your version so I think we're good. I'll look to get this merged soon so you can test with an official build if you want and it will then be available for 4.2.

PR #835 (2024-06-07): Add Slzb06m Network Coordinator Handler
Adds support for the [SLZB06m](https://smlight.tech/product/slzb-06m/) with the Ember chipset.

Closes #831
closed 2024-06-07 by cdjackson