openhab/org.openhab.binding.zigbee #832: IKEA Styrbar - Only 2 of 4 buttons show up as channels

magx2, 2024-04-16:
I can confirm what Cplant wrote [here](https://community.openhab.org/t/ikea-styrbar-only-2-of-4-buttons-show-up-as-channels/153793) that IKEA Styrbar has only 2 buttons available in OH instead of 4.

> I’m on OH 4.1, using a Sonoff Zigbee dongle with the openHAB Zigbee-Binding.
> 
> If I pair the Styrbar button with openHAB, it works great for two of the four buttons (the ones for on/off). The other two buttons, however, (the ones that control the light temperature) do not appear under Channels:

source: 
https://community.openhab.org/t/ikea-styrbar-only-2-of-4-buttons-show-up-as-channels/153793

What info you would like to get in order to be able to fix this bug?

cdjackson, 2024-04-16:
The binding works by detecting device services, and then trying to create channels based on what it finds. If the device works in a non-standard way, then of course this won't work.  Without detailed information on the device, it's pretty much impossible for me to know what this device does and how it works. If we can find out how it works, then in principal someone could create a static definition (as XML) so that this overrides the automatic discovery.

magx2, 2024-04-16:
I have this device. What do you want me to do?

cdjackson, 2024-04-16:
We need to find out how it works. You can try the fingerprint command, but again, this uses standard mechanisms to discover the services and if the device is doing something non-standard, then it probably won't discover it.

You can also set up a sniffer using a second Ember dongle to try and sniff the data on the network to see what happens when the button is pressed.

Or maybe there's information elsewhere on the web that describes how it works?

magx2, 2024-04-17:
From zigbee2mqtt deocs:

> Zigbee-herdsman handles the core Zigbee communication. The module [zigbee-herdsman-converters](https://github.com/koenkk/zigbee-herdsman-converters) handles the mapping from individual device models to the Zigbee clusters they support.

Here is the mapping: 

 - https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/ikea.ts#L780
 - https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/ikea.ts#L549

cdjackson, 2024-04-17:
The fingerprint seems to indicate there's only 1 endpoint, so I'd expect only 1 button.  

I had a quick look at the herdsman converters - it seems (if I understand them right) that the buttons just break down the different parts of the level control functionality. Is anything received at all when you press these buttons? (ie is anything displayed in the log if you have debug logging enabled).

magx2, 2024-04-17:
```xml
<Logger name="org.openhab.binding.zigbee" level="debug" additivity="false">
    <AppenderRef ref="ZIGBEE"/>
</Logger>
```

I've added this for logging but I didn't get anything meaningful. Any advice?

cdjackson, 2024-04-17:
Doo you see anything when you press the other buttons - ie the ones that do work ok?

magx2, 2024-04-18:
Yes, if I press on/off I'm getting some logs

cdjackson, 2024-04-23:
As far as I can tell, this probably ought to work. What channels are actually provided for the two buttons that work?

The next step will be to enable TRACE level debugging for the binding and reinitialise the device. This will allow me to see what's going on with the device discovery and the initialisation, but first please let me know what the channels are and also please provide the logs showing the data received for the buttons that work.

magx2, 2024-04-23:
[image]

Up and down works as dimmer, left/right do not work

[image]

I will provide you logs this week

cdjackson, 2024-04-23:
Ok, thanks. 

From looking at the fingerprint, I'd assumed that the 2 that worked were on and off and the other 2 were brightness. So what do you expect the other two buttons should do?

magx2, 2024-04-26:
```xml
<Logger name="org.openhab.binding.zigbee" level="trace" additivity="false">
	<AppenderRef ref="ZIGBEE"/>
</Logger>
```

Should I add something more?

magx2, 2024-04-26:
> From looking at the fingerprint, I'd assumed that the 2 that worked were on and off and the other 2 were brightness. So what do you expect the other two buttons should do?

I had this paired to smart bulb that could change brightness and color temp. The usage can be described as:

- press up/down - turn on/off the light
- hold up/down - smooth change the brightness of the light
- press left/right - change the color temp (cycle through 3 pre defined options: 100% warm, 100% cold, 50/50)
- hold left/right - smooth change of the color temp (similar to up down)

In general both up/down and left/right works the same, the change only different properties of the bulb. With OH+rules I've already copied the up/down functionality.

cdjackson, 2024-04-27:
Hmmm - were these functions handled through some sort of rule? I don't think there's a direct way to control the colour temp in this way, so either there's a rule involved, or there's some way to program the remote to send very specific commands.  I also just noted that the device doesn't seem to support the colour control cluster, so I guess you use some other command (probably one of the level control commands?) and this triggers a rule?

I somehow need to work out what the low level commands are...

magx2, 2024-04-27:
Check this video: https://www.youtube.com/watch?v=NF5G5wziwso . There are no rules or Ikea gateway, just pure connection Styrbar-Bulb

cdjackson, 2024-04-27:
Ok, then I don't know. According to the fingerprint, the device does not support colour control which I think is the way it should be controlling the colour temp.

magx2, 2024-04-27:
> ```
> <Logger name="org.openhab.binding.zigbee" level="trace" additivity="false">
> 	<AppenderRef ref="ZIGBEE"/>
> </Logger>
> ```
> 
> Should I add something more?

Should I click styrbar with this enabled?

cdjackson, 2024-04-27:
I think this logging should be fine, but I'd need to see the initialisation to get a bit more detail. You've already said that nothing is received when you click the buttons, so changing to TRACE logging won't change that. I need to find out what commands are being used so I can configure the device.

As I said above, the device (apparently) doesn't support colour control, so I'm not sure what commands it sends to change the bulb colour. The other option is to find a sniffer and sniff what's happening on a working system.

magx2, 2024-04-27:
Do you want me to turn off trace logging and pair styrbar one more time?

cdjackson, 2024-04-27:
We need TRACE logging on for the pairing/discovery to see fully what's happening with the channel discovery. However, my thinking around looking at the discovery was when I thought the channels were around on/off and level control - not colour control. I'm not sure this will really help though as it probably won't tell us how the colour control is managed given that the device doesn't seem to be reporting the colour control cluster.

Still, getting the full log, with TRACE, won't hurt.

However...

I just found this issue -: https://github.com/Koenkk/zigbee2mqtt/issues/1232

And this comment -:

[image]

This seems to indicate that the Ikea bridge is programming the device to do something "special". I'll continue to read this thread (tonight - I need to go out and do some work in the garden now) as I suspect it might have the info I'm looking for so for now let's hold off - no sense wasting your time just yet :)

magx2, 2024-07-07:
Hey, any update on this?

cdjackson, 2024-07-07:
No - I'm not sure what else can be done. I don't have this device or the Ikea gateway to be able to work out how it is configured - as above - it looks like the Ikea gateway is sending extra configuration to implement this. It might be possible to program a scene or something, but without the device I'm simply unable to do much.

magx2, 2024-07-08:
Can I send you 12 pounds so you would buy this device https://www.ikea.com/gb/en/p/styrbar-remote-control-smart-white-60488366/ (no gateway is required)?

cdjackson, 2024-07-08:
Unfortunately I'm not living in the UK at the moment - I'm in New Zealand. We don't have an Ikea here (yet) - although one is meant to open soon. I think that makes things a bit difficult to buy stuff...

But if I understand the issue (which possibly I don't) it seems that the Ikea gateway is configuring the remote to do something with the buttons. It states above that "removing and pairing again looses the ability to control the color_temp with the remote" and that pairing with the Ikea bridge seems to be different than pairing direct to a coordinator. The issue also states "it seems the tradfri hub does some special setup".

My guess is that the hub is configuring scenes - but that is a guess. The linked issue is very long with hundreds of comments and I'm not had the time to read through this to work out what the solution is. Are you able to spend some time looking at this to see if you can find the relevant info?

magx2, 2024-07-09:
> that the Ikea gateway is configuring the remote to do something with the buttons

There is no gateway. In Ikea's use case Styrbar is connecting directly to the bulb(s). Check this video - https://www.youtube.com/watch?v=NF5G5wziwso

> My guess is that the hub is configuring scenes

Yeah, but in OH I do not see those buttons at all... If the device would show me 4 buttons instead of 2 I would do everything in OH without a problem

cdjackson, 2024-07-09:
> There is no gateway. 

Strange - the zigbeemqtt thread talks about the Ikea bridge doing some configuring of the device. How does it join a network? Where is the coordinator? Does the bulb act as a coordinator, in which case how are other devices connected to the network? The video is not really particularly informative - we need to understand technically how this works.

> Yeah, but in OH I do not see those buttons at all

Sure - but that's a whole different issue. At the moment, the device is not sending any data when you press the buttons. Until it does that, we can't work out how to configure the binding to receive the data as I have no idea what it is.

magx2, 2024-07-09:
> How does it join a network? Where is the coordinator? Does the bulb act as a coordinator

I think there is no network/zigbee mesh. Styrbar is communicating directly with a light bulb so either styrbar is an coordinator or lighbulb.

Check [this video](https://youtu.be/fH5t9eGfhzQ?si=ErfR6FTRKGOLLpDt&t=443) it shows you how it's pairing light bulbs with styrbar and motion sensor without gateway

cdjackson, 2024-07-09:
> I think there is no network/zigbee mesh.

So every light is connected to only the single remote - nothing else? That seems very limiting if each bulb is effectively its own network

magx2, 2024-07-09:
On the video guy connected a remote and motion sensor to one bulb. idk but maybe you can connect multiple remotes to a bulb

cdjackson, 2024-07-09:
> On the video guy connected a remote and motion sensor to one bulb

Sure - but that doesn't make it a good thing to do. Normally a device can only be included in a single network, so if the bulb is acting as the coordinator, then it can't be in an OH network.

This probably uses the Light Link system that was used in the past for Hue. In any case, if you want to connect it to OH, then it needs to join the OH mesh network.

magx2, 2024-07-09:
> so if the bulb is acting as the coordinator, then it can't be in an OH network.

But if you restart the bulb and the remote they can join zigbee network as normal devices. In this moment I have 3 bulbs and 1 styrbar remote that are in my zigbee/OH network and I can control them via OH

[image]

cdjackson, 2024-07-09:
Sure - so you can have them in a normal network, or directly connected. Again, this is similar to the Hue bulbs. However, if they are in a normal network, then, from what I've read and linked above, the gateway is doing some configuration to configure the device to send other commands.

At the moment I don't have much information on how this works, so it's pretty much impossible to know how to implement it. Unfortunately I don't have the time to read hundreds of messages to see how or if zigbee2mqtt got it working so it would be helpful if you can try and find this (as I suggested earlier).  We need to know how to make the device send commands for all buttons, then we need to work out what the commands are, and then we can look to implement something in OH.