openhab/org.openhab.binding.zigbee #847: Only One Zigbee Coordinator Works at a Time

madhacker3kxl, 2024-07-08:
Only one Zigbee coordinator works at a time. Whichever initializes first, it will be the one that will show Online and the other will be communication error. Example picture where SLZB06 initialized later:

[image]

### Configuration

| Configuration     | Description                                                                    |
|------------------|----------------------------------------------------------|
| Coordinator used |  SLZB06M and Nortec HUBz                                        |
| openHAB version  | 4.2     |
| Hardware         |     LXC                                                       |
| Memory           |    4GB                                                                            |
| Java version     |     Zulu17.50+19-CA                                                      |
| Devices          | Various sensors and outlets                                  |

### Logs

Following log is both Zigbee coordinator is disabled and enabling one at at time and disable again. This was after a clean boot. SLZB06M initialzed first.
```
2024-07-08 10:01:33.861 [DEBUG] [ng.zigbee.ember.handler.EmberHandler] - Initializing ZigBee Ember serial bridge handler.
2024-07-08 10:01:33.861 [DEBUG] [ng.zigbee.ember.handler.EmberHandler] - ZigBee Ember Coordinator opening Port:'rfc2217://192.168.1.208:2001' PAN:2012, EPAN:988499449112ABD6, Channel:25
2024-07-08 10:01:33.861 [DEBUG] [ng.zigbee.ember.handler.EmberHandler] - Ember end device poll timeout set to (225 * 2^5) = 7200 seconds
2024-07-08 10:01:33.861 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Scheduling ZigBee start
2024-07-08 10:01:34.862 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - ZigBee network starting
2024-07-08 10:01:34.862 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - Initialising ZigBee coordinator
2024-07-08 10:01:34.865 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - null: networkStateUpdated called with state=INITIALISING
2024-07-08 10:01:34.869 [DEBUG] [ding.zigbee.internal.ZigBeeDataStore] - 00124B0022417A66: ZigBee reading network state complete.
2024-07-08 10:01:34.878 [DEBUG] [ding.zigbee.internal.ZigBeeDataStore] - 00124B002228F3EE: ZigBee reading network state complete.
2024-07-08 10:01:34.887 [DEBUG] [ding.zigbee.internal.ZigBeeDataStore] - 00124B001C4B2B98: ZigBee reading network state complete.
2024-07-08 10:01:34.894 [DEBUG] [ding.zigbee.internal.ZigBeeDataStore] - 00124B001F918E75: ZigBee reading network state complete.
2024-07-08 10:01:34.904 [DEBUG] [ding.zigbee.internal.ZigBeeDataStore] - 00124B002228F511: ZigBee reading network state complete.
2024-07-08 10:01:34.916 [DEBUG] [ding.zigbee.internal.ZigBeeDataStore] - F0D1B800000736F7: ZigBee reading network state complete.
2024-07-08 10:01:34.923 [DEBUG] [ding.zigbee.internal.ZigBeeDataStore] - 000D6F001200B3E3: ZigBee reading network state complete.
2024-07-08 10:01:34.931 [DEBUG] [ding.zigbee.internal.ZigBeeDataStore] - F0D1B80000121FED: ZigBee reading network state complete.
2024-07-08 10:01:34.932 [DEBUG] [nding.zigbee.serial.ZigBeeSerialPort] - Connecting to serial port [rfc2217://192.168.1.208:2001] at 57600 baud, flow control FLOWCONTROL_OUT_XONOFF.
2024-07-08 10:01:35.033 [ERROR] [nding.zigbee.serial.ZigBeeSerialPort] - Serial Error: Device cannot be opened on [rfc2217://192.168.1.208:2001]. Caused by Unable to establish remote connection to serial port rfc2217://192.168.1.208:2001, call stack: java.lang.IllegalStateException: Unable to establish remote connection to serial port rfc2217://192.168.1.208:2001
        at org.openhab.core.io.transport.serial.rxtx.rfc2217.internal.SerialPortIdentifierImpl.open(SerialPortIdentifierImpl.java:60)
        at org.openhab.binding.zigbee.serial.ZigBeeSerialPort.open(ZigBeeSerialPort.java:157)
        at org.openhab.binding.zigbee.serial.ZigBeeSerialPort.open(ZigBeeSerialPort.java:123)
        at com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp.initialiseEzspProtocol(ZigBeeDongleEzsp.java:1289)
        at com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp.initialize(ZigBeeDongleEzsp.java:432)
        at com.zsmartsystems.zigbee.ZigBeeNetworkManager.initialize(ZigBeeNetworkManager.java:418)
        at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler.initialiseZigBee(ZigBeeCoordinatorHandler.java:431)
        at org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler.lambda$2(ZigBeeCoordinatorHandler.java:557)
        at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539)
        at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
        at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
        at java.base/java.lang.Thread.run(Thread.java:840)
Caused by: java.net.SocketTimeoutException: Connect timed out
        at java.base/sun.nio.ch.NioSocketImpl.timedFinishConnect(NioSocketImpl.java:551)
        at java.base/sun.nio.ch.NioSocketImpl.connect(NioSocketImpl.java:602)
        at java.base/java.net.SocksSocketImpl.connect(SocksSocketImpl.java:327)
        at java.base/java.net.Socket.connect(Socket.java:639)
        at org.apache.commons.net.SocketClient._connect(SocketClient.java:132)
        at org.apache.commons.net.SocketClient.connect(SocketClient.java:268)
        at org.openhab.core.io.transport.serial.rxtx.rfc2217.internal.SerialPortIdentifierImpl.open(SerialPortIdentifierImpl.java:57)
        ... 13 more

2024-07-08 10:01:35.034 [ERROR] [zigbee.dongle.ember.ZigBeeDongleEzsp] - EZSP Dongle: Unable to open serial port
2024-07-08 10:01:35.034 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - null: networkStateUpdated called with state=OFFLINE
2024-07-08 10:01:35.036 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00124B002228F511: Coordinator status changed to OFFLINE.
2024-07-08 10:01:35.036 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00124B002228F511: Coordinator is unknown or not online.
2024-07-08 10:01:35.036 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00124B0022417A66: Coordinator status changed to OFFLINE.
2024-07-08 10:01:35.037 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00124B0022417A66: Coordinator is unknown or not online.
2024-07-08 10:01:35.039 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00124B001C4B2B98: Coordinator status changed to OFFLINE.
2024-07-08 10:01:35.040 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00124B002228F3EE: Coordinator status changed to OFFLINE.
2024-07-08 10:01:35.040 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 00124B002228F3EE: Coordinator is unknown or not online.
2024-07-08 10:01:35.040 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - F0D1B800000736F7: Coordinator status changed to OFFLINE.
[... 231 lines cut ...]
2024-07-08 10:02:32.383 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - ZigBee initialise done. channel=CHANNEL_26, PanId=34009  EPanId=26D2F2BE80D4CD5C
2024-07-08 10:02:32.880 [DEBUG] [ding.zigbee.internal.ZigBeeDataStore] - 943469FFFE73C56E: ZigBee saving network state complete.
2024-07-08 10:02:52.303 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - 943469FFFE73C56E: networkStateUpdated called with state=OFFLINE
2024-07-08 10:02:52.303 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - UNINITIALIZED: Ignoring OFFLINE status: Thing is {}
2024-07-08 10:02:52.303 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - 943469FFFE73C56E: networkStateUpdated called with state=SHUTDOWN
2024-07-08 10:02:52.307 [DEBUG] [ding.zigbee.internal.ZigBeeDataStore] - 943469FFFE73C56E: ZigBee saving network state complete.
2024-07-08 10:02:52.308 [DEBUG] [ding.zigbee.internal.ZigBeeDataStore] - 00124B00222E9346: ZigBee saving network state complete.
2024-07-08 10:02:57.306 [WARN ] [core.thing.internal.ThingManagerImpl] - Disposing handler for thing 'zigbee:coordinator_slzb06:56a865c77e' takes more than 5000ms.
2024-07-08 10:02:57.309 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - bundle org.openhab.binding.zigbee:4.2.0 (295)[org.openhab.binding.zigbee.discovery.ZigBeeDiscoveryService(344)] : dm ZigBeeCoordinatorHandler tracking 11 MultipleDynamic removed {org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler}={service.id=720, service.bundleid=303, service.scope=singleton} (enter)
[... cut ...]
2024-07-08 10:03:02.420 [WARN ] [.core.thing.binding.BaseThingHandler] - Handler Slzb06Handler of thing zigbee:coordinator_slzb06:56a865c77e tried checking if channel zb_temp is linked although the handler was already disposed.
2024-07-08 10:03:02.420 [WARN ] [.core.thing.binding.BaseThingHandler] - Handler Slzb06Handler of thing zigbee:coordinator_slzb06:56a865c77e tried checking if channel uptime is linked although the handler was already disposed.
2024-07-08 10:03:02.422 [WARN ] [.core.thing.binding.BaseThingHandler] - Handler Slzb06Handler of thing zigbee:coordinator_slzb06:56a865c77e tried checking if channel socket_uptime is linked although the handler was already disposed.
```

cdjackson, 2024-07-08:
Unfortunately the logging doesn't show the info from the zigbee library, but I suspect that the issue here is that both of these are Ember chips and they are running very different firmware. There is a constraint in the zigbee library where a static is used to store information for communication with the dongle -:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/issues/1149

I couldn't find info on the HUBz - is that the HUSBZB1 or is it different? If you can find newer firmware, then it would work I think.  I'll also take a look at the library issue again, but that's also messy to resolve.

madhacker3kxl, 2024-07-08:
It is the HUSBZB1. Yes, they are both ember devices. There is no newer firmware version for it. I am not worried about it, as I needed them both online to move devices from one to another, one at a time. You are welcome to close this issue. If you need more logs, let me know.

cdjackson, 2024-07-08:
Thanks.

I'll modify the title to reflect that this is Ember and leave it open for now. I'll take another look to see if this can be resolved in an elegant way in the library as it isn't a nice "feature" to have. Interestingly though, in all the years this is the first time it's come up in OH at least.