openhab/org.openhab.binding.zigbee #848: SLZB-06M will Discover Nodes but Interview Imcomplete

madhacker3kxl, 2024-07-12:
Adding any Zigbee device using SLZB06M will not complete device interview. Same devices will work fine using HUSBZB1. Tried different channels, joining secure mode and various child ageing. 

One I did notice is that `openhab:zigbee ncpstate` does not work. Says "Error: Could not find command: ncpstate" or any other commands starting with `ncp` even though it is listed at supported commands. 

### Configuration

| Configuration     | Description                                                                    |
|------------------|----------------------------------------------------------|
| Coordinator used |  SLZB06M                                       |
| openHAB version  | 4.2     |
| Hardware         |     LXC                                                       |
| Memory           |    4GB                                                                            |
| Java version     |     Zulu17.50+19-CA                                                      |
| Devices          | Various sensors and outlets                                  |

### Logs
```
openhab> log:clear
openhab> log:tail
12:39:15.994 [DEBUG] [lzb06.internal.api.Slzb06Communicator] - SLZB06 response: {"Sensors":{"esp32_temp":"57.78","zb_temp":"56.02","uptime":3387,"socket_uptime":1369,"ram_usage":94,"fs_used":244,"ethernet":true,"wifi_connected":false,"wifi_status":255,"disable_leds":false,"night_mode":false,"auto_zigbee":false}}
12:39:31.414 [INFO ] [gbee.discovery.ZigBeeDiscoveryService] - 00124B00222E9346: Starting ZigBee device discovery
12:39:31.415 [DEBUG] [gbee.discovery.ZigBeeDiscoveryService] - 00124B00222E9346: Creating ZigBee device zigbee:device with bridge zigbee:coordinator_slzb06:56a865c77e
12:39:31.416 [INFO ] [openhab.event.InboxAddedEvent        ] - Discovery Result with UID 'zigbee:device:56a865c77e:00124b00222e9346' has been added.
12:39:31.416 [INFO ] [ig.discovery.internal.PersistentInbox] - Added new thing 'zigbee:device:56a865c77e:00124b00222e9346' to inbox.
12:39:31.416 [DEBUG] [gbee.discovery.ZigBeeDiscoveryService] - 00124B00222E9346: Node discovery not complete
12:39:31.656 [DEBUG] [nding.zigbee.internal.ZigBeeDataStore] - 00124B00222E9346: ZigBee saving network state complete.
12:39:41.806 [WARN ] [b.core.thing.binding.BaseThingHandler] - Handler Slzb06Handler of thing zigbee:coordinator_slzb06:56a865c77e tried checking if channel rx_dat is linked although the handler was already disposed.
12:39:41.807 [WARN ] [b.core.thing.binding.BaseThingHandler] - Handler Slzb06Handler of thing zigbee:coordinator_slzb06:56a865c77e tried checking if channel tx_dat is linked although the handler was already disposed.
12:39:41.807 [WARN ] [b.core.thing.binding.BaseThingHandler] - Handler Slzb06Handler of thing zigbee:coordinator_slzb06:56a865c77e tried checking if channel rx_ack is linked although the handler was already disposed.
12:39:41.807 [WARN ] [b.core.thing.binding.BaseThingHandler] - Handler Slzb06Handler of thing zigbee:coordinator_slzb06:56a865c77e tried checking if channel tx_ack is linked although the handler was already disposed.
12:39:41.808 [WARN ] [b.core.thing.binding.BaseThingHandler] - Handler Slzb06Handler of thing zigbee:coordinator_slzb06:56a865c77e tried checking if channel rx_nak is linked although the handler was already disposed.
12:39:41.808 [WARN ] [b.core.thing.binding.BaseThingHandler] - Handler Slzb06Handler of thing zigbee:coordinator_slzb06:56a865c77e tried checking if channel tx_nak is linked although the handler was already disposed.
12:39:41.809 [DEBUG] [lzb06.internal.api.Slzb06Communicator] - SLZB06 request: http://192.168.7.4/ha_sensors
12:39:41.835 [DEBUG] [lzb06.internal.api.Slzb06Communicator] - SLZB06 response: {"Sensors":{"esp32_temp":"57.22","zb_temp":"54.68","uptime":3413,"socket_uptime":1395,"ram_usage":94,"fs_used":244,"ethernet":true,"wifi_connected":false,"wifi_status":255,"disable_leds":false,"night_mode":false,"auto_zigbee":false}}
12:39:41.835 [WARN ] [b.core.thing.binding.BaseThingHandler] - Handler Slzb06Handler of thing zigbee:coordinator_slzb06:56a865c77e tried checking if channel esp32_temp is linked although the handler was already disposed.
12:39:41.835 [WARN ] [b.core.thing.binding.BaseThingHandler] - Handler Slzb06Handler of thing zigbee:coordinator_slzb06:56a865c77e tried checking if channel zb_temp is linked although the handler was already disposed.
12:39:41.836 [WARN ] [b.core.thing.binding.BaseThingHandler] - Handler Slzb06Handler of thing zigbee:coordinator_slzb06:56a865c77e tried checking if channel uptime is linked although the handler was already disposed.
12:39:41.836 [WARN ] [b.core.thing.binding.BaseThingHandler] - Handler Slzb06Handler of thing zigbee:coordinator_slzb06:56a865c77e tried checking if channel socket_uptime is linked although the handler was already disposed.
12:39:45.995 [DEBUG] [lzb06.internal.api.Slzb06Communicator] - SLZB06 request: http://192.168.7.4/ha_sensors
12:39:46.024 [DEBUG] [lzb06.internal.api.Slzb06Communicator] - SLZB06 response: {"Sensors":{"esp32_temp":"57.22","zb_temp":"55.75","uptime":3417,"socket_uptime":1399,"ram_usage":94,"fs_used":244,"ethernet":true,"wifi_connected":false,"wifi_status":255,"disable_leds":false,"night_mode":false,"auto_zigbee":false}}
12:39:46.732 [DEBUG] [nding.zigbee.internal.ZigBeeDataStore] - 00124B00222E9346: ZigBee saving network state complete.
^C~                                                                                                                                                                  openhab> 
zigbee nodes
Total known nodes in network: 2
Network  Addr  IEEE Address      Logical Type  State      EP   Profile                    Device Type                Manufacturer     Model          
      0  0000  943469FFFE73C56E  COORDINATOR   ONLINE   
  14506  38AA  00124B00222E9346  UNKNOWN       ONLINE

```

cdjackson, 2024-07-12:
I am using the SLZB06 for my network atHome now and have 18 devices currently connected with many different types used, and have had no problem with any device joining the network and being interrogated. I have light switches, battery powered sensors, presence sensors and hot water controller from different manufacturers and all have worked flawlessly for the past 6 weeks since I moved over to using the SLZB06. There is no discovery specific code in the SLZB driver so this all should work exactly the same as other coordinators. Probably the NCP commands don’t work as this is coordinator dependant and it might need to be changed to make this work. I’ll take a look at this in the coming days. Sent from my iPhone

cdjackson, 2024-07-12:
Just to add that if your comment relates to the log entry that discovery is not complete, then this is normal - you will always see this shortly after joining.

Do you see any channels with the devices? If so, discovery is completed just fine. If there are no channels, then the interview failed.

madhacker3kxl, 2024-07-12:
When i add the device it always states it is unknown device with not channels. In the past, I had to run the scan again and it will complete the interview, it didn't work with this one. I have also waited ~10 minutes to see if anything changed. It worked fine with the same device when I tested the snapshot when we got the SLZB implemented. I will try again with a fresh openhab install and see if i can replicate the issue there.

cdjackson, 2024-07-12:
> When i add the device it always states it is unknown device with not channels. 

So, again, this is normal initially. You should then get the device name changed, and only after it's added as a device will you see the channels. Depending on the device, this can take some time - mains powered devices will be quick, battery devices will take much longer, and if you have non-standard battery devices, you may need to wake them up.

I had assumed from previous conversations a month or two ago that your system was working with devices having channels. If I understand this issue, then none of your devices ever show any channels? This is strange as a) the discovery is not related to the coordinator code, it's part of the library, and b) as I mentioned, I have 18 devices currently in my network and they have all worked fine for the past 6 weeks or so. I added another battery device just 2 days ago, and it also added fine.

I've made a change to support the console commands - as mentioned this is caused by the coordinator type not being EMBER.

madhacker3kxl, 2024-07-12:
With our previous conversations it did show the correct name on initial discovery and correct channels after I added the device. The test was using a clean install of openhab in a VM. This issue is with my existing/production installation. 

So spun-up a VM with clean install of openhab and installed the zigbee add-on. It discovered the device just fine, channels and all.
[image]

It might be an issue with my existing install.

cdjackson, 2024-07-12:
Ok, thanks.  Hopefully it works ok - I'm not sure what to do if it doesn't...

I'll have a fix for the ncp commands today - I've made the change already so just need to make the PR.

madhacker3kxl, 2024-07-12:
Got it working. I had to remove zigbee addon and clean out the addons folder that had the jar file located. Then reinstalled it. Something probably went wrong from 4.1 to 4.2 upgrade and having existing zigbee ember device with older firmware.

closed 2024-07-12 by madhacker3kxl