openhab/org.openhab.binding.zigbee #853: Colortemperature and Brightness (HSB channel UNDEF on polling)

m4rcu5, 2024-08-01:
## Outline

Most of my Zigbee bulbs have a _ColorTemperature_ and _Color_ channel. These lights have a separate CCT led and RGB leds.
So it makes sense that if you set a _ColorTemperature_, the Color channel will be reset to **UNDEF** (as no RGB LEDs are in use). Which I believe is also working-as-expected looking at PR #336. However the brightness still has to be controlled by the _Color_ channel.

Initially this works somewhat, you can send a brightness value to the _Color_ channel. Now the _Color_ channel switches to `0,0,B-value`  and the bulb lights. However after the polling interval has passed, the _Color_ channel bounces back to _UNDEF_.

Leaving the lights with a _ColorTemperature_ set, also makes them bounce on every poll. The channels go to a value and back to UNDEF in the same poll.

I've done some more in depth digging into the behavior on the forum (https://community.openhab.org/t/dimmer-and-color-item-to-same-channel/157241/11). I'll add the summary below.

## Configuration

| Configuration | Description |
|------------------|------------------|
| Coordinator used | ITEAD ZBDongle-E |
| openHAB version  | 4.2.0-1 (Debian 12.6) |
| Hardware         | Intel(R) N100, x86_64 |
| Memory           | 16GB (-Xms512m -Xmx4096m) |
| Java version     | zulu17-jre (17.0.12-1) |
| Devices          | MiBoxer RGB+CCT (003C84FFFEABB5F7), Philips Hue White & Color ambiance (001788010C5AC808) |

## Logs

### Color channel UNDEF even when brightness is set

**Starting at the off (`0,0,0`) setting**
```
openhab> openhab:send OfficeLedstripWall_Color 0,0,0

[INFO ] [openhab.event.ItemCommandEvent      ] - Item 'OfficeLedstripWall_Color' received command 0,0,0
[DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 003C84FFFEABB5F7: Command for channel zigbee:device:dc80d093a7:003c84fffeabb5f7:003C84FFFEABB5F7_1_color --> 0,0,0 [HSBType]
[DEBUG] [converter.ZigBeeBaseChannelConverter] - 003C84FFFEABB5F7: Channel zigbee:device:dc80d093a7:003c84fffeabb5f7:003C84FFFEABB5F7_1_color waiting for response to 0,0,0
[DEBUG] [converter.ZigBeeBaseChannelConverter] - 003C84FFFEABB5F7: Channel zigbee:device:dc80d093a7:003c84fffeabb5f7:003C84FFFEABB5F7_1_color received SUCCESS in response to 0,0,0
[DEBUG] [converter.ZigBeeBaseChannelConverter] - 003C84FFFEABB5F7: Channel zigbee:device:dc80d093a7:003c84fffeabb5f7:003C84FFFEABB5F7_1_color updated to 0,0,0
[DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 003C84FFFEABB5F7: Updating ZigBee channel state zigbee:device:dc80d093a7:003c84fffeabb5f7:003C84FFFEABB5F7_1_color to 0,0,0
[DEBUG] [.zigbee.handler.ZigBeeIsAliveTracker] - IsAlive Tracker scheduled task for thingUID=zigbee:device:dc80d093a7:003c84fffeabb5f7 in 14430 seconds
```

**Setting a normal color (including a brightness)**
```
openhab> openhab:send OfficeLedstripWall_Color 123,45,50

[INFO ] [openhab.event.ItemStateChangedEvent ] - Item 'OfficeLedstripWall_Color' changed from 0,0,0 to 123,45,50
[INFO ] [openhab.event.ItemStateChangedEvent ] - Item 'OfficeLedstripWall_Dimmer' changed from 0 to 50
[DEBUG] [rter.ZigBeeConverterColorTemperature] - 003C84FFFEABB5F7: ZigBee attribute reports ZclAttribute [cluster=Color Control, id=8, name=Color Mode, dataType=ENUMERATION_8_BIT, lastValue=0, lastReportTime=Thu Aug 01 16:32:36 CEST 2024, implemented=false]  on endpoint 1
[DEBUG] [.converter.ZigBeeConverterColorColor] - 003C84FFFEABB5F7: ZigBee attribute reports ZclAttribute [cluster=Color Control, id=8, name=Color Mode, dataType=ENUMERATION_8_BIT, lastValue=0, lastReportTime=Thu Aug 01 16:32:36 CEST 2024, implemented=false]
[DEBUG] [converter.ZigBeeBaseChannelConverter] - 003C84FFFEABB5F7: Channel zigbee:device:dc80d093a7:003c84fffeabb5f7:003C84FFFEABB5F7_1_colortemperature updated to UNDEF
[DEBUG] [converter.ZigBeeBaseChannelConverter] - 003C84FFFEABB5F7: Channel zigbee:device:dc80d093a7:003c84fffeabb5f7:003C84FFFEABB5F7_1_color updated to 123.30708,44.88189,50
[DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 003C84FFFEABB5F7: Updating ZigBee channel state zigbee:device:dc80d093a7:003c84fffeabb5f7:003C84FFFEABB5F7_1_colortemperature to UNDEF
[DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 003C84FFFEABB5F7: Updating ZigBee channel state zigbee:device:dc80d093a7:003c84fffeabb5f7:003C84FFFEABB5F7_1_color to 123.30708,44.88189,50
[DEBUG] [.zigbee.handler.ZigBeeIsAliveTracker] - IsAlive Tracker scheduled task for thingUID=zigbee:device:dc80d093a7:003c84fffeabb5f7 in 14430 seconds
[INFO ] [openhab.event.ItemStateChangedEvent ] - Item 'OfficeLedstripWall_Color' changed from 123,45,50 to 123.30708,44.88189,50
[DEBUG] [.zigbee.handler.ZigBeeIsAliveTracker] - IsAlive Tracker scheduled task for thingUID=zigbee:device:dc80d093a7:003c84fffeabb5f7 in 14430 seconds
```

**Setting a color temperature (which will turn the _Color_ channel UNDEF, even though the light is still on)**
```
openhab>  openhab:send OfficeLedstripWall_ColorTemperature 50

[INFO ] [openhab.event.ItemCommandEvent      ] - Item 'OfficeLedstripWall_ColorTemperature' received command 50
[INFO ] [penhab.event.ItemStatePredictedEvent] - Item 'OfficeLedstripWall_ColorTemperature' predicted to become 50
[INFO ] [openhab.event.ItemStateChangedEvent ] - Item 'OfficeLedstripWall_ColorTemperature' changed from UNDEF to 50
[DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 003C84FFFEABB5F7: Command for channel zigbee:device:dc80d093a7:003c84fffeabb5f7:003C84FFFEABB5F7_1_colortemperature --> 50 [PercentType]
[DEBUG] [converter.ZigBeeBaseChannelConverter] - 003C84FFFEABB5F7: Channel zigbee:device:dc80d093a7:003c84fffeabb5f7:003C84FFFEABB5F7_1_colortemperature waiting for response to 50
[DEBUG] [converter.ZigBeeBaseChannelConverter] - 003C84FFFEABB5F7: Channel zigbee:device:dc80d093a7:003c84fffeabb5f7:003C84FFFEABB5F7_1_colortemperature received SUCCESS in response to 50
[DEBUG] [.converter.ZigBeeConverterColorColor] - 003C84FFFEABB5F7: ZigBee attribute reports ZclAttribute [cluster=Color Control, id=8, name=Color Mode, dataType=ENUMERATION_8_BIT, lastValue=2, lastReportTime=Thu Aug 01 16:35:15 CEST 2024, implemented=false]
[INFO ] [openhab.event.ItemStateChangedEvent ] - Item 'OfficeLedstripWall_Color' changed from 123.30708,44.88189,50 to UNDEF
[DEBUG] [rter.ZigBeeConverterColorTemperature] - 003C84FFFEABB5F7: ZigBee attribute reports ZclAttribute [cluster=Color Control, id=8, name=Color Mode, dataType=ENUMERATION_8_BIT, lastValue=2, lastReportTime=Thu Aug 01 16:35:15 CEST 2024, implemented=false]  on endpoint 1
[DEBUG] [converter.ZigBeeBaseChannelConverter] - 003C84FFFEABB5F7: Channel zigbee:device:dc80d093a7:003c84fffeabb5f7:003C84FFFEABB5F7_1_color updated to UNDEF
[... 95 lines cut ...]
[DEBUG] [.zigbee.handler.ZigBeeIsAliveTracker] - IsAlive Tracker scheduled task for thingUID=zigbee:device:dc80d093a7:001788010c5ac808 in 14430 seconds
[INFO ] [openhab.event.ItemStateChangedEvent ] - Item 'ColorlightLivingroomDresser_Color' changed from 0,0,0 to 0,0,100
[INFO ] [openhab.event.ItemStateChangedEvent ] - Item 'ColorlightLivingroomDresser_Dimmer' changed from 0 to 100
[DEBUG] [.converter.ZigBeeConverterColorColor] - 001788010C5AC808: ZigBee attribute reports ZclAttribute [cluster=Level Control, id=0, name=Current Level, dataType=UNSIGNED_8_BIT_INTEGER, lastValue=12, lastReportTime=Thu Aug 01 16:49:48 CEST 2024, implemented=true]
[DEBUG] [.converter.ZigBeeConverterColorColor] - 001788010C5AC808: ZigBee attribute reports ZclAttribute [cluster=On/Off, id=0, name=On Off, dataType=BOOLEAN, lastValue=true, lastReportTime=Thu Aug 01 16:49:48 CEST 2024, implemented=true]
[DEBUG] [.converter.ZigBeeConverterColorColor] - 001788010C5AC808: ZigBee attribute reports ZclAttribute [cluster=Level Control, id=0, name=Current Level, dataType=UNSIGNED_8_BIT_INTEGER, lastValue=254, lastReportTime=Thu Aug 01 16:49:49 CEST 2024, implemented=true]

[INFO ] [openhab.event.ItemCommandEvent      ] - Item 'ColorlightLivingroomDresser_Color' received command REFRESH
[DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 001788010C5AC808: Command for channel zigbee:device:dc80d093a7:001788010c5ac808:001788010C5AC808_11_color --> REFRESH [RefreshType]
[DEBUG] [.converter.ZigBeeConverterColorColor] - 001788010C5AC808: ZigBee attribute reports ZclAttribute [cluster=On/Off, id=0, name=On Off, dataType=BOOLEAN, lastValue=true, lastReportTime=Thu Aug 01 16:50:04 CEST 2024, implemented=true]
[... cut ...]
[INFO ] [openhab.event.ItemStateChangedEvent ] - Item 'ColorlightLivingroomDresser_Dimmer' changed from 100 to UNDEF
[DEBUG] [.converter.ZigBeeConverterColorColor] - 001788010C5AC808: ZigBee attribute reports ZclAttribute [cluster=Level Control, id=0, name=Current Level, dataType=UNSIGNED_8_BIT_INTEGER, lastValue=254, lastReportTime=Thu Aug 01 16:50:04 CEST 2024, implemented=true]
```

**Note:** In most examples you will see a `_Dimmer` item follow along. This is just a Dimmer item linked to the _Color_ channel for easy sending of brightness values. I have tried the examples without this link as well.

andrewfg, 2024-09-14:
@m4rcu5 I am thinking of writing a PR to address this. What I think you are asking for is that when someone changes the color temperature (CT) channel, then the color channel should be set also to the color that matches the CT on the CT locus in the color XY space. That would be doable. However I am concerned that is such CT channel command would be sent to the lamp and thus force it to switch from CT mode to color mode. And that obviously is not desirable at all. So perhaps you can clarify what you need?

m4rcu5, 2024-09-14:
Hi @andrewfg, I think that setting the Color channel to a representative of the ColorTemperature would be really nice (more akin how Google Home handles it), but would be over achieving my goal.
For me, I would be happy if the Color channel would return `0,0,<brightness>` instead of UNDEF when the CT value is changed, or when a poll happens. This would be really close to its current behavior.
To be clear, that 0 Hue and 0 Saturation is not send to the device, but seems to be currently returned by the Color channel when setting the brightness while in HSB mode.

So currently; If you set CT, the Color is set to UNDEF, setting a brightness afterwards, makes it return 0,0,B. And after the channel is polled it plops back to UNDEF. If we can get rid of UNDEF, and just consistently return that 0,0,B value when bulb is in CT mode, that would be great. If instead of 0H 0S, you could map that to a valid color that looks like the brightness, that would be more awesome, but without knowing the mapping of % (on the CT channel) to kelvin (on the actual lamp), I think that is impossible.

andrewfg, 2024-09-14:
Ok I think reading (polling) may return 4 different state combinations..
- Color is defined, CT not => OH color channel = actual HSB; CT channel = UNDEF
- Color is undefined, CT defined => color channel = HSB of locus of CT on XY chart; CT channel = actual CT
- both defined => both channels = actual value
- both undefined => both channels UNDEF

I think writing (commands) may have following combinations..
- On/off command to color channel => send switch command to lamp
- Dimming command to color channel, and CT is UNDEF => send prior HSB with different B
- Dimming command to color channel, and CT is NOT UNDEF => send 0,0,B ?? .. or send HSB value from CT point in XY space ??

I am concerned about the last bullet point, since that risks to flip the lamp from CT mode to color mode. => WDYT?

m4rcu5, 2024-09-14:
> Ok I think reading (polling) may return 4 different state combinations..
> 
>     * Color is defined, CT not => OH color channel = actual HSB; CT channel = UNDEF
:heavy_check_mark: Sound correct

> 
>     * Color is undefined, CT defined => color channel = HSB of locus of CT on XY chart; CT channel = actual CT
I am a bit concerned how this HS/XY value would be calculated, as the CT channel itself is a dimmer value, and the temperature in kelvin at 0% and 100% are only known to the hardware itself (or if you set them in the google home metadata)

> 
>     * both defined => both channels = actual value
I do not think this is a valid combination. The bulb is either in Color or ColorTemperature mode. Digging in the current code, it seems `clusterColorControl` is what defines its state.

> 
>     * both undefined => both channels UNDEF
:heavy_check_mark: sounds logical.

> 
> 
> I think writing (commands) may have following combinations..
> 
>     * On/off command to color channel => send switch command to lamp
Yes, currently it seems to use a ClusterOnOff command.
```    private Future<CommandResult> changeOnOff(OnOffType onoff) throws InterruptedException, ExecutionException {
        boolean on = onoff == OnOffType.ON;

        if (clusterOnOff == null) {
            if (clusterLevelControl == null) {
                logger.warn("{}: ignoring on/off command", endpoint.getIeeeAddress());

                return null;
            } else {
                return changeBrightness(on ? PercentType.HUNDRED : PercentType.ZERO);
            }
        }

        ZclOnOffCommand command;
        if (on) {
            command = new OnCommand();
        } else {
            command = new OffCommand();
        }

        return clusterOnOff.sendCommand(command);
    }
``

> 
>     * Dimming command to color channel, and CT is UNDEF => send prior HSB with different B
Then dimming, I think it sends a ClusterLevelControl command, and not the full HSB values

```
    private Future<CommandResult> changeBrightness(PercentType brightness)
            throws InterruptedException, ExecutionException {
        if (clusterLevelControl == null) {
            if (clusterOnOff == null) {
                logger.warn("{}: ignoring brightness command", endpoint.getIeeeAddress());
                return null;
            } else {
                return changeOnOff(brightness.intValue() == 0 ? OnOffType.OFF : OnOffType.ON);
            }
        }

        int level = percentToLevel(brightness);

        ZclLevelControlCommand command;
        if (clusterOnOff != null) {
            if (brightness.equals(PercentType.ZERO)) {
                return clusterOnOff.sendCommand(new OffCommand());
            } else {
                command = new MoveToLevelWithOnOffCommand(level, configLevelControl.getDefaultTransitionTime());
            }
        } else {
            command = new MoveToLevelCommand(level, configLevelControl.getDefaultTransitionTime());
        }

        return clusterLevelControl.sendCommand(command);
    }
```

> 
>     * Dimming command to color channel, and CT is NOT UNDEF => send 0,0,B ?? .. or send HSB value from CT point in XY space ??
So the same here. If you would send the HSB command, it will go back to color mode. But I think the issue is with the channel updates and not the commands send to the device itself.

In my debugging, the commands to the device are OK, but the representation on the openHAB side is wrong. I'm trying to follow the code and see where updates to the channel are send.

> 
> 
> I am concerned about the last bullet point, since that risks to flip the lamp from CT mode to color mode. => WDYT?

andrewfg, 2024-09-14:
> the CT channel itself is a dimmer value, and the temperature in kelvin at 0% and 100% are only known to the hardware itself

In Philips Hue there is a default of 500 Mirek = 0% and 153 Mirek = 100% .. so I would use that as basis. And we could eventually add a config param for Min/Max Mirek to cover more exotic lamps.

cdjackson, 2024-09-14:
> I am a bit concerned how this HS/XY value would be calculated, as the CT channel itself is a dimmer value, and the temperature in kelvin at 0% and 100% are only known to the hardware itself

That's not correct - the binding knows the min/max - otherwise it wouldn't be able to set the colour temperature which is in mired. The binding uses the min/max values to convert the percentage to mired.

> In Philips Hue there is a default of 500 Mirek = 0% and 153 Mirek = 100% .. so I would use that as basis.

You shouldn't just guess the min/max - just read the min/max mired from the device. Note that if the device doesn't provide the min/max mired, then the binding has some internal defaults, and those should be used - not another set. Otherwise things will be inconsistent.

andrewfg, 2024-09-15:
> if the device doesn't provide the min/max mired, then the binding has some internal defaults, and those should be used

> In Philips Hue there is a default of 500 Mirek = 0% and 153 Mirek = 100%

@cdjackson just for the avoidance of doubt the Zigbee binding default CT range is are 2000 K -- 6500 K .. which is .. (tada) .. 500 Mirek -- 153 Mirek .. so there would not have been any risk of using something other than that :)

cdjackson, 2024-09-15:
Ok, that's fine - it happens to be the same in this instance (which is good :) ), but my point is we should use a single set of constants to avoid any confusion.

andrewfg, 2024-09-15:
@cdjackson in any case the code that I would add does not need to check or process the min/max mired capabilities of the lamp. Reason is that my added code would respond to the CT `attributeUpdated()` event which contains a mired value from the lamp (so obviously anything in that event must be within range of the respective lamp.

---

HOWEVER on a related note: I see that your ZigBeeConverterColorTemperature class currently converts UI dimmer percentages => mired (which is correct) .. but it actually does the conversion percent => kelvin => mired -- which I think is incorrect. Kelvin is too non linear to be useful for things like UI sliders, so the inverse mired was invented so that a 0%..100% control (in mired) gives a more linear change of perceived CT. Therefore I suggest another PR needs to be made so that there is a linear relationship percent <=> mired (i.e. in fact to ignore the Kelvin code entirely). => WDYT?

cdjackson, 2024-09-16:
> @cdjackson in any case the code that I would add does not need to check or process the min/max mired capabilities of the lamp. Reason is that my added code would respond to the CT attributeUpdated() [...]

Agreed

> Therefore I suggest another PR needs to be made so that there is a linear relationship percent <=> mired (i.e. in fact to ignore the Kelvin code entirely). => WDYT?

I'm going to defer to your much greater wisdom in this area. I just had a look at some of the code and I note that `miredToKelvin` and `kelvinToMired` are exactly the same, which kinda feels like it's wrong :)

I'm certainly happy to look at a PR that tidies this up. I personally don't use any colour lamps so it's not something I'm going to get a lot of test milage on.

Oh - I also looked briefly at your other PR, and it looks good at first glance. Thanks.

andrewfg, 2024-09-16:
> miredToKelvin and kelvinToMired are exactly the same, which kinda feels like it's wrong

I will have a look at it. I suppose that the "proper" method via the OH core UoM. Can you perhaps hold my PR until I have something more definitive to say about this?

> not something I'm going to get a lot of test milage on.

Yup. My lights are all on Hue, so I just bought a plain generic Zigbee lamp to run more tests on.

andrewfg, 2024-09-16:
^
@cdjackson actually the current 'miredToKelvin' and 'kelvinToMired' code is quite Ok (i.e. the same calculation does apply for the conversion both ways) .. although there is a round trip loss of precision due to the code using 'int' values.

And the "new" UoM way is as follows .. albeit it yields the same results..

```
   private int kelvinToMired(int kelvin) {
        return QuantityType.valueOf(kelvin, Units.KELVIN).toInvertibleUnit(Units.MIRED).intValue();
    }

    private int miredToKelvin(int mired) {
        return QuantityType.valueOf(mired, Units.MIRED).toInvertibleUnit(Units.KELVIN).intValue();
    }
```

andrewfg, 2024-09-16:
@cdjackson while we are at it, I wonder if we should address this one too..

https://github.com/openhab/org.openhab.binding.zigbee/blob/95097906d4d7f5c78c91a5b4daef64a721ccdc03/org.openhab.binding.zigbee/src/main/java/org/openhab/binding/zigbee/internal/converter/ZigBeeConverterColorTemperature.java#L137

PR #855 (2024-10-05): [zigbee] Map colour temperature states to color HSB states
Previously, when a lamp device was operated in color temperature mode, the color channel/item state within OH and its respective UIs would be shown as UNDEF.

Now, with this PR when a lamp is operated in color temperature mode, the respective color temperature is converted to an actual valid color that is derived from the color temperature point on the color temperature locus on the CIE color xy chart. See curved line on CIE chart below.

**NOTA BENE**: this calculated HSB state is **NOT** sent to the remote device (since that would le [... cut]
closed 2024-10-05 by cdjackson