openhab/org.openhab.binding.zigbee #871: Support for Nous E8 Smoke Detector

jangrewe, 2024-11-06:
### Outline

I've just received a Nous E8 Smoke Detector that is found and can be added when running a scan, but unfortunately results in this:

[image]

If it's of any help, Z2M recently added support for it: https://www.zigbee2mqtt.io/devices/E8.html 

And: https://github.com/Koenkk/zigbee2mqtt/issues/23941 

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used | SONOFF Zigbee 3.0 USB Dongle Plus V2 (E) |
| openHAB version  | 4.2.2 |
| Hardware         | amd64 |
| Memory           |  128G |
| Java version     |  whatever is in the official Docker image |
| Devices          |  see below |

```
openhab> zigbee nodes
Total known nodes in network: 2
Network  Addr  IEEE Address      Logical Type  State      EP   Profile                    Device Type                Manufacturer     Model
      0  0000  8C65A3FFFE525D1A  COORDINATOR   ONLINE
  38527  967F  583BC2FFFE02DC5A  END_DEVICE    ONLINE      1  ZIGBEE_HOME_AUTOMATION     SMART_PLUG                 _TZE284_0zaf1cr8  TS0601
```

### Logs
```
2024-11-06 02:23:40.537 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=6, ackNum=7, reTx=false, data=BE 00 01 18 00]
2024-11-06 02:23:40.545 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=7, ackNum=7, reTx=false, data=BE 80 01 18 00 02]
2024-11-06 02:23:40.549 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=0, notRdy=false]
2024-11-06 02:23:41.056 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 583BC2FFFE02DC5A: Initializing ZigBee thing handler zigbee:device:c5894321b4:583bc2fffe02dc5a
2024-11-06 02:23:41.056 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 583BC2FFFE02DC5A: Coordinator status changed to ONLINE.
2024-11-06 02:23:41.056 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 583BC2FFFE02DC5A: Coordinator is ONLINE. Starting device initialisation.
2024-11-06 02:23:41.057 [DEBUG] [pp.discovery.ZigBeeNetworkDiscoverer] - 583BC2FFFE02DC5A: NWK Discovery starting node rediscovery
2024-11-06 02:23:41.057 [DEBUG] [pp.discovery.ZigBeeNetworkDiscoverer] - 583BC2FFFE02DC5A: NWK Discovery: Rediscovery using unicast to 967F
2024-11-06 02:23:41.057 [DEBUG] [e.transaction.ZigBeeTransactionQueue] - 583BC2FFFE02DC5A: Added transaction to queue, len=1, transaction=ZigBeeTransaction [ieeeAddress=583BC2FFFE02DC5A queueTime=0, state=WAITING, sendCnt=0, command=NetworkAddressRequest [0000/0 -> 967F/0, cluster=0000, TID=--, ieeeAddr=583BC2FFFE02DC5A, requestType=0, startIndex=0]]
2024-11-06 02:23:41.058 [DEBUG] [transaction.ZigBeeTransactionManager] - Transaction Manager: Send Next transaction. outstandingTransactions=0, outstandingQueues=1, sleepy=0/3
2024-11-06 02:23:41.058 [DEBUG] [transaction.ZigBeeTransactionManager] - 967F/0: Sending ZigBeeTransaction [ieeeAddress=583BC2FFFE02DC5A queueTime=1, state=WAITING, sendCnt=0, command=NetworkAddressRequest [0000/0 -> 967F/0, cluster=0000, TID=11, ieeeAddr=583BC2FFFE02DC5A, requestType=0, startIndex=0]]
2024-11-06 02:23:41.058 [DEBUG] [transaction.ZigBeeTransactionManager] - transactionListenerAdded: 1 outstanding
2024-11-06 02:23:41.058 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX CMD: NetworkAddressRequest [0000/0 -> 967F/0, cluster=0000, TID=11, ieeeAddr=583BC2FFFE02DC5A, requestType=0, startIndex=0]
2024-11-06 02:23:41.059 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0000/0, destinationAddress=967F/0, profile=0000, cluster=0000, addressMode=DEVICE, radius=8, apsSecurity=false, ackRequest=true, apsCounter=1B, rssi=--, lqi=--, payload=11 5A DC 02 FE FF C2 3B 58 00 00]
2024-11-06 02:23:41.059 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - ASH TX EZSP: EzspSendUnicastRequest [networkId=0, type=EMBER_OUTGOING_DIRECT, indexOrDestination=967F, apsFrame=EmberApsFrame [profileId=0000, clusterId=0000, sourceEndpoint=0, destinationEndpoint=0, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY, EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY], groupId=0, sequence=1B], messageTag=11, messageContents=11 5A DC 02 FE FF C2 3B 58 00 00]
2024-11-06 02:23:41.060 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=7, ackNum=0, reTx=false, data=BF 00 01 34 00 00 7F 96 00 00 00 00 00 00 40 11 00 00 1B 11 0B 11 5A DC 02 FE FF C2 3B 58 00 00]
2024-11-06 02:23:41.067 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - 583BC2FFFE02DC5A: Start initialising ZigBee Thing handler
2024-11-06 02:23:41.067 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 583BC2FFFE02DC5A: ZigBee node property discovery start
2024-11-06 02:23:41.067 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 583BC2FFFE02DC5A: ZigBee node property discovery using basic cluster on endpoint 967F/1
2024-11-06 02:23:41.068 [DEBUG] [scovery.ZigBeeNodePropertyDiscoverer] - 583BC2FFFE02DC5A: ZigBee node property discovery using OTA cluster on endpoint 967F/1
2024-11-06 02:23:41.068 [DEBUG] [e.transaction.ZigBeeTransactionQueue] - 583BC2FFFE02DC5A: Added transaction to queue, len=1, transaction=ZigBeeTransaction [ieeeAddress=583BC2FFFE02DC5A queueTime=0, state=WAITING, sendCnt=0, command=ReadAttributesCommand [Ota Upgrade: 0000/0 -> 967F/1, cluster=0019, TID=--, identifiers=[2]]]
2024-11-06 02:23:41.068 [DEBUG] [transaction.ZigBeeTransactionManager] - Transaction Manager: Send Next transaction. outstandingTransactions=1, outstandingQueues=1, sleepy=1/3
2024-11-06 02:23:41.072 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=0, ackNum=0, reTx=false, data=BF 80 01 34 00 00 E4]
2024-11-06 02:23:41.072 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=1, notRdy=false]
2024-11-06 02:23:41.072 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspSendUnicastResponse [networkId=0, status=EMBER_SUCCESS, sequence=E4]
2024-11-06 02:23:41.109 [DEBUG] [transaction.ZigBeeTransactionManager] - Transaction Manager: Send Next transaction. outstandingTransactions=1, outstandingQueues=1, sleepy=1/3
2024-11-06 02:23:42.018 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=1, ackNum=0, reTx=false, data=BF 90 01 45 00 00 00 00 00 80 00 00 40 01 00 00 D4 F0 D8 7F 96 FF FF 0C 11 00 5A DC 02 FE FF C2 3B 58 7F 96 02]
2024-11-06 02:23:42.019 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=2, notRdy=false]
2024-11-06 02:23:42.019 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingMessageHandler [networkId=0, type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=0000, clusterId=8000, sourceEndpoint=0, destinationEndpoint=0, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY], groupId=0, sequence=D4], lastHopLqi=240, lastHopRssi=-40, sender=967F, bindingIndex=255, addressIndex=255, messageContents=11 00 5A DC 02 FE FF C2 3B 58 7F 96]
2024-11-06 02:23:42.019 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=967F/0, destinationAddress=0000/0, profile=0000, cluster=8000, addressMode=DEVICE, radius=0, apsSecurity=false, ackRequest=false, apsCounter=D4, rssi=-40, lqi=F0, payload=11 00 5A DC 02 FE FF C2 3B 58 7F 96]
2024-11-06 02:23:42.019 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - 583BC2FFFE02DC5A: Node update. NWK Address=NULL
2024-11-06 02:23:42.020 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - 583BC2FFFE02DC5A: Node 967F is not updated from ZigBeeNode [state=ONLINE, IEEE=583BC2FFFE02DC5A, NWK=----, endpoints=[]]
2024-11-06 02:23:42.020 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: NetworkAddressResponse [967F/0 -> 0000/0, cluster=8000, TID=11, status=SUCCESS, ieeeAddrRemoteDev=583BC2FFFE02DC5A, nwkAddrRemoteDev=967F, startIndex=null, nwkAddrAssocDevList=[]]
2024-11-06 02:23:42.020 [DEBUG] [transaction.ZigBeeTransactionManager] - notifyTransactionCommand: NetworkAddressResponse [967F/0 -> 0000/0, cluster=8000, TID=11, status=SUCCESS, ieeeAddrRemoteDev=583BC2FFFE02DC5A, nwkAddrRemoteDev=967F, startIndex=null, nwkAddrAssocDevList=[]]
2024-11-06 02:23:42.020 [DEBUG] [pp.discovery.ZigBeeNetworkDiscoverer] - 583BC2FFFE02DC5A: NWK Discovery NetworkAddressRequest returned from NetworkAddressResponse [967F/0 -> 0000/0, cluster=8000, TID=11, status=SUCCE [... cut ...]
2024-11-06 02:23:48.625 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=4, notRdy=false]
2024-11-06 02:23:49.625 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameData [frmNum=7, ackNum=4, reTx=false, data=C7 00 01 18 00]
2024-11-06 02:23:49.633 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - <-- RX ASH frame: AshFrameData [frmNum=4, ackNum=0, reTx=false, data=C7 80 01 18 00 02]
2024-11-06 02:23:49.637 [DEBUG] [e.ember.internal.ash.AshFrameHandler] - --> TX ASH frame: AshFrameAck [ackNum=5, notRdy=false]
```

FranzSchi, 2024-11-06:
Here is a similar issue: https://community.openhab.org/t/new-ugly-zigbee-device-humidity-temperature-clock-with-clusters-000a-time-and-ef00-tuya-ts0601-tze200-c7emyjom/131783/11

jangrewe, 2024-11-06:
Sure, i'm running `6.10.3.0`.

cdjackson, 2024-11-16:
Normally when there's no supported clusters, it means the device is either non-standard, or just does something that the binding doesn't support. If it's doing things in a standard way, then I'm happy to look at adding support.

The log doesn't really provide a lot of useful info. Please can you run the fingerprint command from the console -:

`openhab:zigbee fingerprint 0x967F`

jangrewe, 2024-11-17:
Sure, here's the output:
```
openhab> openhab:zigbee fingerprint 0x967F
|>| Node Descriptor
| |> Logical Type               END_DEVICE
| |> MAC Capabilities           [REDUCED_FUNCTION_DEVICE]
| |> Stack Compliance           22
| |> Server Capabilities        []
| |> Buffer Size                82
| |> Incoming Transfer Size     82
| |> Outgoing Transfer Size     82
|
|>| Power Descriptor
| |> Available Power Sources    [MAINS]
| |> Current Power Source       MAINS
| |> Current Power Mode         RECEIVER_ON_IDLE
| |> Power Level                FULL
|
|>| ZDO
| |> ManagementBindRequest      SUCCESS
| |> IeeeAddressRequest         SUCCESS
| |> ManagementLqiRequest       SUCCESS
| |> ManagementRoutingRequest   SUCCESS
|
|>| Basic Information
| |> Generic Device Class
| |> Generic Device Type
| |> Manufacturer Name          _TZE284_0zaf1cr8
| |> Model Identifier           TS0601
| |> Product Code
| |> Product URL
| |> Date Code
| |> Application Version        67
| |> Software Build ID
| |> Hardware Version           1
| |> Zcl Version                3
| |> Stack Version              0
| |
| |>| Endpoint 1
| | |> Profile                  0104  ZIGBEE_HOME_AUTOMATION
| | |> Device Type              0051  SMART_PLUG
| | |> Device Version           1
| | |
| | |>| Input Clusters
| | | |
| | | |>| Cluster 0000 Basic
| | | | |> Type                 Server [Input]
| | | | |> Manufacturer Spec.   No
| | | | |
| | | | |>| Commands Generated
| | | | | |> FAILURE
| | | | |
| | | | |>| Commands Received
| | | | | |> FAILURE
| | | | |
| | | | |>| Attributes Supported
| | | | | |> 0000  ZCL Version                               >> UNSIGNED_8_BIT_INTEGER          3
| | | | | |> 0001  Application Version                       >> UNSIGNED_8_BIT_INTEGER          67
| | | | | |> 0002  Stack Version                             >> UNSIGNED_8_BIT_INTEGER          0
| | | | | |> 0003  HW Version                                >> UNSIGNED_8_BIT_INTEGER          1
| | | | | |> 0004  Manufacturer Name                         >> CHARACTER_STRING                _TZE284_0zaf1cr8
| | | | | |> 0005  Model Identifier                          >> CHARACTER_STRING                TS0601
| | | | | |> 0006  Date Code                                 >> CHARACTER_STRING
| | | | | |> 0007  Power Source                              >> ENUMERATION_8_BIT               3
| | | | | |> FFDE                                            >> UNSIGNED_8_BIT_INTEGER          11
| | | | | |> FFFD  Cluster Revision                          >> UNSIGNED_16_BIT_INTEGER         2
| | | | | |> FFFE  Attribute Reporting Status                >> ENUMERATION_16_BIT              0
| | | |
| | | |>| Cluster 0004 Groups
| | | | |> Type                 Server [Input]
| | | | |> Manufacturer Spec.   No
| | | | |> Unsupported locally
| | | |
| | | |>| Cluster 0005 Scenes
| | | | |> Type                 Server [Input]
| | | | |> Manufacturer Spec.   No
| | | | |
| | | | |>| Commands Generated
| | | | | |> FAILURE
| | | | |
| | | | |>| Commands Received
| | | | | |> FAILURE
| | | | |
| | | | |>| Attributes Supported
| | | | | |> 0000  Scene Count                               >> UNSIGNED_8_BIT_INTEGER          0
| | | | | |> 0001  Current Scene                             >> UNSIGNED_8_BIT_INTEGER          0
| | | | | |> 0002  Current Group                             >> UNSIGNED_16_BIT_INTEGER         0
| | | | | |> 0003  Scene Valid                               >> BOOLEAN                         false
| | | | | |> 0004  Name Support                              >> BITMAP_8_BIT                    0
| | | | | |> FFFD  Cluster Revision                          >> UNSIGNED_16_BIT_INTEGER         2
| | | |
| | | |>| Cluster ED00
| | | | |> Type                 Server [Input]
| | | | |> Manufacturer Spec.   No
| | | | |> Unsupported locally
| | | |
| | | |>| Cluster EF00
| | | | |> Type                 Server [Input]
| | | | |> Manufacturer Spec.   No
| | | | |> Unsupported locally
| | |
| | |>| Output Clusters
| | | |
| | | |>| Cluster 000A Time
| | | | |> Type                 Client [Output]
| | | | |> Manufacturer Spec.   No
| | | | |> Unsupported locally
| | | |
| | | |>| Cluster 0019 Ota Upgrade
| | | | |> Type                 Client [Output]
| | | | |> Manufacturer Spec.   No
| | | | |
| | | | |>| Commands Generated
| | | | | |> FAILURE
| | | | |
| | | | |>| Commands Received
| | | | | |> FAILURE
| | | | |
| | | | |>| Attributes Supported
| | | | | |> 0000  Upgrade Server ID                         >> IEEE_ADDRESS                    FFFFFFFFFFFFFFFF
| | | | | |> 0001  File Offset                               >> UNSIGNED_32_BIT_INTEGER         -1
| | | | | |> 0006  Image Upgrade Status                      >> ENUMERATION_8_BIT               0
| | | | | |> FFFD  Cluster Revision                          >> UNSIGNED_16_BIT_INTEGER         3

```