openhab/org.openhab.binding.zigbee #878: Trigger Channels for RWL21 no longer working, After update OH4.2.2 => OH4.3.0

FranzSchi, 2024-11-19:
I have a philips RWL021 Switch.
This dimmer control is löinked in rules  like shown here

```
configuration: {}
triggers:
  - id: "1"
    configuration:
      channelUID: zigbee:philips_rwl021:62ec522f14:001788011048be20:buttonI
    type: core.ChannelEventTrigger
  - id: "3"
    configuration:
      thingUID: zigbee:philips_rwl021:62ec522f14:001788011048be20
      channelUID: zigbee:philips_rwl021:62ec522f14:001788011048be20:buttonO
    type: core.ChannelEventTrigger
```

after an inital update on console with:
bundle:update -r org.openhab.binding.zigbee/4.3.0.M4 https://openhab.jfrog.io/artifactory/libs-snapshot-local/org/openhab/addons/bundles/org.openhab.binding.zigbee/4.3.0-SNAPSHOT/org.openhab.binding.zigbee-4.3.0-20241118.033607-16.jar

I've seen this error:
```
java.lang.StackOverflowError: null
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:835) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
        at org.openhab.binding.zigbee.handler.ZigBeeThingHandler.triggerChannel(ZigBeeThingHandler.java:836) ~[?:?]
---
```

Then I've also updated the other Plugins and restarted the docker and enabled the logging.
Result: I could no longer this error but now i see lots of "button pressed short" messags.

And for sure the button pressed is not recognized 

See attached log:
[OHLog.zip](https://github.com/user-attachments/files/17813606/OHLog.zip)

cdjackson, 2024-11-19:
Yes, thanks. Not sure why that changed but it seems it was accidentally removed.

FranzSchi, 2024-11-21:
So, tested and it’s working!

closed 2025-04-18 by cdjackson