openhab/org.openhab.binding.zigbee #881: tried updating channel <your channel here> although the handler was already disposed

FranzSchi, 2024-12-11:
### Outline

I see quite often the entry "Handler ZigBeeThingHandler of thing zigbee:device:62ec522f14:a4c1384309cf08d7 tried updating channel A4C1384309CF08D7_1_meteringsumdelivered although the handler was already disposed." 
can this be avoided?

### Configuration

| Configuration    | Description                                                                    |
|------------------|--------------------------------------------------------------------------------|
| Coordinator used |  Ember  6.10.7.0  , EFR32MG2                                      |
| openHAB version  | 4.3.0.M4 |
| Hardware         |   Raspberry Pi                                                                              |
| Memory           | 8 GB                                                                               |
| Java version     |  Docker Image    17.0.13 (undefined)     |
| Devices          | see log                                   |

```
256 x Active x  80 x 1.4.14                x com.zsmartsystems.zigbee
257 x Active x  80 x 1.4.14                x com.zsmartsystems.zigbee.console
258 x Active x  80 x 1.4.14                x com.zsmartsystems.zigbee.console.ember
259 x Active x  80 x 1.4.14                x com.zsmartsystems.zigbee.console.telegesis
260 x Active x  80 x 1.4.14                x com.zsmartsystems.zigbee.dongle.cc2531
261 x Active x  80 x 1.4.14                x com.zsmartsystems.zigbee.dongle.ember
262 x Active x  80 x 1.4.14                x com.zsmartsystems.zigbee.dongle.telegesis
263 x Active x  80 x 1.4.14                x com.zsmartsystems.zigbee.dongle.xbee
312 x Active x  80 x 4.3.0.202411200337    x openHAB Add-ons :: Bundles :: ZigBee Binding
313 x Active x  80 x 4.3.0.M4              x openHAB Add-ons :: Bundles :: ZigBee CC2531 Bridge
314 x Active x  80 x 4.3.0.M4              x openHAB Add-ons :: Bundles :: ZigBee Console
315 x Active x  80 x 4.3.0.M4              x openHAB Add-ons :: Bundles :: ZigBee Console Ember
316 x Active x  80 x 4.3.0.M4              x openHAB Add-ons :: Bundles :: ZigBee Console Telegesis
317 x Active x  80 x 4.3.0.M4              x openHAB Add-ons :: Bundles :: ZigBee Ember Bridge
318 x Active x  80 x 4.3.0.M4              x openHAB Add-ons :: Bundles :: ZigBee Firmware Provider
319 x Active x  80 x 4.3.0.M4              x openHAB Add-ons :: Bundles :: ZigBee Serial Driver
320 x Active x  80 x 4.3.0.M4              x openHAB Add-ons :: Bundles :: ZigBee SLZB06 Bridge
321 x Active x  80 x 4.3.0.M4              x openHAB Add-ons :: Bundles :: ZigBee Telegesis Bridge
322 x Active x  80 x 4.3.0.M4              x openHAB Add-ons :: Bundles :: ZigBee XBee Bridge
```

### Logs

```
2024-12-11 16:49:07.098 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspSendUnicastResponse [networkId=0, status=EMBER_SUCCESS, sequence=A8]
2024-12-11 16:49:07.118 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingRouteRecordHandler [networkId=0, source=E027, sourceEui=A4C1384309CF08D7, lastHopLqi=120, lastHopRssi=-70, relayList=]
2024-12-11 16:49:07.139 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspMessageSentHandler [networkId=0, type=EMBER_OUTGOING_DIRECT, indexOrDestination=E027, apsFrame=EmberApsFrame [profileId=0104, clusterId=0B04, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY, EMBER_APS_OPTION_RETRY], groupId=0, sequence=A8], messageTag=4E, status=EMBER_SUCCESS, messageContents=]
2024-12-11 16:49:07.139 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX STA: msgTag=4E state=RX_ACK
2024-12-11 16:49:07.139 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingMessageHandler [networkId=0, type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=0104, clusterId=0B04, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY], groupId=0, sequence=B8], lastHopLqi=120, lastHopRssi=-70, sender=E027, bindingIndex=255, addressIndex=255, messageContents=18 4E 01 0B 05 00 29 00 00]
2024-12-11 16:49:07.139 [DEBUG] [transaction.ZigBeeTransactionManager] - notifyTransactionProgress: TID=4E, state=RX_ACK, outstanding=2
2024-12-11 16:49:07.139 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=E027/1, destinationAddress=0000/1, profile=0104, cluster=0B04, addressMode=DEVICE, radius=0, apsSecurity=false, ackRequest=false, apsCounter=B8, rssi=-70, lqi=78, payload=18 4E 01 0B 05 00 29 00 00]
2024-12-11 16:49:07.139 [DEBUG] [zigbee.transaction.ZigBeeTransaction] - Transaction state changed: nwk=E027, TID=4E, event=RX_ACK, state=ACKED
2024-12-11 16:49:07.140 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - A4C1384309CF08D7: Node update. NWK Address=NULL
2024-12-11 16:49:07.140 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - A4C1384309CF08D7: Node E027 is not updated from ZigBeeNode [state=ONLINE, IEEE=A4C1384309CF08D7, NWK=----, endpoints=[]]
2024-12-11 16:49:07.140 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=SERVER_TO_CLIENT, disableDefaultResponse=true, manufacturerCode=0, sequenceNumber=4E, commandId=1]
2024-12-11 16:49:07.140 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReadAttributesResponse [Electrical Measurement: E027/1 -> 0000/1, cluster=0B04, TID=4E, records=[ReadAttributeStatusRecord [status=SUCCESS, attributeIdentifier=1291, attributeDataType=SIGNED_16_BIT_INTEGER, attributeValue=0]]]
2024-12-11 16:49:07.140 [DEBUG] [transaction.ZigBeeTransactionManager] - notifyTransactionCommand: ReadAttributesResponse [Electrical Measurement: E027/1 -> 0000/1, cluster=0B04, TID=4E, records=[ReadAttributeStatusRecord [status=SUCCESS, attributeIdentifier=1291, attributeDataType=SIGNED_16_BIT_INTEGER, attributeValue=0]]] 
2024-12-11 16:49:07.141 [DEBUG] [transaction.ZigBeeTransactionManager] - Transaction complete: ZigBeeTransaction [ieeeAddress=A4C1384309CF08D7 queueTime=66, state=COMPLETE, sendCnt=1, command=ReadAttributesCommand [Electrical Measurement: 0000/0 -> E027/1, cluster=0B04, TID=4E, identifiers=[1291]]]
2024-12-11 16:49:07.141 [DEBUG] [rter.ZigBeeConverterMeasurementPower] - A4C1384309CF08D7: ZigBee attribute reports ZclAttribute [cluster=Electrical Measurement, id=1291, name=Active Power, dataType=SIGNED_16_BIT_INTEGER, lastValue=0, lastReportTime=Wed Dec 11 16:49:07 CET 2024, implemented=true]
2024-12-11 16:49:07.141 [DEBUG] [transaction.ZigBeeTransactionManager] - transactionListenerRemoved: 1 outstanding
2024-12-11 16:49:07.141 [DEBUG] [converter.ZigBeeBaseChannelConverter] - A4C1384309CF08D7: Channel zigbee:device:62ec522f14:a4c1384309cf08d7:A4C1384309CF08D7_1_activepower updated to 0 W
2024-12-11 16:49:07.141 [DEBUG] [e.transaction.ZigBeeTransactionQueue] - A4C1384309CF08D7: transactionComplete, state=COMPLETE, outstanding=0
2024-12-11 16:49:07.141 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - A4C1384309CF08D7: Updating ZigBee channel state zigbee:device:62ec522f14:a4c1384309cf08d7:A4C1384309CF08D7_1_activepower to 0 W
2024-12-11 16:49:07.142 [DEBUG] [transaction.ZigBeeTransactionManager] - Transaction Manager: Send Next transaction. outstandingTransactions=1, outstandingQueues=0, sleepy=1/3
2024-12-11 16:49:07.142 [DEBUG] [.zigbee.handler.ZigBeeIsAliveTracker] - IsAlive Tracker scheduled task for thingUID=zigbee:device:62ec522f14:a4c1384309cf08d7 in 630 seconds
2024-12-11 16:49:07.711 [DEBUG] [zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingMessageHandler [networkId=0, type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=0104, clusterId=0B04, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY], groupId=0, sequence=5C], lastHopLqi=244, lastHopRssi=-39, sender=5CF6, bindingIndex=255, addressIndex=255, messageContents=18 5A 0A 08 05 21 86 01]
2024-12-11 16:49:07.711 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=5CF6/1, destinationAddress=0000/1, profile=0104, cluster=0B04, addressMode=DEVICE, radius=0, apsSecurity=false, ackRequest=false, apsCounter=5C, rssi=-39, lqi=F4, payload=18 5A 0A 08 05 21 86 01]
2024-12-11 16:49:07.712 [DEBUG] [tsystems.zigbee.ZigBeeNetworkManager] - A4C138C6335C5577: Node update. NWK Address=NULL
[... cut ...]
2024-12-11 16:49:11.719 [DEBUG] [converter.ZigBeeBaseChannelConverter] - A4C138C6335C5577: Channel zigbee:device:62ec522f14:a4c138c6335c5577:A4C138C6335C5577_1_current updated to 371 A
2024-12-11 16:49:11.720 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - A4C138C6335C5577: Updating ZigBee channel state zigbee:device:62ec522f14:a4c138c6335c5577:A4C138C6335C5577_1_current to 371 A
2024-12-11 16:49:11.720 [DEBUG] [.zigbee.handler.ZigBeeIsAliveTracker] - IsAlive Tracker scheduled task for thingUID=zigbee:device:62ec522f14:a4c138c6335c5577 in 1830 seconds
```

cdjackson, 2024-12-11:
It's hard to know what your system is doing, but this means that there is no handler for this thing - the OH core has either shut it down or not started it yet. This can happen during configuration changes, binding system restarts etc. At the end of the day this is a system/core issue.

FranzSchi, 2024-12-12:
Thanks @cdjackson ,
But is it still a core issue if only this cone channel is already disposed?
As i can see the same thing has multiple channels but only that one is affected.
Have a good day
/Franz

cdjackson, 2024-12-12:
Well, the error is coming from the core. The core seems unable to find the thing handler for this thing (as per the error). I agree - it seems strange if there are other channels working - maybe there's something strange with your configuration.  In any case I don't see how (at the moment anyway) that this can be an issue in the binding given it's the core that manages this and there's nothing here that shows any error in the binding.

There is something a little strange here -:

```
2024-12-11 16:49:08.208 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - A4C1384309CF08D7: Updating ZigBee channel state zigbee:device:62ec522f14:a4c1384309cf08d7:A4C1384309CF08D7_1_meteringsumdelivered to 58467.0
2024-12-11 16:49:08.208 [DEBUG] [converter.ZigBeeBaseChannelConverter] - A4C1384309CF08D7: Channel zigbee:device:62ec522f14:a4c1384309cf08d7:A4C1384309CF08D7_1_meteringsumdelivered updated to 58467.0
2024-12-11 16:49:08.209 [DEBUG] [ng.zigbee.handler.ZigBeeThingHandler] - null: Updating ZigBee channel state zigbee:device:62ec522f14:a4c1384309cf08d7:A4C1384309CF08D7_1_meteringsumdelivered to 58467.0
```

This appears to show the same device updating two different channels, but in the second one it doesn't print the address - it prints `null` instead. If these really are the same thing handler, this seems impossible since these are the same log statement - one that knows the address, and one that doesn't. These also seem to be processing the same channel - so my guess is there's something messed up with your system and there are two handlers somehow.

FranzSchi, 2024-12-12:
Thank you Chris, no it’s well better understood.

FranzSchi, 2024-12-14:
Hi @cdjackson ,

Is this log passage similar?
I have an issue here when restarting OH or enable / disable the item.
Practically it leads into that the first Button press (in this case my doorbell) does nothing, only second or later is reacting wich is not sooo nice.

 
```

events.log:2024-12-14 04:00:56.274 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'zigbee:device:62ec522f14:44e2f8fffe34b950' changed from ONLINE to UNINITIALIZED
events.log:2024-12-14 04:00:56.282 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'zigbee:device:62ec522f14:44e2f8fffe34b950' changed from UNINITIALIZED to UNINITIALIZED (DISABLED)

events.log:2024-12-14 04:00:57.290 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'zigbee:device:62ec522f14:44e2f8fffe34b950' changed from UNINITIALIZED (DISABLED) to INITIALIZING
events.log:2024-12-14 04:00:57.294 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'zigbee:device:62ec522f14:44e2f8fffe34b950' changed from INITIALIZING to UNKNOWN
events.log:2024-12-14 04:07:35.723 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'zigbee:device:62ec522f14:44e2f8fffe34b950' changed from UNKNOWN to ONLINE
events.log:2024-12-14 04:07:35.724 [INFO ] [penhab.event.FirmwareStatusInfoEvent] - Firmware status of thing zigbee:device:62ec522f14:44e2f8fffe34b950 changed to UNKNOWN.
openhab.log:2024-12-13 11:24:39.007 [WARN ] [converter.ZigBeeConverterSwitchOnoff] - 44E2F8FFFE34B950: OnOff converter is not linked to a server and cannot accept commands
openhab.log:2024-12-13 11:26:45.235 [WARN ] [converter.ZigBeeConverterSwitchOnoff] - 44E2F8FFFE34B950: OnOff converter is not linked to a server and cannot accept commands
openhab.log:2024-12-13 11:35:51.578 [WARN ] [converter.ZigBeeConverterSwitchOnoff] - 44E2F8FFFE34B950: OnOff converter is not linked to a server and cannot accept commands
openhab.log:2024-12-13 11:38:20.955 [WARN ] [converter.ZigBeeConverterSwitchOnoff] - 44E2F8FFFE34B950: OnOff converter is not linked to a server and cannot accept commands
openhab.log:2024-12-13 11:41:04.952 [WARN ] [converter.ZigBeeConverterSwitchOnoff] - 44E2F8FFFE34B950: OnOff converter is not linked to a server and cannot accept commands
openhab.log:2024-12-13 11:41:54.695 [WARN ] [converter.ZigBeeConverterSwitchOnoff] - 44E2F8FFFE34B950: OnOff converter is not linked to a server and cannot accept commands
openhab.log:2024-12-13 11:42:37.533 [WARN ] [converter.ZigBeeConverterSwitchOnoff] - 44E2F8FFFE34B950: OnOff converter is not linked to a server and cannot accept commands
openhab.log:2024-12-13 11:43:45.923 [WARN ] [converter.ZigBeeConverterSwitchOnoff] - 44E2F8FFFE34B950: OnOff converter is not linked to a server and cannot accept commands
openhab.log:2024-12-13 12:26:03.403 [WARN ] [converter.ZigBeeConverterSwitchOnoff] - 44E2F8FFFE34B950: OnOff converter is not linked to a server and cannot accept commands
openhab.log:2024-12-13 12:26:27.321 [WARN ] [converter.ZigBeeConverterSwitchOnoff] - 44E2F8FFFE34B950: OnOff converter is not linked to a server and cannot accept commands
openhab.log:2024-12-13 12:33:18.920 [WARN ] [converter.ZigBeeConverterSwitchOnoff] - 44E2F8FFFE34B950: OnOff converter is not linked to a server and cannot accept commands
root@raspi:/var/lib/docker/volumes/openhab3_data_openhab_log/_data# 
```

cdjackson, 2024-12-14:
> Is this log passage similar?

No - it's a completely different log with a very different meaning.

This means that you are trying to send commands to a channel that can't be commanded.

closed 2026-01-06 by cdjackson