openhab/org.openhab.binding.zigbee #893: Update the color channel HSB state when light is turned off.

andrewfg, 2025-01-03:
See #889

Currently the binding only updates the color HSB channel when the light is on, where the B part of the HSB channel shows the actual brightness. When the light is turned off, the B part of the HSB channel is not set to zero, but retains the prior B value.

In future the B part of the color HSB channel in OH should toggle b_actual=>0=>b_actual as the light is toggled on=>off=>on.

cdjackson, 2025-01-04:
> When the light is turned off, the B part of the HSB channel is not set to zero, but retains the prior B value.

Doesn't setting the state within OH to `OFF` do the same thing?  I would expect that if I set the state of a dimmer (or HSB) to `OFF`, then read back it's level, it will show '0'. I think that's what the binding does, and I expect that the UI correctly shows the HSB with B=0.

Or am I misunderstanding this?

> EDIT: this has the effect that a Switch item linked to the color HSB channel still shows on when the light is actually off.

I don't think so. If the device is turned `OFF`, then the binding receives an attribute update for the `OnOffCluster`. This will set the level to 0 (or as mentioned above, it sets the state to `OFF` which is I think the same).

I think that you are assuming that the `LevelControl` attribute is the only one in play here?

Or have you actually tested that this doesn't work on some devices? If so, it would be good to see a logfile showing what the device is sending.

The binding has been like this for a very long time, and this converter received a lot of work by someone who has a lot of bulbs.

andrewfg, 2025-01-04:
@cdjackson if the above is as you say, and an off command already sets b=0 then the version of the binding prior to #889 was just fine. And that is what I tested. However that makes your logic for #889 totally unnecessary since if b is already 0 then it doesn't matter if the channel is updated again with b=0. So perhaps you can explain what was the issue that you had detected for which #889 was a necessary solution?

cdjackson, 2025-01-04:
The issue is that the device can send a report that has b>0 but the device is off.  In this case we don’t want to set b>0 in OH since it’s actually OFF. Sent from my iPhone

andrewfg, 2025-01-04:
I never observed what you say. But I suppose other lights might do it. But then I wonder WHY a light would do such a thing?

cdjackson, 2025-01-04:
It definitely happens and this was reported as a regression in 4.3 on the forum. This is normal for ZigBee and if you have reporting configured to report the colour, then it will report the colour - even if it’s off. It’s the same with the level - in principal it’s possible with ZigBee to set the level without turning on the device (there are separate commands for setting the level, and setting level without on/off). Clearly this requires the ability to report the level independently from the on/off state. Sent from my iPhone

andrewfg, 2025-01-04:
This is actually the same as what Hue (or Govee) does. The color, brightness, and on/off state are orthogonal and asynchronous. The problem "only" arises in OH because we try to munge three orthogonal states into a common HSB channel. I have munging down to a fine art in Hue, but messed it up in Zigbee on the first attempt, but will have another go.

cdjackson, 2025-01-04:
I'm always open to improvements, but as above, this did have quite a lot of work a few years ago and was working fine until recently. The guy who reported the bug reverted back to 4.2 and said it was working fine there, so I think the change in #889 ought to restore this.

andrewfg, 2025-01-04:
If I would propose a new PR it would improve on the reverted code. It would follow and persist interim color and brightness changes whilst the device is off (as you do) but it would also allow OH channels to see the new color immediately, and see the new brightness immediately that the the device goes on again.

cdjackson, 2025-01-04:
I think your proposing to simply update H&S when the bulb is off - leaving B=0. This seems sensible and I'm certainly happy to review a PR on this.

andrewfg, 2025-01-04:
And restoring a prior B when state change is off to on..

cdjackson, 2025-01-04:
That's unnecessary.

If the device reports B=x, and you turn the device OFF then ON, B will still b equal to X. It will (and does) automatically restore to it's previous value.

PR #894 (2025-01-06): Update Color Channel Hue and Saturation when light is off
Resolves #893
closed 2025-01-06 by cdjackson